espruino/Espruino #1826: In jswrap_banglejs_init, the `jswrap_banglejs_ioWr(1,0); // ?` must be about GPS off

ikkala, 2020-05-21:
https://github.com/espruino/Espruino/blob/9d6370cbbfb2df5b1102b7847da4adacb030da2f/libs/banglejs/jswrap_bangle.c#L1363

> jswrap_banglejs_ioWr(1,0); // ?

Based on other similar lines in same file that must be putting GPS off, like in case of HMR off just above it:

> jswrap_banglejs_ioWr(IOEXP_GPS,0); // GPS off

gfwilliams, 2020-05-22:
Thanks! Just changed - not sure how that one happened :)

closed 2020-05-22 by gfwilliams

ikkala, 2020-05-22:
I was stumbled into this when pondering if it would be possible to send Assisted GPS data to the GPS receiver... At least UBlox model [at tech spec page](https://www.espruino.com/Bangle.js+Technical) says that it would be possible but haven't went very much further yet. UBlox AssistNow (mentioned in spec) would provide the data for registered company.. Have you tried/thought about this?

Can you point places in code where sending of this data to UBlox could be done?

Also, I didn't found GitHub issue yet recorded for the idea -- is this so much out of scope or from some reason impossible to do?

gfwilliams, 2020-05-22:
As I understood it, AGPS was only much use when you had an internet connection? If so it's something that'd need to be added to Gadgetbridge,etc.

Or can you add longer-term corrections that would eg make a fix faster for the next month?

Actually sending data to the GPS can be done from JavaScript. There's an example sending some simple commands here: https://github.com/espruino/BangleApps/blob/master/testing/GPS-comms.js

So it should be reasonably easy to extend for different GPS stuff.

I'd be very interested to see if you could come up with something - maybe even if it's just tweaking (enabling?) the Glonass/Beidou GPS stuff as well - although I thought it was on by default, or reducing power usage.

commit 3050d7b6 (2020-05-22): misc - fix #1826