espruino/Espruino #1827: Feature: Assisted GPS

ikkala, 2020-05-22:
Spin-off from discussion in #1826.

> As I understood it, AGPS was only much use when you had an internet connection? If so it's something that'd need to be added to Gadgetbridge,etc.
> 
> Or can you add longer-term corrections that would eg make a fix faster for the next month?

Indeed, I didn't either know the details that much before reading the UBlox spec (linked at [page](https://www.espruino.com/Bangle.js+Technical)): Yes, they have that online mode but they also support modes where almanac data is given only every now and then.

Also I found Polar has very good [layman explanation of A-GPS](https://support.polar.com/en/support/what_is_assisted_gps_a_gps_how_does_it_work) that gives us hint what is possible:

> The A-GPS data updates once a day. The latest A-GPS data file is automatically updated to your Polar device each time you sync it with the Flow web service via FlowSync software. If you sync your Polar device with the Flow web service via Flow app, the A-GPS data file update takes place once a week.
> 
> The A-GPS data file is valid for up to 14 days. Positioning accuracy is relatively high during the first three days, and progressively decreases during the remaining days. Regular updates help to ensure a fast satellite fix.

In that explanation you just replace the word "FlowSync" by "Bangle.js App Loader" and "Polar device" by "Bangle.js device" and we are ready.  ;)

It could be implemented by connecting to Gadgetbridge like you said, and/or like Beer Compass that way you could sync via laptop and via mobile phone. You just need internet connection.

> Actually sending data to the GPS can be done from JavaScript. There's an example sending some simple commands here: https://github.com/espruino/BangleApps/blob/master/testing/GPS-comms.js
> 
> So it should be reasonably easy to extend for different GPS stuff.

That looks very promising -- seeing hex that I saw in UBlox spec (`\xB5\x62\`) there.  :)

> I'd be very interested to see if you could come up with something - maybe even if it's just tweaking (enabling?) the Glonass/Beidou GPS stuff as well - although I thought it was on by default, or reducing power usage.

Let's see how I (or someone else?) have time -- weekend is coming.

gfwilliams, 2020-05-22:
This would be amazing! It should be easy to turn that into an App Loader app - as you say it's just going to be figuring out how to get hold of the latest AGPS data in a way that's easy & legal :)

gfwilliams, 2020-09-22:
Looking into this and it seems trivial to do - UBlox have a server where you can request the data and I'll set up a service to do this.

It's free, *however* you need an access token, and UBlox are taking forever to issue me with one.

ikkala, 2020-09-22:
Hi, and sorry been so silent lately -- have been busy with other duties, perhaps can participate this again soon.

Yes, the UBlox server sound good for me too.  But, if I remember correctly, I didn't have to go yet that far -- I was just testing locally the loading and sending of calibration data from/to watch, but we had those problems with the interrupts or such: The data didn't always transfer totally, it was a bit unreliable...  Perhaps this has evolved after spring?

ikkala, 2020-09-22:
> The data didn't always transfer totally, it was a bit unreliable... 

It was somehow related to this: https://github.com/espruino/Espruino/issues/1859

gfwilliams, 2020-09-22:
The issue would be that if you print a bunch of data at once then Espruino has to wait for all that data to be sent over Serial before it can execute other stuff. That means buffers get full and can't be handled in time.

That's pretty easy to sort out though - just send the String a bit at a time in setInterval, slow enough that all the data can be sent fast enough that it's not blocking.

However even if there's an overflow receiving data in Bangle.js the GPS will still have got all the data so it should still work ok.

ikkala, 2020-09-22:
Yep, I think I tried that trick (waiting enough before trying to debug print anything via console) but it didn't work. But, maybe I messed something -- I will retry that some day.  Or perhaps you could try if you can reliable dump the whole "navigation database" of the GPS receiver, so that you get same or sameish amount of rows there every time..?

gfwilliams, 2020-09-22:
Do you remember how you got the data to upload? Or do you think you could PM me your token for testing? I'm kind of stuck if UBlox aren't giving me a token to use :(

ikkala, 2020-09-22:
As said, I didn't go "that far" yet (=communicating with UBlox servers) -- I didn't have to because there's "navigation database" dump command in GPS receiver that you can use for backing up the current state of the receiver.  Then when you are able to dump that reliably you can try loading that back to the receiver.  :)

gfwilliams, 2020-10-08:
Now implemented: https://banglejs.com/apps/#assistedgps

closed 2020-10-08 by gfwilliams