espruino/Espruino #1828: Stack overflow bug in Storage module

DVR-Ltd, 2020-05-26:
I am experiencing regular problems with the Storage module on an ESP32 running Espruino v2.05. I have produced a script that replicates the problem based upon a real world example. Running on a freshly re-flashed ESP32, this script will cause a stack overflow:

```
var s    = require("Storage"),

	       //256 Char string:
	temp = "0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef" +
	       "0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef" +
	       "0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef" +
	       "0123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef",

	writeFile = (id, len) => {
		var a = 0,
			chunkSize = 0;

		console.log("Erasing " + id);
		s.erase(id);

		console.log("Writing " + id);
		while (a < len) {
			chunkSize = Math.min(256, len - a);

			s.write(id, temp.substr(0, chunkSize), a, (a === 0 ? len : undefined));

			a += chunkSize;
		}
	};

writeFile("web_svg", 1131);
writeFile("web_css", 6405);
writeFile("web_htm", 11249);
writeFile("web_js", 26127);

writeFile("web_svg", 1131);

writeFile("fileserv", 3106);
writeFile("jq_small", 859);
writeFile("parsemql", 1103);
writeFile("varint64", 2230);
writeFile("flex_es6", 5590);

while (true) {
	writeFile(".bootcde", 38693);
}
```

```
 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v05 (c) 2019 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

>Erasing web_svg
Writing web_svg
Erasing web_css
Writing web_css
Erasing web_htm
Writing web_htm
Erasing web_js
Writing web_js
Erasing web_svg
Writing web_svg
Erasing fileserv
Writing fileserv
Erasing jq_small
Writing jq_small
Erasing parsemql
Writing parsemql
Erasing varint64
Writing varint64
Erasing flex_es6
Writing flex_es6
Erasing .bootcde
Writing .bootcde
Erasing .bootcde
Writing .bootcde
Erasing .bootcde
Writing .bootcde
Erasing .bootcde
Writing .bootcde
Erasing .bootcde
Writing .bootcde
Erasing .bootcde
Writing .bootcde
***ERROR*** A stack overflow in task espruinoTask has been detected.
abort() was called at PC 0x40096ce4 on core 0

Backtrace: 0x40096b73:0x3ffcf440 0x40096ccb:0x3ffcf460 0x40096ce4:0x3ffcf480 0x0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2668
load:0x40078000,len:7304
load:0x40080000,len:5312
entry 0x40080274
E (540) spiram: SPI RAM enabled but initialization failed. Bailing out.

 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v05 (c) 2019 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

>
```

gfwilliams, 2020-05-27:
This sounds interesting - however I can't reproduce here (likely to do with file sizes) on an official board. Feels like it might be to do with storage compaction - maybe if the `jshGetFreeStack` function (or whatever it is called) doesn't report correctly, it could be trying to allocate too much memory on the stack.

If you can get some code to make this happen on a board I sell then I'll definitely look into it, but right now I don't have time to look into this for ESP32

DVR-Ltd, 2020-05-27:
I modified the code to include explicit calls to Storage.compact() after erase and littered it with console.log() calls. When it crashed, it was always during compaction.

Possibly related, calling Storage.eraseAll() shows this error message 64 times:
ERROR: >> jshKickWatchDog Not implemented,using taskwatchdog from RTOS

I'm not sure what I've done to my Wifi 1.1 board but it's not accepting being flashed with 2.05. I will dig out a spare I have somewhere.

gfwilliams, 2020-07-13:
Same issue as https://github.com/espruino/Espruino/issues/1777#issuecomment-655630823 - fixing this now

closed 2020-07-14 by gfwilliams