espruino/Espruino #1831: nRF52 OneWire unreliable

gfwilliams, 2020-05-29:
This has been known for a while - basically the softdevice can jump in via an IRQ to do BLE stuff and screw up timing whenever it wants to. It means that OneWire while BLE is active can be unreliable.

It seems there's not really a great way of using any hardware to do OneWire - it needs to be bit-banged. And disabling IRQs during OneWire activity can make the device unstable.

However it looks like we can actually use the timeslot API to request a period of time where the softdevice won't interfere:

https://devzone.nordicsemi.com/nordic/short-range-guides/b/software-development-kit/posts/setting-up-the-timeslot-api

hyhkjiy, 2021-09-08:
You are right, maybe it can be done with nRF5 SDK, but I haven't found an example of running one wire and BLE on nRF52 at the same time. I want to try it but I can't get started. Would you like to provide more reference information for this?

gfwilliams, 2023-03-22:
Forum thread on this too: https://forum.espruino.com/conversations/331390

Now fixed using the timeslot API. We could probably extend this to perform the entire transaction in one timeslot (instead of just individual writes)  but it works for now (albeit slightly more slowly)

commit 5a18c873 (2023-03-22): nRF52: OneWire accesses are now scheduled with the Timeslot API to avoid the softdevice interfering (fix #1831)
closed 2023-03-22 by gfwilliams