espruino/Espruino #1832: Espruino ESP32 missing "ws" module?

Zibri, 2020-05-29:
> var WebSocket = require("ws");
Uncaught Error: Module ws not found
 at line 1 col 29
var WebSocket = require("ws");

also:

>wifi.getHostByName("google.com",console.log);
WARNING: Wifi.getHostByName
WARNING: Wifi.getHostByName: google.com
WARNING: Error: -5 from dns_gethostbyname
WARNING: Wifi.getHostByName CB - google.com 0
=undefined
null
WARNING: Wifi.getHostByName CB - google.com 3ffc78f8

MaBecker, 2020-05-30:
WS is a JS module 

check this page http://www.espruino.com/ws

closed 2020-05-30 by MaBecker

Zibri, 2020-06-01:
> WS is a JS module
> 
> check this page http://www.espruino.com/ws

WebSockets are an integral part of today web comunication.
so they should be included by default, perhaps even converted in internal calls.

MaBecker, 2020-06-01:
Well some parts are easyer to add via module like ws or mqtt.

Espruino is following the concept of adding modules via javascript to keep the firmware small.

You still have the possibility to build your own firmware and add what ever is needed for your projects.

Zibri, 2020-06-01:
I understand, but when I do var WebSocket = require("ws");

It just doesn't work even if ESP32 is connected to my access point.
or should I write:
require("http://www.espruino.com/modules/ws.js") ?

Zibri, 2020-06-01:
```
>var WebSocket = require("ws");
Uncaught Error: Module ws not found
 at line 1 col 29
var WebSocket = require("ws");
                            ^
>var WebSocket = require("http://www.espruino.com/modules/ws.js");
Uncaught Error: Module http://www.espruino.com/modules/ws.js not found
 at line 1 col 64
...espruino.com/modules/ws.js");
                              ^
> 
```

Zibri, 2020-06-01:
```
>wifi.ping("192.168.0.1",console.log);
=undefined
{ "totalCount": 1, "totalBytes": 60, "totalTime": 18, "respTime": 18,
  "seqNo": 1, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 2, "totalBytes": 120, "totalTime": 25, "respTime": 7,
  "seqNo": 2, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 3, "totalBytes": 180, "totalTime": 46, "respTime": 21,
  "seqNo": 3, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 4, "totalBytes": 240, "totalTime": 75, "respTime": 29,
  "seqNo": 4, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 5, "totalBytes": 300, "totalTime": 89, "respTime": 14,
  "seqNo": 5, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 5, "totalBytes": 300, "totalTime": 89, "respTime": 14,
  "seqNo": 6, "timeoutCount": 0, "bytes": 60, "error": 0 }
> 
```

but..

```
>wifi.getHostByName("google.com",console.log);
WARNING: Wifi.getHostByName
WARNING: Wifi.getHostByName: google.com
WARNING: Error: -5 from dns_gethostbyname
WARNING: Wifi.getHostByName CB - google.com 0
=undefined
null
WARNING: Wifi.getHostByName CB - google.com 3ffc78f8
> 
```

gfwilliams, 2020-06-01:
Are you using the Web IDE? And uploading using code on the right hand side? http://www.espruino.com/Troubleshooting#i-get-uncaught-error-module-xyz-not-found-

Zibri, 2020-06-01:
hmm yes.. but what does it have to do with 
`wifi.getHostByName("google.com",console.log);`
?

Zibri, 2020-06-01:
> Are you using the Web IDE? And uploading using code on the right hand side? http://www.espruino.com/Troubleshooting#i-get-uncaught-error-module-xyz-not-found-

oh.. I got it, the require is interpreted by the IDE and not by the firmware.. very strange implementation.
Anyway, the getHostByName does not work.

Zibri, 2020-06-01:
WebSocket is working but host resolution is NOT.

`Uncaught InternalError: Unable to locate host`