espruino/Espruino #1833: ESP32 http server problem

Zibri, 2020-05-29:
```
function onPageRequest(req, res) { 
  var a = url.parse(req.url, true);
  if (a.pathname.substr(-1)=="/") { // a slash at the end, list the directory
    res.writeHead(200, {'Content-Type': 'text/html'});
    res.write("<html><body><p>Temp is "+Math.round(((E.getTemperature()-32)*5/9) * 100) / 100+" C </p>");    
    res.end("</body></html>");
  } 
}

require("http").createServer(onPageRequest).listen(80);

```

using this code, after 8-10 refreshes of the page the server stops without any error.

>process.env
={
  VERSION: "2v05.118",
  GIT_COMMIT: "b3403b73",
  BOARD: "ESP32",
  FLASH: 0, STORAGE: 262144, RAM: 524288,
  SERIAL: "a4cf120a-7ef4",
  CONSOLE: "Serial1",
  MODULES: "Flash,Storage,hea" ... "r,crypto,neopixel",
  EXPTR: 1073484860 }

MaBecker, 2020-06-01:
This is what I do to avoid favicon calls

http://forum.espruino.com/comments/14551636/

Zibri, 2020-06-01:
Still.. the socket blocking should *not* crash the ESP32.

MaBecker, 2020-06-01:
>  Still.. the socket blocking should not crash the ESP32

ok, you are welcome tofixit - it’s open source.

opichals, 2020-06-01:
> using this code, after 8-10 refreshes of the page the server stops without any error.

I wonder what is the expected behavior. I can imagine the server should probably start refusing connections on TCP level in case the server implementation is blocking sockets for any reason too long to cope with the client load.

> the socket blocking should not crash the ESP32.

@Zibri From what you are writing above it did not crash it or did it?

Zibri, 2020-06-01:
> @Zibri From what you are writing above it did not crash it or did it?

the second code does not crash it.

opichals, 2020-06-03:
@Zibri Did it crash or did it just not respond?

Zibri, 2020-06-03:
The first code does not crash it but it stops responding and even a reboot doesn't solve the problem, I must physically disconnect it.

The second code just works because it doesn't leave any connection hanging...

MaBecker, 2020-08-31:
For embedded system you easy run out of sockets if you not clean  them up properly
like ignoring a favicon request. 

```JS
var pc=0;
function onPageRequest(req, res) { 
  var a = url.parse(req.url, true);
  if (a.pathname.substr(-1)=="/") { // a slash at the end, list the directory
    temp = Math.round(((E.getTemperature()-32)*5/9) * 100) / 100;
    pc++;
    res.writeHead(200, {'Content-Type': 'text/html'});
    res.write(`<!DOCTYPE html>
<head><link rel="icon" href="data:,"></head>         // clean up the icon request if you not handle it
<body>
<p>Temp is ${temp} C</p>
<p>page load counter is ${pc}</p>
</body></html>
`);
    res.end();
  }
}

http = require("http");
http.createServer(onPageRequest).listen(80);

```

closed 2020-08-31 by MaBecker