espruino/Espruino #1838: Bangle.js: GPS-raw event strings vs. UBX Protocol binary messages

ikkala, 2020-06-01:
When investigating different approaches to the [A-GPS](https://github.com/espruino/Espruino/issues/1827) realized that the current GPS-raw event, providing messages as strings, works well with NMEA messages introduced in u-blox GPS receiver spec (linked [at page](https://www.espruino.com/Bangle.js+Technical)) but when it comes to u-blox specific UBX protocl, the events are failing to provide all the data.

For example, [the UBX-ACK-NAK (0x05 0x00) messages at GPS-comms.js example](https://github.com/espruino/BangleApps/blob/d8c3b73334c8fdb471f71b9c4a9091873182f752/testing/GPS-comms.js#L16) never come fully through because the zero byte is terminating the string. The message payload coming after the zero is missing totally.

Basically all messages containing zero bytes are truncated.

To fix this, either:
- Change GPS-raw event to provide ArrayBuffer instead of string (ArrayBuffer is trivial to convert to string is JS land when needed)
- or, introduce another event type for u-blox messages

In both options, the underlying implementation must understand which of the protocol it is currently receiving.

gfwilliams, 2020-06-01:
Hi - There is actually a third option which I think might be more sensible and not annoy people currently using GPS-raw.

Strings can contain zeros, so we just need to tweak the way we create the string such that zeros are not a big deal?

ikkala, 2020-06-01:
> Strings can contain zeros, so we just need to tweak the way we create the string such that zeros are not a big deal?

Sounds good... How will that go? (yes, I don't do C/C++ as my day job)

> not annoy people currently using GPS-raw

Btw, have people used already GPS-raw a lot? At least in the apps in BangleApps it's not used. And I tried to scan through issue trackers (BangleApps and Espruino) and didn't find trace that GPS-raw would have been yet used that much.

As an abstraction, the u-blox UBX protocol messages are binary, not strings. Of course, providing events as strings lower the bar to people -- in that sense the string is good choice.  :)

Anyway, the underlying logic (or similar) is need to separate the very string based NMEA protocol from the UBX binary protocol. I tried to do my best to be fault tolerant there.

ikkala, 2020-06-02:
Ok, I learned the code base a bit more and if [in pull requests row](https://github.com/espruino/Espruino/pull/1839/files#diff-418f7b9de67288c8abbc7e03bd8d4fa4R1652) I change the jsvNewArrayBufferWithData to jsvNewStringOfLength the GPS-raw events are strings but zeros are coming through nicely too.

Would this change be enough?

gfwilliams, 2020-06-02:
> have people used already GPS-raw a lot?

There are one or two (eg picking up satellite fix info). Unfortunately what you see online is only a small proportion of what happens - many people play around and then don't share what they do publicly, and mostly I only know when something breaks and they complain :)

> jsvNewArrayBufferWithData to jsvNewStringOfLength

Yes, I think that'd be perfect. In JS it's easy enough to change back with `E.toUint8Array/similar`.

Just checking, is the whole 172 byte buffer definitely required? Given the relative lack of memory ideally we'd keep statically allocated buffers to a minimum, but if that's what is needed for a useful message I guess that's what we have to have.

thanks!

ikkala, 2020-06-02:
> and mostly I only know when something breaks and they complain :)

This typical, usually we "don't have time" to go and thank about good work (though happily I have seen some thanks to you already here in issue tracker already), so let's try to fix that: Thanks for the interesting project!  :)

> > jsvNewArrayBufferWithData to jsvNewStringOfLength
>
> Yes, I think that'd be perfect. In JS it's easy enough to change back with E.toUint8Array/similar.

Yup, it's now there, in pull request.

> Just checking, is the whole 172 byte buffer definitely required?

Actually that's the thing I'm too a bit unsure here: As the comment says (`UBX-MGA-DBD Navigation database dump entry message max size`) it's based on that one specific message type length, and this is just a compromise. I'm not sure how big all of the UBX message can be...

On the other hand, I understood from the spec that NMEA protocol messages can also be bigger than 82 bytes -- there's a special "Limit82 Mode" where you explicitly say to receiver to keep the messages short but I guess the mode is not currently used (?).

gfwilliams, 2020-06-02:
Thanks!

Wrt the message size: I don't enable the Limit82 mode but I never actually hit issues with message length for the default set of enabled messages.

However you're saying that for the stuff you're wanting to do at the moment, you probably wouldn't have issues even with the 82 byte buffer?

I'm just thinking maybe there's a better solution here - keep the original 82 byte buffer, then if that buffer gets full, queue it with JSBT_GPS_PARTIAL_DATA and append it to a JS string (which can grow in size pretty much without limit) - or for now we could just push it as a gps-raw event and let any code on the JS side that's expecting a bigger message do the reconstruction - that'd be extremely easy.

I know it seems silly over a few bytes, but we have extremely limited memory - another 172 bytes is basically 11 JS variables that have to get dropped for available program code.

If you could find a UBX command that would dump enough data that oversize strings could really be tested then it'd really help to make sure it was working ok though.

So for now I'll merge this, and maybe create an issue to drop the buffer size and reconstruct into a JS var...

PR #1839 (2020-06-02): GPS-raw event: handle UBX protocol
Bangle.js: Changed GPS-raw event type from string to ArrayBuffer
and handle u-blox UBX Protocol messages.

Fixes #1838
commit 0574da31 (2020-06-02): GPS-raw event ArrayBuffer, handle UBX protocol

Bangle.js: Changed GPS-raw event type from string to ArrayBuffer
and handle u-blox UBX Protocol messages.

Fixes #1838
closed 2020-06-02 by gfwilliams