espruino/Espruino #1840: Espruino ESP32 dns revolver not working.

Zibri, 2020-06-01:
```
>wifi.getHostByName("google.com",console.log)
WARNING: Wifi.getHostByName
WARNING: Wifi.getHostByName: google.com
WARNING: Error: -5 from dns_gethostbyname
WARNING: Wifi.getHostByName CB - google.com 0
=undefined
null
WARNING: Wifi.getHostByName CB - google.com 0
```
but

```
>wifi.ping("192.168.0.1",console.log)
=undefined
{ "totalCount": 1, "totalBytes": 60, "totalTime": 67, "respTime": 67,
  "seqNo": 1, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 2, "totalBytes": 120, "totalTime": 90, "respTime": 23,
  "seqNo": 2, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 3, "totalBytes": 180, "totalTime": 98, "respTime": 8,
  "seqNo": 3, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 4, "totalBytes": 240, "totalTime": 107, "respTime": 9,
  "seqNo": 4, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 5, "totalBytes": 300, "totalTime": 191, "respTime": 84,
  "seqNo": 5, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 5, "totalBytes": 300, "totalTime": 191, "respTime": 84,
  "seqNo": 6, "timeoutCount": 0, "bytes": 60, "error": 0 }
>wifi.ping("172.217.21.14",console.log)
=undefined
{ "totalCount": 1, "totalBytes": 60, "totalTime": 317, "respTime": 317,
  "seqNo": 1, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 2, "totalBytes": 120, "totalTime": 530, "respTime": 213,
  "seqNo": 2, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 3, "totalBytes": 180, "totalTime": 752, "respTime": 222,
  "seqNo": 3, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 4, "totalBytes": 240, "totalTime": 966, "respTime": 214,
  "seqNo": 4, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 5, "totalBytes": 300, "totalTime": 1168, "respTime": 202,
  "seqNo": 5, "timeoutCount": 0, "bytes": 60, "error": 0 }
{ "totalCount": 5, "totalBytes": 300, "totalTime": 1168, "respTime": 202,
  "seqNo": 6, "timeoutCount": 0, "bytes": 60, "error": 0 }
>
```
for some reason this works:

```
>ws=new WebSocket("ws://echo.websocket.org",{
:  origin: 'https://zibri.org'
:})
={ socket: null,
  host: "ws://echo.websocket.org",
  port: 80, "protocolVersion": 13,
  origin: "https://zibri.org",
  keepAlive: 60000, masking: true,
  path: "/",
  protocol: undefined,
  lastData: "",
  key: {
    source: "bm15dTg5ajZma3UwNHJ1Mw==",
    hashed: "16jsc5MJ8YrTu40cvtObYrHsVr4="
   },
  connected: undefined,
  headers: {  }
 }
> 

```

but this doesn't:
```
>ws=new WebSocket("ws://tjs.dyns.gq/testing",{
:  origin: 'https://zibri.org'
:})
Uncaught InternalError: Unable to locate host
 at line 1 col 81
...},this.onConnect.bind(this))
                              ^
in function "initializeConnection" called from line 1 col 41
var c=new g(a,b);c.initializeConnection();return c
                                        ^
in function "WebSocket" called from line 3 col 2
})

```
but as you can see the host exists
https://dnschecker.org/#A/tjs.dyns.gq
and
https://tjs.dyns.gq

MaBecker, 2020-08-30:
Well not sure what's going wrong here....

First call returns `null`, second the ip plus the information ` Already resolved`  

```JS

>wifi= require('Wifi')
=function () { [native code] }
>wifi.getIP()
={
  ip: "192.168.194.148",
  netmask: "255.255.255.0",
  gw: "192.168.194.1",
  mac: "a4:cf:12:76:64:b4"
 }
>wifi.getHostByName("google.com",(ip)=>{console.log(ip);});
WARNING: Wifi.getHostByName
WARNING: Wifi.getHostByName: google.com
WARNING: Error: -5 from dns_gethostbyname
WARNING: Wifi.getHostByName CB - google.com 0
=undefined
null
WARNING: Wifi.getHostByName CB - google.com 3ffc78c0
>wifi.getHostByName("google.com",(ip)=>{console.log(ip);});
WARNING: Wifi.getHostByName
WARNING: Wifi.getHostByName: google.com
WARNING: Already resolved
WARNING: Wifi.getHostByName CB - google.com 3ffdd0e0
=undefined
224.208.253.63
> 

```

For my taste too much jsWarn().

MaBecker, 2022-11-22:
Same result for esp.idf 3.3.6 - so it might be a ESP32 implementation bug.

MaBecker, 2024-03-22:
Remove all warning, check branch https://github.com/espruino/Espruino/tree/esp32_cleanup_jswarn

closed 2024-03-22 by MaBecker