espruino/Espruino #1843: Bangle.js: Save RAM in GPS handling

gfwilliams, 2020-06-02:
After https://github.com/espruino/Espruino/pull/1839 - and discussed in https://github.com/espruino/Espruino/issues/1838

Basically most messages from GPS are <82 chars, but it seems they *can* be quite long with UBX protocol.

We should drop the size of the line buffer back to 82 chars, and then just store larger strings (if they exist) in a JS var.

ikkala, 2020-06-02:
From: https://github.com/espruino/Espruino/issues/1838#issuecomment-637385011

> Wrt the message size: I don't enable the Limit82 mode but I never actually hit issues with message length for the default set of enabled messages.
> 
> However you're saying that for the stuff you're wanting to do at the moment, you probably wouldn't have issues even with the 82 byte buffer?

I communicated unclear: The `UBX-MGA-DBD` "Navigation database dump" entry message max size is really 172 bytes. From there came the Buffer size increase -- I want to play with that because storing of the nav db content could help with [A-GPS implementation](https://github.com/espruino/Espruino/issues/1827).

> I'm just thinking maybe there's a better solution here - keep the original 82 byte buffer, then if that buffer gets full, queue it with JSBT_GPS_PARTIAL_DATA and append it to a JS string (which can grow in size pretty much without limit) - or for now we could just push it as a gps-raw event and let any code on the JS side that's expecting a bigger message do the reconstruction - that'd be extremely easy.

These options sounds good!  Though, my skills with this code base are not yet there so that I could implement this.  :)

Let's see what A-GPS actually need -- you know, it's possible that it cannot be done properly in user-land (=JS), but it must be done in Espruino code.

> I know it seems silly over a few bytes, but we have extremely limited memory - another 172 bytes is basically 11 JS variables that have to get dropped for available program code.

No problem, I totally understand this -- have been playing with very low memory microcontrollers earlier in life...

> If you could find a UBX command that would dump enough data that oversize strings could really be tested then it'd really help to make sure it was working ok though.

The mentioned `UBX-MGA-DBD` "Navigation database dump", when I right now run it against my Bangle.js it gives for:

```
Bangle.setGPSPower(1);
//Bangle.on('GPS',print);

Bangle.on('GPS-raw',function (d) {
  if (d[0]=="$") return;
  if (d.startsWith("\xB5\x62\x05\x01")) print("GPS ACK");
  else if (d.startsWith("\xB5\x62\x05\x00")) print("GPS NACK");
  else { // if (!d.startsWith("\x0x24")) {
    print("GPS-raw:", typeof d, d.length);
    print("GPS-raw str:", d);
    print("GPS", E.toUint8Array(d).join(","));
  }
});
function writeGPScmd(cmd) {
  var d = [0xB5,0x62]; // sync chars
  d = d.concat(cmd);
  var a=0,b=0;
  for (var i=2;i<d.length;i++) {
    a += d[i];
    b += a;
  }
  d.push(a,b);
  Serial1.write(d);
}
writeGPScmd([0x13, 0x80, 0, 0]);
```

lots of rows including these:

```
GPS-raw: string 54
GPS-raw str: µb4&àJc°D¿c#&ý½¡Rýc.ùg§é
GPS 181,98,19,128,52,0,2,0,0,30,20,6,18,16,38,28,0,0,224,74,1,3,29,99,176,68,191,151,1,99,35,38,158,253,189,12,161,0,82,253,0,99,25,46,249,0,11,140,139,0,103,167,233,0
GPS-raw: string 54
¡DýcØÌ3·´ý| &àJc°DÛp¡Ú  ?
GPS 181,98,19,128,52,0,2,0,0,32,20,6,18,16,38,28,0,0,224,74,1,3,31,99,176,68,219,112,3,161,218,32,18,9,63,13,161,0,68,253,0,99,194,216,204,0,51,183,156,0,180,253,124,0
GPS-raw: string 126
GPS-raw str: µbxÿàJ0Áà1À2³´¨GÀGÇÉ
GPS 181,98,19,128,120,0,3,0,0,255,0,0,0,0,0,0,0,0,224,74,1,3,25,0,0,20,1,0,0,0,1,0,0,0,0,48,6,0,60,8,137,8,18,7,18,0,193,3,0,0,25,0,0,21,0,0,224,49,0,0,192,50,0,0,128,179,0,0,0,180,0,0,168,71,0,0,192,71,0,0,128,199,0,0,0,201,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
GPS-raw: string 94
r­3}Q@Ä» Ð¿õ`Åÿ'g~mü Vjÿz:ªlY ÐK©S­EóëTÖèÿkgÝÄ@g1
GPS 181,98,19,128,88,0,5,0,0,14,20,6,5,2,30,3,6,30,224,74,1,3,13,114,173,20,148,51,1,29,60,8,1,125,81,0,64,196,187,0,160,19,159,208,191,245,96,197,2,194,255,39,103,126,109,252,4,9,86,106,255,31,122,58,0,0,170,108,89,32,208,21,24,75,169,83,141,173,69,243,139,235,84,214,232,255,133,107,103,221,196,64,103,49
GPS-raw: string 94
 vÿz:ªl Ð!½6W'û®ç.¦GhfX÷OýÄáÿ¯l½?*ÁâA=)ç[
GPS 181,98,19,128,88,0,5,0,0,24,20,6,4,14,15,3,6,30,224,74,1,3,23,126,173,20,79,19,1,5,60,8,1,253,130,0,0,196,225,255,175,19,108,189,63,3,42,193,226,65,61,41,231,91,137,13,3,9,118,25,255,7,122,58,0,0,170,108,137,32,208,135,33,189,146,54,87,22,143,39,251,174,135,231,133,46,166,71,104,23,102,88,247,31
GPS-raw: string 94
GPS-raw str: µbXàJ­ïH@D&ÿ¯:ô@Á7)ïÚ} \².
z:ªl øôãñ«ù.TÝã#õ4ÏB
GPS 181,98,19,128,88,0,5,0,0,28,20,6,4,14,30,3,6,30,224,74,1,3,27,126,173,20,239,154,3,72,60,8,1,129,129,0,64,68,38,255,175,19,58,19,0,244,64,193,2,2,55,41,199,8,239,218,125,9,92,178,46,10,122,58,0,0,170,108,137,32,248,144,244,227,241,171,147,135,7,11,249,151,46,84,221,227,35,245,4,52,18,207,66,6

```

Note for example line `GPS-raw: string 126`.

ps. Copy-pasting from Espruino Web IDE is quite challenging...

gfwilliams, 2020-06-03:
Just looking at this - do you have an example of something that triggers:

```
if (ubloxMsgPayloadEnd < ubloxCount) { // Length is some odd way horribly wrong
```

So I can test?

ikkala, 2020-06-03:
Yep, that's for sanity. 

And, speaking about the checks, I have also the UBX protocol checksum checking pull request coming to you -- and the fix also contains providing the checksum number to GSP-raw event (they are currently missing).

ikkala, 2020-06-03:
...as promised: https://github.com/espruino/Espruino/pull/1844

commit 162d6a3e (2020-06-03): Bangle.js: Save RAM in GPS handling, allow arbitrary length data returned (fix #1843)Reduce GPS buffer size
closed 2020-06-03 by gfwilliams