espruino/Espruino #1847: GPS-raw produces odd lines after the renewal that saves RAM

ikkala, 2020-06-04:
With latest 4ed174d954df5f8cfcb2b776ac3c903470d11011 code, including the GPS-raw renewal that saves RAM (162d6a3e0f0867192f75cefd9de6d244b1fccbc3) an UBX-MGA-DBD "Poll the navigation database" `0xB5,0x62, 0x13, 0x80, 0, 0` command that dumps navigation database produces the following:
```
0xb5 0x62 0x13 0x80 0x54 0x0 0x1 0x0 0x0 0xa 0x14 0x6 0x3 0x13 0x3b 0x0 0x4 0x0 0xe0 0x4a 0x1 0x3 0x9 0x24 0xa4 0x44 0x17 0x17 0x1 0x4d 0x53 0xa7 0xff 0x5 0x6b 0x45 0x24 0xd0 0xab 0x12 0xe0 0x2 0xa7 0xdc 0x62 0xa2 0x7c 0x70 0x57 0x27 0x14 0xc1 0xbd 0x94 0x5e 0x3f 0xd 0xa1 0x1a 0x4f 0x9d 0xff 0x26 0x3 0xb8 0x33 0x8e 0x2 0x42 0xd 0xdd 0x20 0x1a 0x4f 0x60 0x0 0xcd 0xff 0x0 0x68 0x3 0xf 0xf1 0x2e 0x34 0x0 0x48 0xe0 0x1 0x0 0x43 0x39
0x3a 0x53 0x17 0x0 0xa8 0xe0 0x1 0x0 0x2d 0xa1 0x3a 0x53 0x17 0x0 0xa8 0xe0 0x1 0x0 0x2d 0xa1
0xb5 0x62 0x13 0x80 0x54 0x0 0x1 0x0 0x0 0x20 0x14 0x6 0x3 0x13 0x3b 0x0 0x4 0x0 0xe0 0x4a 0x1 0x3 0x1f 0x20 0xa4 0x44 0x88 0x3c 0x1 0x39 0x58 0xa8 0xff 0x1 0x3a 0x9f 0x99 0x98 0x50 0x6b 0xd 0x2 0xe1 0xcf 0xdf 0xcc 0x8b 0xeb 0xf7 0x26 0xa9 0x67 0xc0 0x9c 0x8e 0xb9 0xd 0xa1 0x1a 0x4f 0x3f 0x0 0x25 0xfb 0xce 0x32 0xe3 0xfb 0x13 0x11 0x73 0x1b 0x1a 0x4f 0x9 0x0 0xf3 0xff 0x0 0xe 0x3a 0xf 0xd9 0x9b 0x9 0x0 0x40 0xe0 0x1 0x0 0x9e 0x5c
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x1 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x0 0x7b 0xb0 0x44 0x43 0x76 0x1 0xfa 0x72 0x51 0xe2 0x18 0xb4 0xc 0xa1 0x0 0x6d 0xfd 0x0 0x7b 0xeb 0x14 0x78 0x0 0x2c 0xb 0x20 0x0 0x62 0x6d 0x52 0x0 0x69 0x6 0x0 0x3c 0x4c 0xbf 0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x1 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x0 0x7b 0xb0 0x44 0x43 0x76 0x1 0xfa 0x72 0x51 0xe2 0x18 0xb4 0xc 0xa1 0x0 0x6d 0xfd 0x0 0x7b 0xeb 0x14 0x78 0x0 0x2c 0xb 0x20 0x0 0x62 0x6d 0x52 0x0 0x69 0x6 0x0 0x3c 0x4c 0xbf
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x6 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x5 0x7b 0xb0 0x4 0x1b 0x6c 0x1 0xe5 0x6c 0xf 0x98 0x18 0x1f 0xc 0xa1 0x0 0x6c 0xfd 0x0 0x7b 0xf5 0xbe 0x77 0x0 0x94 0xfe 0xd4 0x0 0x17 0xc5 0x58 0x0 0xd6 0xf6 0x3f 0x3c 0x18 0x7a
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x8 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x7 0x7b 0xb0 0x44 0x7 0xee 0x1 0x1d 0xc9 0x2b 0x8a 0x11 0xfe 0xc 0xa1 0x0 0x2d 0xfd 0x0 0x7b 0x8a 0x7b 0x4c 0x0 0xc9 0x4b 0xfd 0x0 0x47 0xa9 0xae 0x0 0xda 0x7 0x80 0x3c 0x98 0xee
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x15 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x14 0x7b 0xb0 0xc4 0xf 0xec 0x1 0x63 0x90 0xc3 0x3a 0x7 0xc2 0x18 0xa1 0x0 0x54 0xfd 0x0 0x7b 0xd4 0xd 0x75 0x0 0xc4 0xc 0xca 0x0 0x88 0xf2 0xfb 0x0 0x8 0x8 0x80 0x3c 0xe 0xae
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x18 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x17 0x7b 0xb0 0xc4 0x85 0x5e 0x3 0xaa 0x5e 0x4f 0x3d 0xfc 0xcd 0xc 0xa1 0x0 0x52 0xfd 0x0 0x7b 0x3a 0x49 0xf5 0x0 0x87 0x25 0x1b 0x0 0x3 0x5e 0x5b 0x0 0xf1 0x7 0x80 0x3c 0x12 0xc7
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x1a 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x19 0x7b 0xb0 0x4 0xa6 0x2f 0x1 0x98 0xe5 0x26 0xc9 0x2 0xa2 0xc 0xa1 0x0 0x3a 0xfd 0x0 0x7b 0x32 0x7 0x21 0x0 0x53 0x36 0x9 0x0 0x20 0x93 0xf4 0x0 0xe7 0x10 0x0 0x3c 0xf8 0xb6
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x1c 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x1b 0x7b 0xb0 0x4 0x99 0xb3 0x3 0xcd 0xcc 0x96 0xd3 0x16 0xfd 0xc 0xa1 0x0 0x4f 0xfd 0x0 0x7b 0x15 0x35 0x26 0x0 0x8d 0xc2 0xc6 0x0 0xac 0x59 0xc7 0x0 0xea 0xfa 0x3f 0x3c 0x79 0x7b
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x1e 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x1d 0x7b 0xb0 0x44 0x28 0xf2 0x1 0x67 0x2b 0x26 0x9a 0xfd 0xb6 0xc 0xa1 0x0 0x52 0xfd 0x0 0x7b 0xe2 0x25 0xf9 0x0 0xd8 0x96 0x8b 0x0 0x14 0xd3 0x31 0x0 0x9 0xf7 0xbf 0x3c 0xd8 0x4a
```

Note the second message: It doesn't start by the UBX protocol header.
Also, fourth line: It has the `0xb5 0x62 0x13 0x80` sequence inside of it, not only in the beginning.

The commit before the RAM saving with big static buffer produces following:

```
0xb5 0x62 0x13 0x80 0x54 0x0 0x1 0x0 0x0 0xa 0x14 0x6 0x3 0x13 0x3b 0x0 0x4 0x0 0xe0 0x4a 0x1 0x3 0x9 0x24 0xa4 0x44 0x17 0x17 0x1 0x4d 0x53 0xa7 0xff 0x5 0x6b 0x45 0x24 0xd0 0xab 0x12 0xe0 0x2 0xa7 0xdc 0x62 0xa2 0x7c 0x70 0x57 0x27 0x14 0xc1 0xbd 0x94 0x5e 0x3f 0xd 0xa1 0x1a 0x4f 0x9d 0xff 0x26 0x3 0xb8 0x33 0x8e 0x2 0x42 0xd 0xdd 0x20 0x1a 0x4f 0x60 0x0 0xcd 0xff 0x0 0x68 0x3 0xf 0xf1 0x2e 0x34 0x0 0x48 0xe0 0x1 0x0 0x43 0x39
0xb5 0x62 0x13 0x80 0x54 0x0 0x1 0x0 0x0 0x1c 0x14 0x6 0x3 0x13 0x3b 0x0 0x4 0x0 0xe0 0x4a 0x1 0x3 0x1b 0x54 0xa4 0x44 0x4 0x58 0x3 0xb8 0x64 0xa8 0xff 0xe8 0xf0 0x4a 0xb5 0x9a 0x60 0x4c 0x6d 0x9 0xa8 0x52 0x44 0x26 0x6a 0x21 0xd3 0x27 0x2b 0xf1 0xbd 0xc6 0x17 0x56 0xd 0xa1 0x19 0x4f 0xe2 0xff 0xfe 0x9 0xd6 0x2f 0xfc 0x8 0xd1 0x2 0x8d 0x2c 0x19 0x4f 0x50 0x0 0x7d 0x0 0x0 0x83 0x4 0xf 0x3a 0x53 0x17 0x0 0xa8 0xe0 0x1 0x0 0x2d 0xa1
0xb5 0x62 0x13 0x80 0x58 0x0 0x1 0x0 0x3 0x10 0x14 0x6 0x3 0x13 0x3b 0x0 0x4 0x0 0xe0 0x4a 0x1 0x3 0x7a 0x59 0xa4 0x44 0x3d 0x73 0x1 0xfe 0x96 0xd2 0x8b 0xf 0x45 0xb9 0xc3 0xb 0xb6 0x8f 0xc0 0xb 0x82 0xe 0x0 0x0 0xe0 0xff 0x1f 0xc8 0x7a 0x2a 0x40 0x0 0x62 0x1 0x38 0x4e 0x2a 0xcc 0x3b 0x4e 0xc3 0xe5 0xad 0x12 0x5f 0xea 0xff 0x0 0x2a 0x57 0xeb 0xca 0x5 0x2d 0x7a 0x1 0xfd 0xd2 0x49 0x97 0x91 0x4c 0x84 0xa0 0x61 0x2c 0xad 0xd0 0xe0 0x12 0x1c 0x27 0x7 0x1d
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x4 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x3 0x7b 0xb0 0x4 0xa0 0x7c 0x1 0x33 0x35 0x6 0x1f 0xb 0xe1 0xc 0xa1 0x0 0x48 0xfd 0x0 0x7b 0x29 0x6d 0xce 0x0 0x42 0xdd 0x95 0x0 0x8e 0xf7 0x7f 0x0 0x99 0xff 0x3f 0x3c 0xf3 0xc1
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x6 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x5 0x7b 0xb0 0x4 0x1b 0x6c 0x1 0xe5 0x6c 0xf 0x98 0x18 0x1f 0xc 0xa1 0x0 0x6c 0xfd 0x0 0x7b 0xf5 0xbe 0x77 0x0 0x94 0xfe 0xd4 0x0 0x17 0xc5 0x58 0x0 0xd6 0xf6 0x3f 0x3c 0x18 0x7a
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x8 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x7 0x7b 0xb0 0x44 0x7 0xee 0x1 0x1d 0xc9 0x2b 0x8a 0x11 0xfe 0xc 0xa1 0x0 0x2d 0xfd 0x0 0x7b 0x8a 0x7b 0x4c 0x0 0xc9 0x4b 0xfd 0x0 0x47 0xa9 0xae 0x0 0xda 0x7 0x80 0x3c 0x98 0xee
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x15 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x14 0x7b 0xb0 0xc4 0xf 0xec 0x1 0x63 0x90 0xc3 0x3a 0x7 0xc2 0x18 0xa1 0x0 0x54 0xfd 0x0 0x7b 0xd4 0xd 0x75 0x0 0xc4 0xc 0xca 0x0 0x88 0xf2 0xfb 0x0 0x8 0x8 0x80 0x3c 0xe 0xae
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x18 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x17 0x7b 0xb0 0xc4 0x85 0x5e 0x3 0xaa 0x5e 0x4f 0x3d 0xfc 0xcd 0xc 0xa1 0x0 0x52 0xfd 0x0 0x7b 0x3a 0x49 0xf5 0x0 0x87 0x25 0x1b 0x0 0x3 0x5e 0x5b 0x0 0xf1 0x7 0x80 0x3c 0x12 0xc7
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x1a 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x19 0x7b 0xb0 0x4 0xa6 0x2f 0x1 0x98 0xe5 0x26 0xc9 0x2 0xa2 0xc 0xa1 0x0 0x3a 0xfd 0x0 0x7b 0x32 0x7 0x21 0x0 0x53 0x36 0x9 0x0 0x20 0x93 0xf4 0x0 0xe7 0x10 0x0 0x3c 0xf8 0xb6
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x1c 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x1b 0x7b 0xb0 0x4 0x99 0xb3 0x3 0xcd 0xcc 0x96 0xd3 0x16 0xfd 0xc 0xa1 0x0 0x4f 0xfd 0x0 0x7b 0x15 0x35 0x26 0x0 0x8d 0xc2 0xc6 0x0 0xac 0x59 0xc7 0x0 0xea 0xfa 0x3f 0x3c 0x79 0x7b
0xb5 0x62 0x13 0x80 0x34 0x0 0x2 0x0 0x0 0x1e 0x14 0x6 0x13 0x13 0x38 0x1c 0x0 0x0 0xe0 0x4a 0x1 0x3 0x1d 0x7b 0xb0 0x44 0x28 0xf2 0x1 0x67 0x2b 0x26 0x9a 0xfd 0xb6 0xc 0xa1 0x0 0x52 0xfd 0x0 0x7b 0xe2 0x25 0xf9 0x0 0xd8 0x96 0x8b 0x0 0x14 0xd3 0x31 0x0 0x9 0xf7 0xbf 0x3c 0xd8 0x4a
```

I tried to read the new code... it looks quite clear to me but wasn't able to spot bug directly.

Any ideas, @gfwilliams ?

ikkala, 2020-06-04:
The JS code:

```
Bangle.setGPSPower(1);

function writeGPScmd(cmd) {
  var d = [0xB5,0x62]; // sync chars
  d = d.concat(cmd);
  var a=0,b=0;
  for (var i=2;i<d.length;i++) {
    a += d[i];
    b += a;
  }
  d.push(a,b);
  Serial1.write(d);
}

Bangle.on('GPS-raw',function (d) {

  if (d[0]=="$") return;

  const arr = [];
  for (const c of E.toUint8Array(d)) {
    arr.push(c);
  }
  print(arr.map(x => "0x" + x.toString(16)).join(" "));
});

writeGPScmd([0x13, 0x80, 0, 0]); // UBX-MGA-DBD Poll the navigation database

```

ikkala, 2020-06-04:
The result is same with multiple attempts -- so most likely the NMEA messages (that are excluded from the output) are not messing with the buffer.. if they would, then the output would be perhaps more random.

gfwilliams, 2020-06-05:
Do you think you could modify it to output just the first few bytes on each line and see if it still happens?

What I found when debugging was that when you output the decoded data in full, you're outputting on average 4 bytes for each character received. The speed of Bluetooth LE comms is basically only around 9600 baud anyway (the same speed the data comes in from GPS), so when you output big chunks of data, buffers get full and data gets dropped.

I think that is what is happening in this case.

Normally stuff like this would trigger the FIFO_FULL error flag, but since we're not going through the FIFO that isn't happening. Maybe if we checked the state of the GPS_DATA_LINE bit before setting it we could report the problem correctly.

ikkala, 2020-06-05:
I did some testing, and looks like the BLE connection is not the root of the problem:
- First I tried do what you proposed, I only took 4 character and printed them, like this:
```
Bangle.on('GPS-raw',function (d) {

  if (d[0]=="$") return;

  const arr = [];
  let i = 0;
  for (const c of E.toUint8Array(d)) {
    arr.push(c);
    i += 1;
    if (i === 4) break;
  }
  const str = arr.map(x => "0x" + x.toString(16)).join(" ");
  print(str);

});
```
- It worked: Boring but unbroken list of `0xb5 0x62 0x13 0x80` came out
- But then I thought that hey, if I don't print that stuff at all but I put that into global helper array and print it later very, very slowly by using setInterval with 1 second interval so that I don't overload the BLE buffer:
```
const printArr = [];

Bangle.on('GPS-raw',function (d) {

  if (d[0]=="$") return;

  const arr = [];
  for (const c of E.toUint8Array(d)) {
    arr.push(c);
  }
  printArr.push(arr.map(x => "0x" + x.toString(16)).join(" "));

});

setTimeout(() => {
  setInterval(() => {
    const str = printArr.shift();
    if (str) {
      print(str);
    }
  }, 1000);
}, 5000);
```
- Well, that didn't work: Strings were printed one by one, one per second, but some of them were **not** starting by `0xb5 0x62 0x13 0x80` -- like in the original report of this issue
- But then, if I offload the "heavy" work of string processing away from the GPS-raw event handler I can stream the result as fast as I want through BLE, this works like a charm:
```
const printArr = [];

Bangle.on('GPS-raw',function (d) {
  if (d[0]=="$") return;
  printArr.push(d);
});

setTimeout(() => {
  printArr.forEach(d => {
    const arr = [];
    for (const c of E.toUint8Array(d)) {
      arr.push(c);
    }
    let p = arr.map(x => "0x" + x.toString(16)).join(" ");
    print(p);
  });
}, 5000);
```

ikkala, 2020-06-05:
So, is just about warning people doing too heavy things in that GPS-raw event handler or do you have some fix for this?  Does the problem exist in case of the other event handlers as well?

gfwilliams, 2020-06-08:
Thanks - so yeah, it looks like it's what i said above - we set the FIFO_FULL flag and then the error gets reported to the user, same as any other case where the event loop is too busy to receive the data. Just done this now.

commit dfc70eb2 (2020-06-08): Bangle.js: Now warn if GPS data overflows (fix #1847)
closed 2020-06-08 by gfwilliams