espruino/Espruino #1853: Regex issue

matjaz, 2020-06-06:
I've tried formatting number using regex.

    '1234567.89'.replace(/\B(?=(\d{3})+(?!\d))/g, ",")

which does not work.

gfwilliams, 2020-06-08:
Thanks, but the Espruino regex docs do clearly say it doesn't contain the full JS regex spec.

PRs are welcome for extra functionality (or just erroring on unsupported stuff), but this is a severely restricted embedded device and the RegExp implementation is meant to handle 95% of what people normally use regex for.

I think for stuff like the thousands separator it's going to be better to just write it in JS code.

matjaz, 2020-06-08:
yeah. for Bangle it's done here https://github.com/matjaz/BangleApps/commit/fd4e0ceea4f07204a8ba934cc6ccf7a0f9aa99e7#diff-369f213dc47e483871668d8b63615915R184

gfwilliams, 2020-10-29:
This RegExp seems to use the following that aren't supported yet:

* `x{3}` numeric quantifier
* `\B` Assertion Matches a non-word boundary
* `x(?=y)` Lookahead assertion:
* `x(?!y)` Negative lookahead assertion

https://developer.mozilla.org/en-US/docs/Web/JavaScript/Guide/Regular_Expressions/Assertions

Honestly these seem like they're too painful to support in Espruino with maybe the exception of numeric quantifiers.

Rather than having one bug for each of these I'm just going to close this and roll everything into the first of the 3 RegExp issues here

gfwilliams, 2020-10-29:
rolled into https://github.com/espruino/Espruino/issues/1257

closed 2020-10-29 by gfwilliams