espruino/Espruino #1854: Pretokenization issue

nebbishhacker, 2020-06-07:
Pretokenization seems to be mixing up symbols or something.

Steps to reproduce:
- Open the web IDE
- Turn on "Pretokenize code before upload"
- Upload the following code to a bangle.js device:
```
function foo() {
  return 42;
}

print(foo());
```

Expected result:
```
> 42
```

Actual result:
```
>Uncaught SyntaxError: BREAK statement outside of SWITCH, FOR or WHILE loop
 at line 1 col 1
break 42;
     ^
in function "foo" called from line 3 col 11
print(foo());
          ^
```

My bangle.js has firmware version 2v05.155. If I use the emulator instead of an actual device I get the expected result.

nebbishhacker, 2020-06-07:
Here's another test case that produces an error when packaged as an app and loaded using the app loader with pretokenization enabled, but *not* when loaded using the web IDE.

```
function rotatePoints() {}

function drawSnow() {
  var points;
  for(let i = 0; i<=6; ++i) {
    for(let j = -1; j<=1; j += 2) {
      rotatePoints();
    }
  }
}

drawSnow();
```

The error produced:

```
Uncaught ReferenceError: "ii" is not defined
 at line 1 col 58
let points;for(let i=0;i<=6;++i){for(let j=-1;j<=1;j+=2){rotatePoints();}}
                                                                          ^
in function "drawSnow" called from line 3 col 10
drawSnow();
         ^
```

MaBecker, 2020-06-07:
Hmm, just tried the snippets in the Emulator without error. 

Used this settings for the Emulator

[image]

nebbishhacker, 2020-06-07:
Yes, the emulator seems to work fine.

I only get an error from the first snippet when using an actual bangle.js and version 0.72.1 of the web IDE. Version 0.73.7 doesn't give me the error, so I'm guessing it was already fixed by the time I noticed.

I only get an error from the second snippet when I add it to the app loader as an app, and load it onto my watch that way.

I added an app containing the second snippet to https://nebbishhacker.github.io/BangleApps/, so you should be able to reproduce using the following steps:
- go to https://nebbishhacker.github.io/BangleApps/
- ensure pretokenization is enabled
- install the app named "Pre-Tokenization Test"
- connect to the watch using the web IDE
- open the "Pre-Tokenization Test" app on the watch
- an error should appear in the web IDE

gfwilliams, 2020-06-08:
I've just updated the chrome web app as that was horribly out of date. I've been neglecting it because Google keep threatening to kill it any day now.

So I think your first issue isn't a problem any more.

But yeah, your second issue is odd, and I can reproduce here.

It's not pretokenisation per se, because if I load the same code into RAM with `drawSnow=eval("("+drawSnow.toString()+")")` then it's totally fine, despite having the exact same contents:

```
Flash
#99[r1,l0] FlashString [1 blocks] "\xACpoints;\xA7(\xADi=0;i\x8E6;\x98i){\xA7(\xADj=-1;j\x8E1;j\x962){rotatePoints();}}"
RAM
#186[r1,l0] FlatString [5 blocks] "\xACpoints;\xA7(\xADi=0;i\x8E6;\x98i){\xA7(\xADj=-1;j\x8E1;j\x962){rotatePoints();}}"
```

And you can reproduce it on linux builds too in which case you get a `ASSERT(jsvGetLocks(var)>0) FAILED`:

```
require('Storage').write("tokentest.app.js",atob("qnJvdGF0ZVBvaW50cygpe30KqmRyYXdTbm93KCl7rHBvaW50czunKK1pPTA7aY42O5hpKXunKK1qPS0xO2qOMTtqljIpe3JvdGF0ZVBvaW50cygpO319fQpkcmF3U25vdygpOw=="))
load("tokentest.app.js")
```

And I imagine on emulator, if I'd implemented flash storage :)

So definitely something that needs fixing quick - I'm amazed this didn't come up before actually.

So

commit e206a837 (2020-06-09): Bangle.js: Fix occasional execution errors when executing from Flash (fix #1854)
closed 2020-06-09 by gfwilliams