espruino/Espruino #1864: How to handle callbacks in firmware?

MaBecker, 2020-06-15:
There are different ways how call back in wifi and ethernet libs are handled

my understanding is that a callback function should have at least to parameter 

* error 
* information
* ....

Is this the style how a call back should be implemente.
Not sure about
* why is JsVar *wlanObj used in WIZnet implementation but then NOT_USED(wlanObj)?
* is jsvIsFunction(callback) enough to detect if a callback is given 
* use jsvUnLock(params) or  jsvUnLock(params[0])?
* should a return always return a value or only if there is no callback?

```
// in firmaware c code
JsVar * jswrap_ethernet_getHostname(JsVar *wlanObj, JsVar *callback) {
  NOT_USED(wlanObj);
  const char *errorMessage = 0;
  char *hostname = getHostname();

  // Schedule callback if a function was provided
  if (jsvIsFunction(callback)) {
    JsVar *params[2];
    params[0] = errorMessage ? jsvNewFromString(errorMessage) : jsvNewWithFlags(JSV_NULL);
    params[1] = jsvNewFromString(hostname);
    jsiQueueEvents(NULL, callback, params, 2);
    jsvUnLock(params);
    return 0;
  }
  else { 
    return jsvNewFromString(hostname);
  }
}
```

MaBecker, 2020-06-22:
I could need some advise.

gfwilliams, 2020-06-22:
> my understanding is that a callback function should have at least to parameter

Just one parameter is ok as well. Sometimes there will be nothing to return - just a callback indicating that the function was complete (eg DHCP?)

Especially for Network I think we should try and keep the form of first arg = error though, even if it isn't that nice

> why is JsVar *wlanObj used in WIZnet implementation but then NOT_USED(wlanObj)?

That's just to stop the compiler complaining. The first arg of a method is always 'this' - but we never use that because there can only be one wiznet

> is jsvIsFunction(callback) enough to detect if a callback is given

Yes, that's be fine

> use jsvUnLock(params) or jsvUnLock(params[0])?

`jsvUnLock(params[0])` - there's also jsvUnLockMany if you're giving lots of parameters.

`jsvUnLock(params)` will likely produce a compile warning anyway

> should a return always return a value or only if there is no callback?

You might as well always return a value if it is available.

MaBecker, 2020-06-24:
Does   `JsVar *params[2];` need a single ore double unlock? 

  ```
// Schedule callback if a function was provided
  if (jsvIsFunction(callback)) {
    JsVar *params[2];
    params[0] = jsvNewWithFlags(JSV_NULL);
    params[1] = jsStatus;
    jsiQueueEvents(NULL, callback, params, 2);
    jsvUnLock2(params[1],params[0]);
}
```

use

        jsvUnLock2(params[1],params[0]);

or 

            jsvUnLock(params[0]);

gfwilliams, 2020-06-25:
It depends on what values you're passing. What is jsStatus?

If you created it new  - eg `jsvNewWithFlags` as you did for `params[0]` - then you need to unlock it.

But if it's a variable that belongs to something else or is freed elsewhere you don't want to unlock it.

MaBecker, 2020-06-25:
This is the full get_Status() function, does this more jsvUnLock() calls?

```
JsVar * jswrap_ethernet_getStatus( JsVar *wlanObj, JsVar *callback) {
  NOT_USED(wlanObj);

  uint8_t tmp;
  uint8_t tmpstr[6] = {0,};

  JsNetwork net;
  if (!networkGetFromVar(&net)) return 0;

  JsVar *jsStatus = jsvNewObject();

  jsvObjectSetChildAndUnLock(jsStatus, "state", jsvNewFromString(state[networkState]));

  ctlwizchip(CW_GET_ID,(void*)tmpstr);  
  jsvObjectSetChildAndUnLock(jsStatus, "id",jsvNewFromString(tmpstr));

  ctlwizchip(CW_GET_PHYLINK, (void*)&tmp);
  jsvObjectSetChildAndUnLock(jsStatus, "phylink", jsvNewFromString(phylink[tmp]));
  
  ctlwizchip(CW_GET_PHYPOWMODE,(void*)&tmp);
  jsvObjectSetChildAndUnLock(jsStatus, "phypowmode", jsvNewFromString(phypowm[tmp]));

#if  _WIZCHIP == 5500 
  wiz_PhyConf tmpPhycConf;
  ctlwizchip(CW_GET_PHYCONF,(void*)&tmpPhycConf);
  jsvObjectSetChildAndUnLock(jsStatus, "by", jsvNewFromInteger(tmpPhycConf.by));
  jsvObjectSetChildAndUnLock(jsStatus, "mode", jsvNewFromInteger(tmpPhycConf.mode));
  jsvObjectSetChildAndUnLock(jsStatus, "speed",  tmpPhycConf.speed ? jsvNewFromInteger(10) :  jsvNewFromInteger(100));
  jsvObjectSetChildAndUnLock(jsStatus, "duplex", jsvNewFromInteger(tmpPhycConf.duplex));
#endif

  networkFree(&net);

  // Schedule callback if a function was provided
  if (jsvIsFunction(callback)) {
    JsVar *params[2];
    params[0] = jsvNewWithFlags(JSV_NULL);
    params[1] = jsStatus;
    jsiQueueEvents(NULL, callback, params, 2);
    jsvUnLock(params[0]);
  } 
  return jsStatus;
}

```

gfwilliams, 2020-06-25:
No, that looks perfect :)

MaBecker, 2020-06-26:
This one does not return the errorMessage and give me an error

Any hint to to fix it?

```
>eth.setHostname("abcd");
=
Uncaught InternalError: Variable type cannot be converted to string
```

funtion:

```
bool jswrap_ethernet_setHostname(JsVar *wlanObj, JsVar *jsHostname, JsVar *callback){
  NOT_USED(wlanObj);
  char hostname[13];
  const char *errorMessage = 0;

  jsvGetString(jsHostname, hostname, sizeof(hostname));
  jsvUnLock(jsHostname);

  if (strlen(hostname) < 8) {
    errorMessage = "hostname too short (min 8 and max 12 char )";
  }
  else {
    setHostname(hostname);
  }

  // Schedule callback if a function was provided
  if (jsvIsFunction(callback)) {
    JsVar *params[1];
    params[0] = errorMessage ? jsvNewFromString(errorMessage) : jsvNewWithFlags(JSV_NULL);
    jsiQueueEvents(NULL, callback, params, 1);
    jsvUnLock(params[0]);
  }
  return errorMessage==0;
}
```

gfwilliams, 2020-06-26:
What's the JSON header for the function? that could have something to do with it...

MaBecker, 2020-06-26:
/*JSON{
  "type" : "method",
  "class" : "Ethernet",
  "ifdef" : "USE_WIZNET",
  "name" : "setHostname",
  "generate" : "jswrap_ethernet_setHostname",
  "params" : [
    ["options","JsVar","An optional `callback(err)` function to be called back with null or error text."]
  ],
  "return" : ["bool","True on success"]
}
Set hostname allow to set the hosname used during the dhcp request.
min 8 and max 12 char, best set before calling `eth.setIP()`
Default is WIZnet010203, 010203 is the default nic as part of the mac.
Best to set the hosname before calling setIP().
*/

MaBecker, 2020-06-26:
OK, I see whats wrong - so check the header helps :)

MaBecker, 2020-06-26:
Using a call back works.

```
>eth.setHostname("abcd",(e) => {console.log(e);});
=false
hostname too short (min 8 and max 12 char )
```

Can't get it running without callback  `eth.setHostname("abcd");` causes this error - Any further hints?

```
>eth.setHostname("abcd");
=
Uncaught InternalError: Variable type cannot be converted to string
```

used function code:

```
/*JSON{
  "type" : "method",
  "class" : "Ethernet",
  "ifdef" : "USE_WIZNET",
  "name" : "setHostname",
  "generate" : "jswrap_ethernet_setHostname",
  "params" : [
    ["hostname","JsVar","hostname as string"],
    ["callback","JsVar","An optional `callback(err)` function to be called back with null or error text."]
  ],
  "return" : ["bool","True on success"]
}
Set hostname allow to set the hosname used during the dhcp request.
min 8 and max 12 char, best set before calling `eth.setIP()`
Default is WIZnet010203, 010203 is the default nic as part of the mac.
Best to set the hosname before calling setIP().
*/
bool jswrap_ethernet_setHostname(JsVar *wlanObj, JsVar *jsHostname, JsVar *callback){
  NOT_USED(wlanObj);
  char hostname[13];
  const char *errorMessage = 0;

  jsvGetString(jsHostname, hostname, sizeof(hostname));
  jsvUnLock(jsHostname);

  if (strlen(hostname) < 8) {
    errorMessage = "hostname too short (min 8 and max 12 char )";

  }
  else {
    setHostname(hostname);
  }

  // Schedule callback if a function was provided
  if (jsvIsFunction(callback)) {
    JsVar *params[1];
    params[0] = errorMessage ? jsvNewFromString(errorMessage) : jsvNewWithFlags(JSV_NULL);
    jsiQueueEvents(NULL, callback, params, 1);
    jsvUnLock(params[0]);
  }
  return errorMessage==0;
}
```

gfwilliams, 2020-06-29:
Ahh - yeah, you're doing `jsvUnLock(jsHostname);` and it's not needed.

Basically:

* UnLock 'new' JsVars that are created from function calls when you're done with them
* If you're passing vars out as a function return value, *don't* unlock them
* If you've been given vars in a function argument, *don't* unlock them

If you're able to compile without `RELEASE=1` (eg by removing some libraries you're not using for debugging) then there are assertions that will catch these problems for you

MaBecker, 2020-07-01:
Now all callbacks are working as designed

closed 2020-07-01 by MaBecker