espruino/Espruino #1865: Math.floor returns values equal to null

mfikes, 2020-06-21:
In 2v06:

```
>Math.floor(0.5) == null
=true
>Math.floor(1.5) == null
=false
```

Both results should be `false`

MaBecker, 2020-06-21:
Not sure about this, check the examples on this MDN page:

https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Equality

mfikes, 2020-06-21:
@MaBecker On that page, under "Comparison with type conversion" it has this example:

```
0 == null;            // false
```

Did you have something else in mind?

MaBecker, 2020-06-21:
Ups, still on 2v05

mfikes, 2020-06-21:
FWIW, the issue was originally noticed on 2v04... but I checked to see that it is still reproducible on 2v06.

gfwilliams, 2020-06-22:
Thanks - just checked into this and:

```
>trace(Math.floor(0)) 
#33[r0,l1] Double 0 
>trace(0)
#31[r0,l1] Integer 0 

>0==null
false
>0.0==null
true
```

So the issue appears to be with comparison of floating point numbers with null, rather than Math.floor itself

commit 3b74efba (2020-06-29): Fix 0.0==null comparison (fix #1865)
closed 2020-06-29 by gfwilliams