espruino/Espruino #1869: [Documentation][Code Enhancement] [BangleJs]Improve HRM Handler

ayusun, 2020-06-27:
Hi I was going through the code of hrmHandler because for some reason it shows 200 alot. Sometimes it even shows values like 120, 157(etc) even when I am on resting or lying down. While I wa sgoing through the code, I got few questions in my head on what initial approach.

1) The value read from sensor moves from -128 to 127. Does the value follow this formula (https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fsaadc.html&cp=2_1_0_36_2&anchor=saadc_digital_output)

2) What is CMin and Cmax.  How did we come up with the formula for them (There are few magic numbers that are being used, So I can make best guess for them)
3) Read somewhere that maximum value of BPM is `220`
4) Generally, sensor data have noise and we tend to pass them through filter. Wondering if that's already been taken care of
5) In some cases `nrf_analog_read()` returns `-16384`. Should we add a `if` condition for this case? https://github.com/espruino/Espruino/blob/04a50800c73b7ede33cc89a1679339284b77e36d/targets/nrf5x/jshardware.c#L998
6) Is there a way to enable debugging at runtime? for this module

gfwilliams, 2020-06-29:
Hi,

> The value read from sensor moves from -128 to 127. Does the value follow this formula 

The value is linearly related to the voltage that comes in - but since the voltage comes from an analog Op-Amp the actual range of values itself is meaningless. You're just looking for a pattern.

> What is CMin and Cmax. How did we come up with the formula for them (There are few magic numbers that are being used, So I can make best guess for them)

They are the range or array indices over which the autocorrelation filter correlates: https://en.wikipedia.org/wiki/Autocorrelation#:~:text=Autocorrelation%2C%20also%20known%20as%20serial,the%20time%20lag%20between%20them.

It's not worth doing the correlation for BPM values that are not possible.

> Read somewhere that maximum value of BPM is 220

In humans? or Bangle.js itself. You could always bump BPM_MAX to 220.

> Generally, sensor data have noise and we tend to pass them through filter. Wondering if that's already been taken care of

Yes, that's what the code with CMIN/CMAX is for. It's an autocorrelation filter (see the link above).

IMO the most useful thing would be to make `confidence` produce a more realistic confidence value. Some times it can say 100 when there's no useful BPM signal at all.

> In some cases nrf_analog_read() returns -16384. Should we add a if condition for this case?

Have you ever seen this happening? This is handled in https://github.com/espruino/Espruino/blob/master/libs/banglejs/jswrap_bangle.c#L666

> Is there a way to enable debugging at runtime? for this module

Yep - just comment out  https://github.com/espruino/Espruino/blob/master/libs/banglejs/jswrap_bangle.c#L1769

The `correlation` array will be the most useful

gfwilliams, 2020-06-29:
Just to add, for fast iteration I'd just translate the code under `if (bangle && (bangleTasks & JSBT_HRM_DATA)) {` into JS and then have it run using the `hrmHistory` array you get in the HRM event. It'll be much easier to tweak and try things out

closed 2020-07-13 by gfwilliams