espruino/Espruino #1877: Why not add jsDebug as jsWarn when RELEASE  is not set

MaBecker, 2020-07-12:
This implementation in a central point allows to use jsDebug during development and if RELEASE it is gone.

https://github.com/espruino/Espruino/blob/f99ccaac3ec411090819c6a35cd9aea4fc131929/libs/network/esp32/jswrap_esp32_network.c#L45-L49

* So no more #ifndef RELASE ..... #endif to get ride of warnings.

* It is still possible to use jsWarn() if wanted.

Would /src/jsutils.c be the correct place to add this?

MaBecker, 2020-07-13:
Or call it  `jsDevInfo()` because this is really what it is.

gfwilliams, 2020-07-13:
That sounds like a great idea - maybe stick it up in `jsutils.h` near `jsAssertFail` - and I'd use `jsiConsolePrintf` (unless jsWarn is there *just* for ESP8266 memory issues).

What about something like:

```
#if defined(DEBUG) || __FILE__==DEBUG_FILE
   #define jsDebug(format, ...) jsiConsolePrintf("[" __FILE__ "]:" format, ## __VA_ARGS__) 
 #else 
   #define jsDebug(format, ...) do { } while(0) 
 #endif 
```

Untested but the idea is:

* Debug statements only in there if you compile with DEBUG=1 (which you'd do for debugging)
* OR if you do `DEBUG_FILE="src/jsflash.c" make` then you can get a build with *just* the debug prints in that file.

The big chunk of work is really just changing all the prints all over the code to `jsDebug` though :)

One final  suggestion: Is it worth actually having a first argument:

`jsDebug(DBG_INFO, format, ...)` and  `jsDebug(DBG_VERBOSE, format, ...)`  

Even if we ignore it for now, just so that we don't spam the console unless it's absolutely required.

MaBecker, 2020-07-13:
Cool, so let's give it a try.

MaBecker, 2020-07-13:
implemented with https://github.com/espruino/Espruino/pull/1880

closed 2020-07-14 by MaBecker