espruino/Espruino #1892: Possible command line options to adopt the firmware to your need without changing the board definitions?

MaBecker, 2020-07-20:
So during using Espruino I know it is  possible to change firmware  content via

* to add modules/classes with USE_<Module/Class>

*  DEFINES="-D<makefile flags or values>"

**How to use SETDEFINES?**

https://github.com/espruino/Espruino/blob/fdc1b81cbca57f928bd361f53487530702e64134/Makefile#L40-L41

https://github.com/espruino/Espruino/blob/fdc1b81cbca57f928bd361f53487530702e64134/Makefile#L169-L172

Create a specific file with options like to this

echo USE_FILESYSTEM=0 >> ../myESP32.settings

Just tried it, so ok, got this pars covered now as well.

**What about other board setting?**

e.g variables

Maybe @jumjum123 can bring in some light,  because he covered all this possibilities in  [EFEU](http://forum.espruino.com/conversations/334202/#comment15358607)

gfwilliams, 2020-07-21:
Yep - pretty sure this is something @jumjum123 added for EFEU.

At some point though I think it's more sensible just to make your own BOARD.py file - the whole thing exists so you can customise the build properly - it seems a bit over the top to have two ways of modifying the build.

fanoush, 2020-07-21:
also there is now a way to supply board name in board file so you can have multiple board files for same board. I have several board files for same device with different build options producing same name in process.env.BOARD. see https://github.com/espruino/Espruino/pull/1817

MaBecker, 2020-07-23:
> At some point though I think it's more sensible just to make your own BOARD.py file

Yes this sounds like the best approach.

BOARD="Path plus Board Name" would be nice too.

Let's just wait a few more days to give @jumjum123 some time to add his comments before close this question.

jumjum123, 2020-07-25:
There are several additional args to Makefile born during EFEU.
Idea behind was to have an easy to handle option to customize build without having to change anything in Espruino tree

Options which came in during this process, are described in Makefile lines 38 to 50
If somebody needs somemore explanation, please drop me a line.
Right now, only a handful people ever used these extensions.
Nevertheless, still believe, they are helpful ;-)

MaBecker, 2020-07-27:
Just added #1898 to document custom board files.

closed 2020-07-27 by MaBecker