espruino/Espruino #1896: What about adding something  to .gitignore to allow custom board files?

MaBecker, 2020-07-25:
For my project I have different board files with different settings, it would be nice to permanently ignore them.

A simple way to ignore them is to add a prefix or a postfix to the board file name and create a new entry in .gitignore.

I would go for a postfix  like **_CUST**  and an entry like this **/board/*_CUST.py** for .gitignore.

closed 2020-07-27 by gfwilliams

MaBecker, 2020-07-27:
Great - Thanks!

gfwilliams, 2020-07-27:
Just added for `_CUSTOM.py` - but IMO this sort of thing really isn't worth doing in general.

Mainly because without documentation it's effectively useless - I seriously doubt anyone except you will end up using this :)

MaBecker, 2020-07-27:
If you like I can  document this in  e.g. https://github.com/espruino/Espruino/blob/master/README_BuildProcess.md or where ever you suggest :)

gfwilliams, 2020-07-27:
That'd be great actually - in https://github.com/espruino/Espruino/blob/master/README_BuildProcess.md#board-definition-file-- ?

Mentioning the custom board name addition would be really neat as well.

commit 8cc1df74 (2020-07-27): fix #1896