espruino/Espruino #1899: Feature: support both internal and external (SPI) flash in Storage module

fanoush, 2020-07-29:
It would be nice to have both internal flash for code and also external SPI flash primarily for data supported by Storage module. Currently only one way (memory area, size) can be selected in board file.

 Any ideas how this could be done with current Storage API or other modules? Was thinking file paths like "ext/filename" or maybe drives "int:filename", "ext:filename" with one of them being default and optional. Loading two storage modules  would be a bit strange(?). Or could module accept parameters like  require("Storage")("SPI") to pick the second one like this?

This could be usable for NRF52832 devices with SPI flash which only need SPI for data files but still have enough space for code in internal flash. Idea being that internal flash is faster and directly mapped so does not need extra RAM copy of data.

Would you merge stuff making Storage module more complicated like this? Or maybe some other full blown filesystem for SPI would be better than modifying Storage code?

gfwilliams, 2020-07-29:
Yes, this could be a good idea. Even on Bangle.js certain things (like the bootloader?) would probably benefit.

Internally I don't think it's a huge problem to have >1 set of start/end addresses for the functions (maybe easiest to just have 2 global variables that are constants if the feature isn't available).

In terms of choosing where files are, IMO `"ext/filename"` would be a mistake (unless you could then read the file with just `"filename"`) as it doesn't give you the option of moving pre-existing files into internal flash to speed things up.

Maybe just an extra options argument to `.write`? Or a `Storage.setWriteLocation(...)`?

fanoush, 2020-07-29:
Oh, you mean the namespace would be shared and you would decide only when **creating** the file where it is located but then it would be unified view, so "file1" and "file2" could be each in different area. And when opening file you would search both areas. Well yes that could be used too and maybe would look or be simple to use (at first). Somehow I feel this could also produce complicated hard to understand cases - like file created in wrong place and filling it while there is still free space in the other. Or even file spanning both areas (as a bug or feature) behaving differently when different parts are read (also garbage collection or deleting files could be interesting in such case).

gfwilliams, 2025-04-15:
This has actually been done for a while I think? Still appear to maybe be some issues enabling it for Bangle.js 2 though

closed 2025-04-15 by gfwilliams