espruino/Espruino #1901: require("net").createServer() missing event "data" for Espruino WIFI

MaBecker, 2020-08-03:
This sample is behaving different http://www.espruino.com/Internet#sockets

Command used for testing

OSX:  `echo '{"i":'$i'}'| nc -w 1 192.168.194.197  8124; let i++; echo`
Linux: `echo '{"i":'$i'}'| nc -l 1 192.168.194.197  8124; let i++; echo`

OK on 

* Espruino for Linux
* MDBQ42T with W550io
* PICO with W550io

Missing calls on

* Espruino Wifi

used snippet:
```
var socketServer = {
    init: () => {
        server = require("net").createServer(function(c) {
            c.on('data', function(data) {
                c.write(JSON.stringify(data));
                c.end();
                console.log("data:",data);
            });
        });
        server.listen(myPort);
        console.log("I: Server is up and waiting for connections");
    }
};

// setup network  connection
// Espruino WIFI
wifi = require("Wifi");
wifi.stopAP();
wifi.connect(WIFI_AP,{"password":WIFI_PWD},()=> {
  console.log("I: connected to AP");
  wifi.getIP((err,ip) => {console.log(err,ip);});
   // wifi.turbo(true);
   socketServer.init();
});

// start socket server
setTimeout(socketServer.init,1E3);
```

wifi.debug() show received data
```
>wifi.debug()
={ wifiMode: 1, connected: 1,
  socks: [ undefined, undefined, undefined, undefined, undefined, true ],
  sockData: [
    "{\"i\":30}\n{\"i\":31}\n",
    "",
    "",
    "",
    "",
    ""
   ]
 }
> 
```

Not sure if this a coding, a Espressif AT or a WiFi module issue.

MaBecker, 2020-08-03:
Just ran some more tests, other Espruino devices loose them too.

gfwilliams, 2020-08-03:
Thanks - I'll take a look. And you never get the data event?

Which other devices don't work? Is it just the ones that use ESP8266 in AT command mode?

MaBecker, 2020-08-03:
MDBT42Q and PICO with W550io

MaBecker, 2020-08-03:
> And you never get the data event?

yes but only a few times 

using this bash script to run up to 100 calls

```
export IP="192.168.194.20"
export PORT="8124"

let i=0
while [ $i -lt 100 ]
do
    echo '{"i":'$i'}'| nc -w 1 $IP $PORT
    echo
    let i++
    sleep 2
done
```

gfwilliams, 2020-08-03:
> MDBT42Q and PICO with W550io

I thought you said those ones worked in the initial bug post?

> yes but only a few times

Like only a few times out of 100? Or 50%?

MaBecker, 2020-08-03:
Never lost a single call when using Espruino for Linux

MaBecker, 2020-08-03:
> I thought you said those ones worked in the initial bug post?

Yes,  just shot a few calls, but to be sure have to come up with a test and hoped to get 100 of 100.

> Like only a few times out of 100? Or 50%?

sorry it is very random

* best case 50% of 100 worked
* worst case 20% of 100 worked

gfwilliams, 2020-08-03:
> and hoped to get 100 of 100.

Ok, so just to be clear, those do not work reliably either? Just more reliably than the Espruino WiFi?

MaBecker, 2020-08-03:
> Ok, so just to be clear, those do not work reliably either?

Yes.

> Just more reliably than the Espruino WiFi?

Hard so say because it turned out to be  so random.

gfwilliams, 2020-08-03:
Do you not get errors when you run your code? I do when I run it because of the two calls to `socketServer.init();`... 

But after that, just started an Espruino WiFi - I've run it a few times and I get ~98/100 succeeded. The ones that didn't display a partial bit of a message and then this on the Espruino:

```
Uncaught Error: This socket is closed.
 at line 9 col 29
c.write(JSON.stringify(data));
                            ^
```

Now I guess it's possible that there's some issue there (data only received when the socket closes?) but that's a long way off your under 50% - and it could just be a side-effect of only waiting 1s for a response when my WiFi router is on the opposite side of the house :)

Please could you try:

* Remove `wifi.stopAP` - it's meant to be used async, so that could maybe cause you issues when you try and use it in a sync way
* Remove `setTimeout(socketServer.init,1E3);` which will be trying to create a second socket server on the same port.

MaBecker, 2020-08-03:
> Do you not get errors when you run your code? I do when I run it because of the two calls to socketServer.init();...

Yes I do,  now switched it to `-w  2` to keep to connection open up to 2sec.

Still get those messages:

```
Uncaught Error: This socket is closed.
 at line 9 col 29
c.write(JSON.stringify(data));
                            ^
```

New test code:

```
var WIFI_AP = '', WIFI_PWD = '';

var myPort = 8124;

var socketServer = {
    init: () => {
        server = require("net").createServer(function(c) {
            c.on('data', function(data) {
                c.write(JSON.stringify(data));
                c.end();
                console.log("data:",data);
            });
        });
        server.listen(myPort);
        console.log("I: Server is up and waiting for connections");
    }
};

wifi = require("Wifi");
wifi.connect(WIFI_AP,{"password":WIFI_PWD},()=> {
  console.log("I: connected to AP");
  wifi.getIP((err,ip) => {console.log(err,ip);});
  //wifi.turbo(true);
  socketServer.init();
});

```

this is a good example to show that data was received but no `c.on('data'....` was emited

```
data: {"i":3}
data: {"i":4}        
{"i":5}
{"i":6}
data: {"i":7}
data: {"i":8}
> 

data: {"i":9}       <- no emit for call 9 and processed output with call 10
{"i":10}
Uncaught Error: This socket is closed.
 at line 9 col 29
c.write(JSON.stringify(data));
                            ^
in function called from system
data: {"i":12}    <- no emit for call 12 and processed output with call 13
{"i":13}
Uncaught Error: This socket is closed.
 at line 9 col 29
c.write(JSON.stringify(data));
                            ^
in function called from system
data: {"i":15}
data: {"i":16}
data: {"i":17}      <- no emit for call 17 and processed output with call 18
{"i":18}
Uncaught Error: This socket is closed.
 at line 9 col 29
c.write(JSON.stringify(data));
                            ^
in function called from system
data: {"i":20}
> 
```

And  Espruino Wifi calling `c.write(...)` or `e.end(...)` does not return data to the client, LINUX, PICO and MDBT42T do return data.

MaBecker, 2020-08-03:
**Important update: Using a node js client works perfekt for all boards including response**

client code:
```
var ioUnitIP = process.env.IP || "192.168.194.20";
var ioUnitPort = process.env.PORT || 8124;
var cTest = {"c":"on"};
var cmd;
if (process.env.CMD !== undefined)
    cmd = JSON.parse(process.env.CMD);
else
    cmd = cTest;

console.log("> connect to:", ioUnitIP + ":" + ioUnitPort, "and send:", JSON.stringify(cmd));

var net = require('net');

var client = new net.Socket();
client.connect(ioUnitPort, ioUnitIP, function() {
    console.log('> connected');
    client.write(JSON.stringify(cmd));
});

client.on('data', function(data) {
    console.log('> received:', data.length, data.toString(), data);
});

client.on('close', function() {
    console.log('> connection closed');
});
```

run
```
IP=192.168.1.100 PORT=8124 CMD='{"i":1}' node ioUnit_tcp_client.js
```

output:
```
> connect to: 192.168.1.100:8124 and send: {"i":1}
> connected
> received: 11 "{\"i\":1}" <Buffer 22 7b 5c 22 69 5c 22 3a 31 7d 22>
> connection closed
```

Question: Is it expected that nc or netcat should work too?

gfwilliams, 2020-08-04:
Hmm - that's an interesting one. But I guess on the Node.js client you don't disconnect 1 sec after sending?

So it seems to me like Espruino Wifi (and probably other AT-command devices) have a problem where if the client sends data and disconnects soon after, the disconnect can happen before the received data has been processed, and then it's appended to the next connection?

> And Espruino Wifi calling c.write(...) or e.end(...) does not return data to the client, LINUX, PICO and MDBT42Q do return data.

Ok, so what about  LINUX, PICO (WIZnet) and MDBT42Q (WIZnet)? Do these actually work ok?

If not, I think there is more than one issue at play here.

MaBecker, 2020-08-04:
Well `echo "msg" |  nc`  will immediately close the connection, that’s why -w 2 or -l 2 is required. nc is waiting for a server close or until time is up and than close the connection to the sever.

Linux:
* nc and Nodejs client work 100%

WIZnet-devices:
* are not emitting data for all calls
* nc is receiving data  send via c.write()
* nc is not receiving the close send via c.end()
* Nodejs client work 100%

AT-devices:
* are not emitting data for all calls
* are not loosing data, checked with wifi.debug
* nc is not receiving data send via c.write()
* nc is not receiving the close send via c.end()
* Nodejs client work 100%

Hope this helps to narrow down this issue.

MaBecker, 2020-08-04:
**Important update: Using a  socket c code is working perfekt for all boards including response**

```
#include <stdio.h> 
#include <sys/socket.h> 
#include <arpa/inet.h> 
#include <unistd.h> 
#include <string.h> 
#include <stdlib.h>

int main(int argc, char const *argv[]) 
{ 
    int sock = 0, valread; 
    struct sockaddr_in serv_addr; 
    char buffer[1024] = {0}; 

    if(argc != 4) {
        fprintf(stderr,"usage: client <server ip> <port> <data>\n");
        return -1;
    }

    if ((sock = socket(AF_INET, SOCK_STREAM, 0)) < 0) 
    { 
        printf("\n Socket creation error \n"); 
        return -1; 
    } 
    
    // Convert IPv4 and IPv6 addresses from text to binary form 
    if(inet_pton(AF_INET, argv[1], &serv_addr.sin_addr)<=0)  
    { 
        printf("\nInvalid address/ Address not supported \n"); 
        return -1; 
    } 

    serv_addr.sin_family = AF_INET; 
    serv_addr.sin_port = htons(atoi(argv[2])); 
   
    if (connect(sock, (struct sockaddr *)&serv_addr, sizeof(serv_addr)) < 0) 
    { 
        printf("\nConnection Failed \n"); 
        return -1; 
    } 

    send(sock , argv[3] , strlen(argv[3]) , 0 ); 
    valread = read( sock , buffer, 1024); 
    printf("%s\n",buffer ); 

    return 0; 
} 
```

MaBecker, 2020-08-04:
**Lessons learned - do not try to use nc for socket tests with Espruino**

**Better use a nodejs client.js or a c client.c.**

closed 2020-08-04 by MaBecker