espruino/Espruino #1902: setWatch(debounce) can cause multiple of the same event

gfwilliams, 2020-08-07:
Based on: 

http://forum.espruino.com/conversations/352012/#comment15451361

Simply doing a setWatch is enough (nRF52):

```
>setWatch(print,D3,{repeat:true,edge:0,debounce:5});
=1
{ "state": true, "lastTime": undefined, "time": 1596812242.14101791381 }
{ "state": false, "lastTime": 1596812242.14101791381, "time": 1596812242.26583480834 }
{ "state": true, "lastTime": 1596812242.26583480834, "time": 1596812242.27486801147 }
{ "state": false, "lastTime": 1596812242.27486801147, "time": 1596812242.37411117553 }
{ "state": true, "lastTime": 1596812242.37411117553, "time": 1596812242.39818954467 }
{ "state": false, "lastTime": 1596812242.39818954467, "time": 1596812242.60046005249 }
{ "state": true, "lastTime": 1596812242.60046005249, "time": 1596812242.61651229858 }
{ "state": false, "lastTime": 1596812242.61651229858, "time": 1596812242.79046249389 }
{ "state": true, "lastTime": 1596812242.79046249389, "time": 1596812242.85891342163 }
{ "state": false, "lastTime": 1596812242.85891342163, "time": 1596812242.96047592163 }
{ "state": true, "lastTime": 1596812242.96047592163, "time": 1596812242.98742294311 }
{ "state": false, "lastTime": 1596812242.98742294311, "time": 1596812243.62267684936 }
{ "state": true, "lastTime": 1596812243.62267684936, "time": 1596812244.01333236694 }
{ "state": false, "lastTime": 1596812244.01333236694, "time": 1596812244.25713729858 }
{ "state": true, "lastTime": 1596812244.25713729858, "time": 1596812244.26382064819 }
{ "state": false, "lastTime": 1596812244.26382064819, "time": 1596812244.49920272827 }
{ "state": true, "lastTime": 1596812244.49920272827, "time": 1596812244.50875473022 }
{ "state": true, "lastTime": 1596812244.50875473022, "time": 1596812244.70891952514 } <--------
{ "state": true, "lastTime": 1596812244.70891952514, "time": 1596812244.91866683959 } <--------
{ "state": true, "lastTime": 1596812244.91866683959, "time": 1596812245.05624008178 } <--------
{ "state": false, "lastTime": 1596812245.05624008178, "time": 1596812245.12023544311 }
{ "state": true, "lastTime": 1596812245.12023544311, "time": 1596812245.14352035522 }
{ "state": false, "lastTime": 1596812245.14352035522, "time": 1596812245.86642074584 }
{ "state": true, "lastTime": 1596812245.86642074584, "time": 1596812245.91857528686 }
{ "state": false, "lastTime": 1596812245.91857528686, "time": 1596812246.04916000366 }
```

Hard to reproduce with just a button but rubbing some wires together to generate lots of noise will do it.

Note the times are all far enough apart, it's just that there isn't a falling edge in between

enaon, 2020-08-19:
I am not sure if this is the right place, I just checked an image made by @fanoush (2v06.100) and I think the edge reporting is now reverced.  edge=1 fires when button goes from true to false.

fanoush, 2020-08-19:
the build is from commit https://github.com/espruino/Espruino/commit/a4655c383c1af7bc89391c22e832b6fbaf475fcc and the setWatch is on P8 smartwatch button BTN1 which is on normal non-negated pin D17 so there should be no surprises in the setup.

gfwilliams, 2020-08-20:
Thanks - looks like while the edge is reported correctly, the edge triggering code was wrong when debounce was enabled. Should be sorted now

gfwilliams, 2020-09-14:
This is fixed now

rbtrrnss, 2020-10-06:
After updating to 2v07, lastTime behaves differently to 2v06.

The same line (very similar to the one above)
`setWatch(print,BTN,{repeat:true,edge:"falling",debounce:5});`
prints as follows:

in 2v07
```
{ "state": false, "lastTime": undefined, "time": 1602002704.78241443634, "pin": D0 }
{ "state": false, "lastTime": 1602002704.78241443634, "time": 1602002706.12823963165, "pin": D0 }
{ "state": false, "lastTime": 1602002706.12823963165, "time": 1602002707.23483753204, "pin": D0 }
```
which indicates that "lastTime" is the **last falling edge**, e.g. "time between button presses", which makes it impossible to detect the "button down" time, afaik

in 2v06
```
{ "state": false, "lastTime": 1602003013.97906208038, "time": 1602003014.12249469757 }
{ "state": false, "lastTime": 1602003015.60372638702, "time": 1602003015.73724079132 }
{ "state": false, "lastTime": 1602003016.50109577178, "time": 1602003016.65325641632 }
```
which seems to me that "lastTime" is the **last rising edge**, which helped getting a "button down" time (time - lastTime)

rbtrrnss, 2020-10-06:
Hopefully, this is the right place to put this.
After updating to 2v07, `lastTime` behaves differently to 2v06.

The same line (very similar to the one above), here on a Puck.js
`setWatch(print,BTN,{repeat:true,edge:"falling",debounce:5});`
prints as follows:

in 2v07
```
{ "state": false, "lastTime": undefined, "time": 1602002704.78241443634, "pin": D0 }
{ "state": false, "lastTime": 1602002704.78241443634, "time": 1602002706.12823963165, "pin": D0 }
{ "state": false, "lastTime": 1602002706.12823963165, "time": 1602002707.23483753204, "pin": D0 }
```
For the first trigger, there is no `lastTime` defined, although with the `setWatch(edge:"falling")` the rise should be detected nontheless? Also, the timecode for `lastTime` is identical to `time` of the previous.
Which indicates that `lastTime` is the **last falling edge**, e.g. "time between button presses", which makes it impossible to detect the "button down" time, which I'm using to differentiate between long and short presses.

in 2v06
```
{ "state": false, "lastTime": 1602003013.97906208038, "time": 1602003014.12249469757 }
{ "state": false, "lastTime": 1602003015.60372638702, "time": 1602003015.73724079132 }
{ "state": false, "lastTime": 1602003016.50109577178, "time": 1602003016.65325641632 }
```
which seems to me that `lastTime` is the **last rising edge**, which helped getting a "button down" time `len = time - lastTime`

How would it be possible to measure the length of a button press on 2v07?

gfwilliams, 2020-10-07:
Thanks! That's definitely not intentional - I'll look into it.

For now you should be able to have:

```
setWatch(a,BTN,{repeat:true,edge:"rising",debounce:5});
setWatch(b,BTN,{repeat:true,edge:"falling",debounce:5});
```

And measure the time between, but it's not ideal

commit df71b754 (2020-10-07): Fix setWatch debounce lastTime regression from 2v07 (fix #1902)
closed 2020-10-07 by gfwilliams