espruino/Espruino #1908: readJSON/writeJSON/erase make the file broken

vidalouiswang, 2020-08-13:
I saved some json objects with writeJSON function, then when I try to read it, error occurred in the readJSON. It shows me Uncaught Error: Got [ERASED], expected ','. 
I try to read this file with `read ` function and I found the file was broken, the content like this: 
`{"name":"test", "type":"dw"???????????????????????????????????????????????????????????????????`

And the
`require("Storage").list()`
shows a broken file list:
>fs.list()
=[ 
  "server", 
  "wifiInfo", 
  "users", 
  "spname", 
  ".boot0",  ... 
  "&ad\t,id)a",      <==here
  "de.@`hta!",       <==here
  "funInitia", 
  "action", 
  "tasks"
 ]

The function readJSON/writeJSON/erase may lead to the problem cause I only used these three functions when the problem happened.

Espruino version: 2.06 running on NodeMCU.

MaBecker, 2020-08-13:
Honestly, erase_flash, flash  and then try again.

vidalouiswang, 2020-08-14:
> Honestly, erase_flash, flash and then try again.

I tried. But same problem.

MaBecker, 2020-08-14:
Ok, do you like to share some details like board, firmware version and source to reproduce this.
This would definitely help to fix this ;-)

vidalouiswang, 2020-08-14:
> Ok, do you like to share some details like board, firmware version and source to reproduce this.
> This would definitely help to fix this ;-)

Espruino version: 2.06 running on NodeMCU. 
The function readJSON/writeJSON/erase may lead to the problem cause I only used these three functions when the problem happened.

MaBecker, 2020-08-14:
please share sample code to reproduce this issue

vidalouiswang, 2020-08-15:
> please share sample code to reproduce this issue

OK. Here is the code:
`let arr = fs.readJSON("action");
/*a:
[{ 
  "name": "\\u5f00\\u5173", 
  "type": "dw", 
  "pin": "NodeMCU.D0", 
  "state": "a", 
  "id": "2576b617001feda733d2b1b8b10c002b1a49f6a3", 
  "subState": false 
},{
    ...
}]
*/
let ac = null;
for (let i of arr) {
    if (i.id == "xxx") {
        ac = i;
    }
}
if (ac) {
    ac.subState = true;
}
fs.writeJSON("action", arr);`

vidalouiswang, 2020-08-15:
> please share sample code to reproduce this issue

And I did another test.
Change `subState` every second  and write to flash using writeJSON. Here is the result from Serial:
[https://vida.wang/1.jpg](url)

MaBecker, 2020-08-15:
I see no way to help you with this, the sample is producing no errors.

vidalouiswang, 2020-08-16:
> I see no way to help you with this, the sample is producing no errors.

Ok. It is a random error, You can try to use `writeJSON` frequently, like every second or 100 milliseconds. 
Now I have to make a backup every time `readJSON` and `writeJSON`.
Anyway, thank you.

MaBecker, 2020-08-16:
Make sure you are not writing to storage during upload, put you code in a `onInit(){}` function and use a  `setTimeout(onInit,1E3);` to start your code.

vidalouiswang, 2020-08-16:
> Make sure you are not writing to storage during upload, put you code in a `onInit(){}` function and use a `setTimeout(onInit,1E3);` to start your code.

 Your method is same as mine. I am using this method to start my code month ago.

MaBecker, 2020-09-24:
Has issue #1912 cover this issue too?

gfwilliams, 2020-09-24:
I think unlikely as this isn't causing a reboot?

closed 2021-02-25 by vidalouiswang