espruino/Espruino #1909: What about access the palette the sam way as buffer in graphics?

MaBecker, 2020-08-14:
Led panel do not have backlight, dimming them is done be lower or higher color values

Working with led panels and using a 4bit color depth, I thought about changing the brightness.

Espruino has implemented a nice 16 color palette which is fixed as fas as I unterstand.

https://github.com/espruino/Espruino/blob/b9bedd42a50967e048aede8ec188644cbb394acb/libs/graphics/jswrap_graphics.c#L40

Is it possible to make the palette accessible for read and write as `g.palette` like  buffer using `g.buffer`

gfwilliams, 2020-08-17:
The best bet would be:

* Create a 4 bit graphics instance with `g = createArrayBuffer(64,64,4,{msb:true})`
* Create a palette: `pal = new Uint16Array(16)` 
* Wrap this into an image with palette: `img = {width:64,height:64,bpp:4,palette:pal,buffer:g.buffer}`
* Draw the image into a 16 bit graphics instance with `.drawImage`

The other option is you're talking about LEDs? I'm not entirely sure on the use case, but for some of the LCD panels we use http://www.espruino.com/Reference#l_E_mapInPlace to convert paletted data

MaBecker, 2020-08-17:
Thanks, that's a nice way, have to try that.

> I'm not entirely sure on the use case, but for some of the LCD panels we use http://www.espruino.com/Reference#l_E_mapInPlace to convert paletted data

Yes saw that's used  for the [/ILI9163](http://www.espruino.com/modules/ILI9163.js) module.

closed 2020-08-17 by MaBecker