espruino/Espruino #1910: ESP32 DHT11 might not be working/ Software not reading pins?

comcloudway, 2020-08-14:
Connected my DHT11 to my ESP32 (3.3V G22(/D22) nil GND)
Using the default code slightly modified to print the complete temperature object:
`var dht = require("DHT11").connect(D22);
dht.read(function (a) {console.log(a);});`
I get the following results:
`{ "err": true, "checksumError": false,
  "raw": "",
  "temp": -1, "rh": -1 }`
According to the [Espruino DHT11 Documentation Page](http://www.espruino.com/DHT11), `"err":true` means that there was no data recieved

> If no data received at all: {"temp": -1, "rh": -1, err : true, "checksumError": false}.

Tried it out using two different [ESP32 NodeMCU Devkits](https://www.amazon.de/dp/B071P98VTG/?coliid=I3K0PDI77R5J90&colid=276WZLL52L8XD&psc=1&ref_=lv_ov_lig_dp_it), two diffrent [DHT11s](https://www.amazon.de/dp/B07CQVJMB1/?coliid=I2BWABI4TUEA91&colid=276WZLL52L8XD&psc=1&ref_=lv_ov_lig_dp_it)  and using three diffrent espruino firmware versions:
[1v96](https://www.espruino.com/binaries/espruino_1v96_esp32/) , [2v05](https://www.espruino.com/binaries/espruino_2v05_esp32/) and [2v06](https://www.espruino.com/binaries/espruino_2v06_esp32/) and I got the same error every time.

Also tried changing the flashing rate from 40m to 80m but the error was still the same

BTW flashing with this command:
`esptool.py    \
        --chip esp32                                \
        --port /dev/ttyUSB0                         \
        --baud 921600                               \
        --after hard_reset write_flash              \
        -z                                          \
        --flash_mode dio                            \
        --flash_freq 80m                            \
        --flash_size detect                         \
        0x1000 bootloader.bin                       \
        0x8000 partitions_espruino.bin              \
        0x10000 espruino_esp32.bin`

Trying to manually analogRead any pin throws this Error:
`>analogRead(D22)`
`E (5602655) RTC_MODULE: /root/espruino/EspruinoBuildTools/esp32/build/esp-idf/components/driver/rtc_module.c:1428 (adc1_config_channel_atten):ADC Channel Err
E (5602659) RTC_MODULE: /root/espruino/EspruinoBuildTools/esp32/build/esp-idf/components/driver/rtc_module.c:1502 (adc1_get_raw):ADC Channel Err
=0.06298828125
`

Decided to post here instead of the forums because it seems to be an software-related Error and I found two posts suggesting it should work [One over on the Forum](http://forum.espruino.com/conversations/316936/) and the other one [is the corresponding Github issue](https://github.com/espruino/Espruino/issues/1346).

There are many forum posts on how to get the DHT11 working using an ESP32 but none of the solutionions worked for me

MaBecker, 2020-08-14:
DHT11 is presenting the values as a digital sequence. 

Check the data sheet and the module https://www.espruino.com/modules/DHT11.js

If possible, use a DHT22 because it is more reliable.

Search the forum, there are many DHXX posts 

http://forum.espruino.com/search/?q=dht11&defaults=true

comcloudway, 2020-08-15:
I'm shown 7 Forum posts and I've went through all of them before filing an issue and as I mentioned 

> There are many forum posts on how to get the DHT11 working using an ESP32 but none of the solutions worked for me

I also went through the module code but the callback in setWatch is never called in my case even after trying multiple different pins.

Unfortunately I don't own a DHT22 and I know that my DHT is working ( at least it was when using micropython on my esp32)

As stated by you and shown to me by the module source code it never has to read an analog value, that means that my problem isn't related but it still isn't able to read analog values, am I supposed to file an issue for it as I wasn't able to find any posts regarding that issue

MaBecker, 2020-08-15:
So if the watch callback is never triggered,  this could have two reasons:

* there is no response after 50ms -> try 75ms
* check the wiring, possible D22 is not connected to `2 (S)` or missing GND or ........

use this code for further investigation

```
Modules.addCached("ST7735S-UB-MAC", function() {
    function DHT11(pin) {
        this.pin = pin;
    }

    DHT11.prototype.read = function(cb, n) {
        if (!n) n = 10;
        var d = "";
        var ht = this;
        digitalWrite(ht.pin, 0);
        pinMode(ht.pin, "output"); // force pin state to output
        // start watching for state change
        this.watch = setWatch(function(t) {
            d += 0 | (t.time - t.lastTime > 0.00005);
        }, ht.pin, { edge: 'falling', repeat: true });
        // raise pulse after 1ms
        setTimeout(function() {
            pinMode(ht.pin, 'input_pullup');
            pinMode(ht.pin);
        }, 20);
        // stop looking after 50ms
        setTimeout(function() {
            if (ht.watch) { ht.watch = clearWatch(ht.watch); }
            var cks =
                parseInt(d.substr(2, 8), 2) +
                parseInt(d.substr(10, 8), 2) +
                parseInt(d.substr(18, 8), 2) +
                parseInt(d.substr(26, 8), 2);
            if (cks && ((cks & 0xFF) == parseInt(d.substr(34, 8), 2))) {
                cb({
                    raw: d,
                    rh: parseInt(d.substr(2, 8), 2),
                    temp: parseInt(d.substr(18, 8), 2)
                });
            } else {
                if (n > 1) setTimeout(function() { ht.read(cb, --n); }, 500);
                else cb({ err: true, checksumError: cks > 0, raw: d, temp: -1, rh: -1 });
            }
        }, 50); // <-- increase this in steps of 25 and check you get results
    };

    exports.connect = function(pin) {
        return new DHT11(pin);
    };
});

var dht = require("DHT11").connect(D22);
dht.read(function(a) { console.log("Temp is " + a.temp.toString() + " and RH is " + a.rh.toString()); });
```

comcloudway, 2020-08-15:
Tried what you suggested and increased the number up to 175 but still got the same data returned `{ "err": true, "checksumError": false,
  "raw": "",
  "temp": -1, "rh": -1 }` so I decided to connected the dht to my arduino and it kind of works although I don't think the output alternating between an Error and wrong sensor data counts(actual Temp: 26.7°C)
`Temperature: -21.00°C
Humidity: 82.00%
Error reading temperature!
Error reading humidity!`

comcloudway, 2020-08-15:
Ok weirdly enough after installing Micropython the DHT11 is only recognised when connected to 5V and a 1k Resistor from 5V to G22

When switching from 5v to 3.3v or removing the Resistor it is no longer able to communicate (Basically same setup as with the Espruino Software)

It now reads 20°C and 82% Humidity but while the 20°C seem more reasonable now there are 27°C in here and the approximate Temp. range is +-1 according to the Documentation although I've read 1-4 somewhere.

But I'm still unable to find the problem though will measure the temperature again later when room temp has either fallen or risen to confirm that my dht11 is indeed working.

MaBecker, 2020-08-15:
Well, the DHT11 is not really my first choice of sensor for temp and hum. Seems like your's is broken.

comcloudway, 2020-08-15:
Got it in a starter kit didn't know at that time but when it was working it was really precise (at least for me) I know others saying its pretty inaccurate but for me it was always spot on.

Weirdly it appears that out of 6 dht11s where five were shipped last weak none worked.

Am I supposed to file an issue because my analogRead isn't working? Because it definitely doesn't belong here.

closed 2020-08-29 by MaBecker