espruino/Espruino #1912: Copy data from a file to another file will cause reboot

vidalouiswang, 2020-08-16:
Espruino 2.06, NodeMCU

When copy a file in flash, like `require("Storage").write("newfile",require("Storage").read("oldfile"))` will make ESP8266 reboot. Unless the old file only have a few data. 

In my test, the file have 2000 bytes.

vidalouiswang, 2020-08-16:
Because of the #1908 I have to make a backup for restore. But the problem is how to restore the backup file to main file if this action will cause reboot.
And also Espruino can't rename a file.

MaBecker, 2020-08-16:
try this sample code:

```
s = require('Storage');

onInit= ()=>{
s.write("text",
`Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum. Stet clita kasd gubergren, no sea takimata sanctus est Lorem ipsum dolor sit amet. Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum. Stet clita kasd gubergren, no sea takimata sanctus est Lorem ipsum dolor sit amet. Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum. Stet clita kasd gubergren, no sea takimata sanctus est Lorem ipsum dolor sit amet.   
Duis autem vel eum iriure dolor in hendrerit in vulputate velit esse molestie consequat, vel illum dolore eu feugiat nulla facilisis at vero eros et accumsan et iusto odio dignissim qui blandit praesent luptatum zzril delenit augue duis dolore te feugait nulla facilisi. Lorem ipsum dolor sit amet, consectetuer adipiscing elit, sed diam nonummy nibh euismod tincidunt ut laoreet dolore magna aliquam erat volutpat.   
Ut wisi enim ad minim veniam, quis nostrud exerci tation ullamcorper suscipit lobortis nisl ut aliquip ex ea commodo consequat. Duis autem vel eum iriure dolor in hendrerit in vulputate velit esse molestie consequat, vel illum dolore eu feugiat nulla facilisis at vero eros et accumsan et iusto odio dignissim qui blandit praesent luptatum zzril delenit augue duis dolore te feugait nulla facilisi.   
Nam liber tempor cum soluta nobis eleifend option congue nihil imperdiet doming id quod mazim placerat facer possim assum. Lorem ipsum dolor sit amet, consectetuer adipiscing elit, sed diam nonummy nibh euismod tincidunt ut laoreet dolore magna aliquam erat volutpat. Ut wisi enim ad minim veniam, quis nostrud exerci tation ullamcorper suscipit lobortis nisl ut aliquip ex ea commodo consequat.`);

s.write("copy-of-text",s.read("text"));

};

setTimeout(onInit,1E3);

```

vidalouiswang, 2020-08-16:
> try this sample code:
>
> ```
> s = require('Storage');
>
> onInit= ()=>{
> s.write("text",
> `Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore [...]

Same result. I added "wait" to let you know where it runs.
[image]

MaBecker, 2020-08-16:
Ok, watchdog kicks in, it takes to long to copy in one line.

try to split this into two calls:

```
t = s.read("text");
s.write("copy-of-text",t);

```

Hmm, as you have a 4MB board you could flash the 4MB firmware which has much more save pages, if you plan to save more stuff.
http://www.espruino.com/EspruinoESP8266#build-content

vidalouiswang, 2020-08-16:
I also tried this method
`t = s.read("text");`
`s.write("copy-of-text",t);`

Then it rebooted again.

[image: image.png]

And, I will use ESP8266-01s or NodeMCU for different reasons, because
ESP826-01s have less volume.

Or, are there other methods to copy data?

vidalouiswang, 2020-08-16:
> Ok, watchdog kicks in, it takes to long to copy in one line.
>
> try to split this into two calls:
>
> ```
> t = s.read("text");
> s.write("copy-of-text",t);
> ```
>
> Hmm, as you have a 4MB board you could flash [...]

I also tried this method
`t = s.read("text");`
`s.write("copy-of-text",t);`

Then it rebooted again.

[image]

And, I will use ESP8266-01s or NodeMCU for different reasons, because
ESP826-01s have less volume.

Or, are there other methods to copy data?

MaBecker, 2020-08-16:
> Or, are there other methods to copy data?

Not as far as I know.

gfwilliams, 2020-08-17:
Can you try:

```
require("Storage").write("newfile",""+require("Storage").read("oldfile"))
```

The issue could be that because the files are memory mapped, ESP8266 crashes when it tries to read the file contents byte by byte

vidalouiswang, 2020-08-17:
> Can you try:
> 
> ```
> require("Storage").write("newfile",""+require("Storage").read("oldfile"))
> ```
> 
> The issue could be that because the files are memory mapped, ESP8266 crashes when it tries to read the file contents byte by byte

It works! Thanks a lot!
I should know that when I found `let tmp = s.read("large-file")` only takes 2 blocks of memory! Hahaha.

And again, thanks to you two for helping me!  Have a nice day! :D

vidalouiswang, 2020-08-17:
> > Can you try:
> > ```
> > require("Storage").write("newfile",""+require("Storage").read("oldfile"))
> > ```
> > 
> > 
> > The issue could be that because the files are memory mapped, ESP8266 crashes when it tries to read the file contents byte by byte
> 
> It works! Thanks a lot!
> I should know that when I found `let tmp = s.read("large-file")` only takes 2 blocks of memory! Hahaha.
> 
> @MaBecker And again, thanks to you two for helping me! Have a nice day! :D

MaBecker, 2020-08-17:
Great, have fun and don't miss to [donate ](http://www.espruino.com/Donate)to Gordon!

vidalouiswang, 2020-08-17:
> Great, have fun and don't miss to [donate ](http://www.espruino.com/Donate)to Gordon!

Yeah. 👍

opichals, 2020-08-17:
> The issue could be that because the files are memory mapped, ESP8266 crashes when it tries to read the file contents byte by byte

@gfwilliams Could you help me understand here? Why does it crash on the memmapped string and not on a copy created from it? Is it just because it is too slow and the watchdog kicks in?

MaBecker, 2020-08-17:
> @gfwilliams Could you help me understand here? Why does it crash on the memmapped string and not on a copy created from it? Is it just because it is too slow and the watchdog kicks in?

Hi @opichals, I mixed it, 4 is the watchdog, it is the old ESP8266 issue that cause fault when a byte access is performed, because only word accesses are permitted.

There is  a brach named [ESP8266_RODA](https://github.com/espruino/Espruino/tree/ESP8266_RODATA)T that could handle it if completed, but this stuff is way above my head ......

gfwilliams, 2020-08-17:
As @MaBecker says, it's the issue that ESP8266 can't read single bytes from Flash. There *should* be a compiler flag that allows the compiler to sort it out, but we never had any luck with that and have to put in explicit word reads whenever we think a byte might be in flash (see `READ_FLASH_UINT8/16`).

There's also the possibility to handle the memory access interrupt, read the correct byte and pass it back - which seems great, but which we haven't been able to sort out either.

Right now the Espruino source is peppered with hacks to work around this ESP8266 byte read issue, so it'd be a great thing to sort out properly (rather than just piling on more hacks!).

opichals, 2020-08-17:
@gfwilliams @MaBecker Ah, thanks! Got it. Forgot about this one.

closed 2020-08-19 by vidalouiswang