espruino/Espruino #1913: What about a LED Panel custom library

MaBecker, 2020-08-17:
It is a lot of fun working with those panel and theire are many different kinds available.

Just got my self some P4 with a fm6126 chip and 64x64 rbg leds and added some code on the this [forum post](http://forum.espruino.com/comments/15461100/)

Doing some research on this and found that using `shifOut()` is working great together with the  graphics array buffer when using option interlace.

Found a nice article for STM32 which explains how those panel work

* https://www.galliumstudio.com/2020/01/20/episode-1-a-taste-of-color/

and a repo that has a sample driver

* https://github.com/bikefrivolously/led_matrix

For unsupported boards like the ESP32 there a many i2s dma parallel solution to push out the data extremely fast. 

* implement a custom lib and name it eq led_panel
* pass panel size and bpp
* pass all required pins like R1,G1,B1,R2,G2,B2, A,B,C,D,E(for 64x64), LAT,CLK OE
* return g as jspNewObject(0, "Graphics")
* add function  as connect(), clear() and flip()

What do you thinks about this?

gfwilliams, 2020-08-17:
Personally as JumJum noted in the forum post, I'm not sure this would really be that good a use of time and I doubt it would stay working for long enough to be useful.

What about optimising `shiftOut` for ESP32? You could detect when it is called using pins that can be handled by ESP32 hardware, and then call into an ESP32 specific function that was extremely fast?

MaBecker, 2020-08-17:
> What about optimising shiftOut for ESP32? You could detect when it is called using pins that can be handled by ESP32 hardware, and then call into an ESP32 specific function that was extremely fast?

Thanks, so let's try this first.

closed 2020-08-17 by MaBecker