espruino/Espruino #1914: Banglejs/Tensorflow: Uncaught Error: 4 bytes lost due to alignment.

dhassault, 2020-08-17:
Hi,

When trying the example provided there: https://github.com/espruino/Espruino/tree/master/libs/tensorflow
I get the following error:
```
>Uncaught Error: 4 bytes lost due to alignment. To avoid this loss, please make sure the tensor_arena is 16 bytes aligned.
 at line 4 col 50
var tf = require("tensorflow").create(2048, model);
```
I tried to add `alignas(16) uint8_t tensor_arena[2048];` but didn't work.

Any idea? 

I'm using tensorflow 2.2 and the last build 206.83 on the bangle.js. The model is generated directly from the notebook in colab.

gfwilliams, 2020-08-17:
Thanks - that's an interesting one. I'll take a look

gfwilliams, 2020-08-17:
Ok, this should now be fixed in new builds

commit bf6022c4 (2020-08-17): Align Tensorflow arena to 16 bytes - new Tensorflow requires this (fix #1914)
closed 2020-08-17 by gfwilliams