espruino/Espruino #1915: Why is any GPIO task on ESP32 so much slower than on original Espruino boads?

MaBecker, 2020-08-17:
Those STM32 boards are so much faster when it comes to GPIO's than the ESP32 - amazing.

So why is the ESP32  with 240 MHz and a dual core so much slower?

MaBecker, 2020-08-18:
shiftOut differences:

* Added pin info for bit-banded pin addresses (and jshGetPinAddress) - only for STM32s

*  jshIsPinValid  - not for STM32s

so there is potential for optimisation.

jumjum123, 2020-08-18:
Compare simple functions like:
var a= 0; for(var i = 0; i < 100000;i++) a++;
Is cpu the problem, or compiler, or RTOS, or ???

gfwilliams, 2020-08-18:
I think in the JS execution case it's probably that while the CPU is fast it's having to grab the code to execute out of QSPI flash on demand? I imagine if you found the 'hot' interpreter functions and loaded them into RAM then execution speed would raise dramatically

MaBecker, 2020-08-18:
>  I imagine if you found the 'hot' interpreter functions and loaded them into RAM then execution speed would raise dramatically

Yes, with ESP32 i2s parallel dma you can have about 500fps for a 64x64 panel without any flicker.

MaBecker, 2020-08-19:
Reading the [esp32 technical reference manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) state that pins are register mapped

input 
* one regsiter for D0-D31 and one for D32-D39

output
* one register to set  D0-D31 and one for D32-D39  high
* one register to set  D0-D31 and one for D32-D39  low

**So the  shiftOut will never be as fast as on STM32 boards with technic used for shiftOut**

But ESP32 offers i2s, which can do parallel shiftout with dma, which nearly no cpu impact.

Just added a sample video to this [conversation](http://forum.espruino.com/comments/15471472)

gfwilliams, 2020-08-20:
It's possible that `jshGetPinAddress` could be extended to be an address + bitmask. That would benefit nRF52 greatly as well

MaBecker, 2020-08-20:
The ESP32 has a limit of 4MHz for setting gpio's via register. Just throttled i2s from 40MHz to  3MHz and the result is still ok.

> It's possible that jshGetPinAddress could be extended to be an address + bitmask. That would benefit nRF52 greatly as well

Great! So working on this is a win for supported boards  a well.

For the ESP32 there is one register plus bitmap  to set a pin and a different register plus bitmap to set a pin low.

MaBecker, 2020-08-20:
ok, this is a code snippet that could be used for the ESP32 function `jshGetPinAddress()`

```
typedef struct {
  uint32_t addr;
  uint8_t bitmask;
} JshPinAddrBitMap;

typedef enum {
  JSGPAF_INPUT,
  JSGPAF_OUTPUT,
} JshGetPinAddressFlags;

#define GPIO_OUT_W1TS_REG  0x3FF44008 
//#define GPIO_OUT_W1TC_REG  0x3FF4400C  // (GPIO_OUT_W1TS_REG+4) 
#define GPIO_OUT1_W1TS_REG 0x3FF44014 
//#define GPIO_OUT1_W1TC_REG 0x3FF44018  // (GPIO_OUT1_W1TC_REG+4) 
#define GPIO_IN_REG        0x3FF4403C
#define GPIO_IN1_REG       0x3FF44040

volatile JshPinAddrBitMap jshGetPinAddress(Pin pin, JshGetPinAddressFlags flags) {

  JshPinAddrBitMap pinAddrBitMap = {0,0};

  if (!jshIsPinValid(pin)) return pinAddrBitMap;
 
  // bitmask
  if (pin < 32)
    pinAddrBitMap.bitmask = pin;
  else
    pinAddrBitMap.bitmask = pin-32;

  // pin addr
  if (flags == JSGPAF_INPUT) {
    pinAddrBitMap.addr = GPIO_IN_REG;
    if (pin  > 31) 
      pinAddrBitMap.addr += 4;
  } else {
    pinAddrBitMap.addr = GPIO_OUT_REG;
    if (pin > 31) 
      pinAddrBitMap.addr += 4;
  }
  return (JshPinAddrBitMap) pinAddrBitMap;
}

```

MaBecker, 2024-04-24:
@jumjum123  can you please share your code for this.

jumjum123, 2024-04-25:
Hallo Mark,

welchen Code meinst Du ?
Ich bin heute ab ca. 19:00 zu Hause erreichbar.

Grüsse
Jürgen

MaBecker, 2026-09-22:
Found it
https://github.com/orgs/espruino/discussions/3010

gfwilliams, 2026-09-22:
Looking at this there are some things that could help now:

* It does `setInterval(function(){sfnc(arr); },boardDur);` but `arr` could be bound into sfnc so you can do just `setInterval(sfnc,boardDur);`
* Crucially, we have `require("timer").add({type : "EXEC",...)` now - which will schedule the function to be called in such a way that it will interrupt executing JS code, so even if some JS code is running the display will still be able to update

edit: this is really cool though and would make a great module, as I believe it's totally device independent

MaBecker, 2026-09-22:
And we cloud have a real performance kicker for ESP32 `jshPinSetValue()`

```c

#if ESP_IDF_VERSION_MAJOR>=5
#include "hal/gpio_ll.h"
#endif
#include "soc/gpio_struct.h"

/*
 * High-speed pin-write function.
 * Optimized for IDF 3.x to 5.x and fully compatible with C3, S3, and P4.
 */
void jshPinSetValue(Pin pin, bool value) {
  if (pinInfo[pin].port & JSH_PIN_NEGATED) value = !value;
    gpio_num_t gpioNum = pinToESP32Pin(pin);

#if ESP_IDF_VERSION_MAJOR >= 5  
    // Uses the official low-level hardware abstraction layer from Espressif.
    // Resolves at compile-time to the optimal register instructions for C3, S3, and P4.
    gpio_ll_set_level(&GPIO, gpioNum, value);
  gpio_ll_set_level(&GPIO, gpioNum, value);
#else
  // Manual, ultra-fast register access for IDF 3.x / 4.x legacy targets
  if (gpioNum < 32) {
    if (value) {
      GPIO.out_w1ts = (1ULL << gpioNum);
    } else {
      GPIO.out_w1tc = (1ULL << gpioNum);
    }
  } else {
    if (value) {
      GPIO.out1_w1ts.val = (1ULL << (gpioNum - 32));
    } else {
      GPIO.out1_w1tc.val = (1ULL << (gpioNum - 32));
    }
  }
#endif
}

```

MaBecker, 2026-09-22:
```c

/**
 * Ultra-fast pin-read function optimized for ISR contexts.
 * Fully compatible with IDF 3.x to 5.x and targets like C3, S3, and P4.
 */
bool CALLED_FROM_INTERRUPT jshPinGetValue(Pin pin) {
  gpio_num_t gpioNum = pinToESP32Pin(pin);
  bool value;

#if ESP_IDF_VERSION_MAJOR >= 5
  // Direct hardware register read via low-level abstraction layer
  //value = gpio_ll_get_level(&GPIO, gpioNum);
#else
  /* Legacy ultra-fast direct register read for classic ESP32 */
  if (gpioNum < 32) {
    value = (GPIO.in >> gpioNum) & 0x1;
  } else {
    value = (GPIO.in1.val >> (gpioNum - 32)) & 0x1;
  }
#endif
  if (pinInfo[pin].port & JSH_PIN_NEGATED) value=!value;
  return value;
}

MaBecker, 2026-09-23:
One more reason to switch to idf5

gfwilliams, 2026-09-23:
I just looked at this and even on idf4:

* `gpio_set_level` calls `gpio_hal_set_level` (after a relatively simple check)
* `gpio_hal_set_level` is defined to be `gpio_ll_set_level`

So I'm not 100% sure you're really gaining much in terms of speed. Did you check it?

One thing that'll hurt is the call to `gpio_matrix_out/esp_rom_gpio_connect_out_signal` for every call which I put a FIXME on. I've just put a fix in for that - please can you have a check and see if it works?

gfwilliams, 2026-09-23:
The thing that would *really* speed this up is for STM32 we have a `jshGetPinAddress` function. On STM32 due to bit-banding there's actually a unique address that you can write a value to to flip a pin.

On ESP32/nRF52/others it's more that there's one address you have to write a value to to set, and another to clear, so it's not quite as nice. I just did a branch at https://github.com/espruino/Espruino/tree/jshGetPinAddress - note that it's not tested at all.

Maybe you could give it a try? the classic IDF3 build fails, but otherwise I think we're ok

MaBecker, 2026-09-24:
> The thing that would _really_ speed this up is for STM32 we have a `jshGetPinAddress` function. On STM32 due to bit-banding there's actually a unique address that you can write a value to to flip a [...]

Just flashed the ESP32_IDF5 version from jshGetPinAddress  and figured out that the code is no longer working.......

**for JumJum's code** 
```js
>magic()
=undefined
Uncaught Error: Expecting Object or undefined, got Array
    at :1:9
sfnc(arr);
        ^
```

**for MaBecker's code**
```js
Uncaught Error: Expecting Object or undefined, got Array
    at :1:9
sfnc(arr);
```
        ^

started a module 

```js

// Register module in the cache
Modules.addCached("P4_FM6126_64x64", `
function Led64x64(opts) {

  this.width = opts.width;
  this.height = opts.height;
  this.maxLed = opts.maxLed;
  this.boardDur = opts.interval;

  this.pins = {
    A: opts.A,
    B: opts.B,
    C: opts.C,
    D: opts.D,
    E: opts.E,

    R1: opts.R1,
    G1: opts.G1,
    B1: opts.B1,

    R2: opts.R2,
    G2: opts.G2,
    B2: opts.B2,

    LAT: opts.LAT,
    OE: opts.OE,
    CLK: opts.CLK
  };

  this.tmr = null;
  this.sfnc = null;
  this.dfnc = null;
  this.arr = [];
  this.gr = null;
}

Led64x64.prototype.init = function() {

  this.initFM6126();
  this.initPins();

  return this.initGraphic();
};

Led64x64.prototype.initFM6126 = function() {

  var p = this.pins;
  var maxLed = this.maxLed;

  var C12 = [
    1,1,1,1,1,1,0,0,
    0,0,0,0,0,0,0,0
  ];

  var C13 = [
    0,0,0,0,0,0,0,0,
    0,1,0,0,0,0,0,0
  ];

  digitalWrite(
    [p.OE, p.LAT, p.CLK],
    0b100
  );

  // Register 11

  for (var l = 0; l < maxLed; l++) {

    var y1 = l % 16;

    digitalWrite(
      [
        p.R1,
        p.G1,
        p.B1,
        p.R2,
        p.B2,
        p.G2
      ],
      C12[y1] ? 0b111111 : 0
    );

    digitalWrite(
      p.LAT,
      (l > maxLed - 12) ? 1 : 0
    );

    digitalWrite(p.CLK, 1);
    digitalWrite(p.CLK, 0);
  }

  digitalWrite(p.LAT, 0);
  digitalWrite(p.CLK, 0);

  // Register 12

  for (var k = 0; k < maxLed; k++) {

    var y2 = k % 16;

    digitalWrite(
      [
        p.R1,
        p.G1,
        p.B1,
        p.R2,
        p.B2,
        p.G2
      ],
      C13[y2] ? 0b111111 : 0
    );

    digitalWrite(
      p.LAT,
      (k > maxLed - 13) ? 1 : 0
    );

    digitalWrite(p.CLK, 1);
    digitalWrite(p.CLK, 0);
  }

  digitalWrite(p.LAT, 0);
  digitalWrite(p.CLK, 0);
};

Led64x64.prototype.initPins = function() {

  var p = this.pins;

  this.sfnc = shiftOut.bind(
    null,
    [
      p.R2,
      p.G2,
      p.B2,
      undefined,
      p.R1,
      p.G1,
      p.B1
    ],
    {
      clk: p.CLK
    }
  );

  this.dfnc = digitalWrite.bind(
    null,
    [
      p.OE,
      p.LAT,
      p.LAT,
      p.E,
      p.D,
      p.C,
      p.B,
      p.A,
      p.OE
    ]
  );
};

Led64x64.prototype.initGraphic = function() {

  this.gr = Graphics.createArrayBuffer(
    this.width,
    this.height,
    4,
    {
      interleavex: true
    }
  );

  var ledBuf = this.gr.buffer;

  this.arr = [];

  this.arr.push({
    callback: this.pins.OE.reset.bind(
      this.pins.OE
    )
  });

  var scanLines = this.height / 2;

  for (var i = 0; i < scanLines; i++) {
    this.arr.push(
      new Uint8Array(
        ledBuf,
        i * this.width,
        this.width
      )
    );
    this.arr.push({
      callback: this.dfnc.bind(
        null,
        65 | (i << 1)
      )
    });
  }
  this.arr.push({
    callback: this.pins.OE.set.bind(
      this.pins.OE
    )
  });
  return this.gr;
};

Led64x64.prototype.start = function() {
  if (this.tmr)
    this.stop();
  var me = this;
  this.tmr = setInterval(
    function() {
      me.sfnc(me.arr);
    },
    this.boardDur
  );
};

Led64x64.prototype.stop = function() {
  if (this.tmr) {
    clearInterval(this.tmr);
    this.tmr = null;
  }
};

Led64x64.prototype.clear = function() {
  if (this.gr) {
    this.gr.setBgColor(0);
    this.gr.clear();
  }
};

exports.connect = function(opts) {
  return new Led64x64(opts);
};
`);

var options, panel, g, times;

var BLACK = 0, RED = 1, BLUE = 2, MAGENTA = 3, GREEN = 4, YELLOW = 5, TUTQUOISE = 6, WHITE = 15;

onInit = () => {
  // Pin definitions for the board (e.g., ESP32)
  options = {
    width: 64,
    height: 64,
    maxLed: 256,
    interval: 20, // Refresh rate in ms
    A: D19, B: D23, C: D18, D: D5, E: D15,
    R1: D13, G1: D21, B1: D12,
    R2: D27, G2: D17, B2: D4,
    LAT: D22, OE: D2, CLK: D14,
  };

  // Initialize the matrix
  panel = require("P4_FM6126_64x64").connect(options);
  g = panel.init();
  panel.start();

  //staticTest();
  dynamicTest();
};

staticTest = () => {
  g.setColor(RED);
  g.drawRect(0, 0, 63, 63);
  g.setColor(WHITE);
  g.drawString("Hello!", 10, 25);
};

dynamicTest = () => {
  times = 0; // init counter
  setInterval(function() {
    g.setColor((Math.random() * 10)|0);
    g.fillPoly([Math.random() * 63, Math.random() * 63,
      Math.random() * 63, Math.random() * 63,
      Math.random() * 63, Math.random() * 63
    ], true);
    g.setColor(WHITE);
    g.setFont("6x8", 2).drawString(times, 0, 0, true);
    times++;
  }, 1E3);
};

setTimeout(onInit, 1E3);

```

```js
Uncaught Error: Expecting Object or undefined, got Array
    at :1:15
me.sfnc(me.arr);
              ^
```

Could someone please take a quick look and check the code?

gfwilliams, 2026-09-25:
Did it work on master and it's just the branch that's broken?

MaBecker, 2026-09-25:
> Did it work on master and it's just the branch that's broken?

broken on master too.

gfwilliams, 2026-09-25:
Ok. Just tried on one I had in a box running 2v29.85  (non-IDF5) and it works. `master` IDF5 is broken for me too, but non-IDF5 works! I'm looking into it more now

gfwilliams, 2026-09-25:
Something low level is wrong with the ESP32 IDF5 build for some reason. Still checking but it boils down to:

```
// other builds (including ESP32C3_IDF5)
>print.bind(null,"Hello")() 
Hello

// ESP32 IDF5
>print.bind(null,"Hello")() 
undefined
```

So binding isn't working!

Also FYI my speed improvements may not even work right now because of the 'undefined' pin in shiftOut putting it into 'fallback' mode

gfwilliams, 2026-09-25:
Ok, it's fixed now! I also fixed the speed issue.

So now, running that code before, sfnc took 23ms to run, now it's 12ms. I don't have a display hooked up but it doesn't fail so I'm hopeful. It'd be nice to get that branch merged.

MaBecker, 2026-09-27:
Finish with testing, it is much faster now with those changes mentione in https://github.com/orgs/espruino/discussions/3010

There is one issue which is not  related to https://github.com/espruino/Espruino/tree/jshGetPinAddress

_In WedIDE each stroke  in the repl make  the panel flicker_

have to flash different firmware release to find out when this happens the first time, as this is not a issue on older release like `espruino_2v06.101_esp32`

@gfwilliams Merging the jshGetPinAddress branch into the master branch is fine from my side.

PS:  please close this issue after the merge.

closed 2026-09-28 by gfwilliams