espruino/Espruino #1917: boot failed

vidalouiswang, 2020-08-19:
[image]

I used `require("Storage").write` before this happened. Then it will reboot every 5 seconds and give no response to Serial input, only flash firmware can make it back.

vidalouiswang, 2020-08-19:
It will shows 
```Storage is corrupt.
Erasing Storage Area...
after firmware flashed.

MaBecker, 2020-10-29:
Hi can you provide a code snippet to reproduce the error?

MaBecker, 2021-02-24:
This can also caused by trying to read stored coding with  newer Espruino version. In this case flash_erase and then flash again. Can you still reproduce the error with the latest Espruino version?

vidalouiswang, 2021-02-24:
> This can also caused by trying to read stored coding with newer Espruino version. In this case flash_erase and then flash again. Can you still reproduce the error with the latest Espruino version?

I dont use this code any more and I haven't see this error in the 2.08

closed 2021-02-24 by MaBecker