espruino/Espruino #1920: nRF52 power draw with utility timer

gfwilliams, 2020-08-26:
The utility timer uses TIMER1, and TIMER1 uses ~0.5mA when enabled

Often the utility timer is used for stuff that has to happen quickly and accurately (eg. Serial port emulation, digitalPulse) but there are times (eg. Bangle.js) when we use it just because we want something to run in an IRQ and don't care +/- 1ms when it runs.

It'd be nice if in these cases we could use the Nordic app_timer code used for setInterval/etc since this *does still use IRQs, we just don't use them*

I guess one option might be to allow Utility Timer tasks to be flagged with whether they need high precision or not, and if not they can be handled using app_timer?

gfwilliams, 2020-08-26:
Above changes just modify Bangle.js to use Nordic's native APP_TIMER, which takes the urgency off getting this done

gfwilliams, 2020-08-27:
Mentioned in http://forum.espruino.com/conversations/352818

catch22out, 2020-08-29:
abaaba

closed 2024-02-02 by gfwilliams