espruino/Espruino #1922: MDBT42Q and console.log() if not connect to WEBIDE

MaBecker, 2020-08-28:
Got this test scenario: 

- MDBT42Q with a WiznetIO device.
- running a socket server 
- js test code has a lot of console.log() 
- sending and receiving many test data to the socket server 

When connected via WEBIDE everything works as expected.
When disconnected socket server is not responding anymore  after some amount of calls.
Removing all console.log() fixes the behavior. 

Question: As I like to keep the console.log(), what can be added to the code to fix this?

gfwilliams, 2020-08-28:
Does `E.setConsole(null)` fix it?

I'm not really sure why you'd be having issues, unless maybe the default UART pins D6/D8 (http://www.espruino.com/MDBT42Q#pinout) are being used for something and the UART hasn't been explicitly disabled?

In the future, please could you ask this sort of question on the forum? It feels like the better place for it really.

MaBecker, 2020-08-28:
> I'm not really sure why you'd be having issues, unless maybe the default UART pins D6/D8

No

 > Does E.setConsole(null) fix it?

No

because it is a connection problem between the USB-C Ethernet and my notebook...... it slips out too easily....

 > In the future, please could you ask this sort of question on the forum? It feels like the better place for it really.

shure, will do.

closed 2020-08-28 by MaBecker