espruino/Espruino #1923: memory leak with SPI() or new SPI()

fanoush, 2020-08-31:
as per forum topic http://forum.espruino.com/conversations/348343/
it looks like when creating software SPI or I2C or Serial there is a memory leak when deleting in
```
>var s=SPI();process.memory().free
=2575
>for(var i=0;i<2000;i++)s=SPI();process.memory().free
=570
```
it is same with `s=SPI()` or `s=new SPI()` while both do something slightly different, the leak is in both.

I guess I don't fully understand difference between `var t=new Type()` vs `var t=Type()`, seems to often do same thing e.g. `d=Date()`, `d=new Date()` both creates date however with SPI() it seem different
```
>
>var s=new SPI()
=SPI: {  }
>s
=SPI: {  }
>var s=SPI()
=undefined
>s
=undefined
>
```
Also what is maybe related that there is
```
>global.SPI
=function () { [native code] }
```
so with s=SPI() this method is called while `new SPI()` creates new instance of SPI object, pretty confusing :-)
What is also strange and may be related to the leak is that after `var s=new SPI()` there are two objects in global named SPI as autocomplete shows and it can be deleted once. See
[image]

gfwilliams, 2020-08-31:
Fixed with https://github.com/espruino/Espruino/commit/426ac06a62ca7835cedc0d95003bb195f130ed81

But actually just `s = SPI()` will give you an object, but it won't be a proper instance of SPI. `new` is the thing that creates an actual usable object: https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/new

closed 2020-08-31 by gfwilliams

fanoush, 2020-08-31:
The memory bug looks fixed but software SPI is not working for me. Looks like `SPI()` and `new SPI()` does the same wrong thing now.
```
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v06.119 (c) 2019 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>var s=new SPI()
={  }
>var s=SPI()
={  }
>for(var i=0;i<2000;i++)s=new SPI();process.memory().free
=2565
> 
>var fc=new SPI();fc.setup({sck:D29,miso:D31,mosi:D30,mode:0});
Uncaught Error: Function "setup" not found!
 at line 1 col 21
var fc=new SPI();fc.setup({sck:D29,miso:D31,mosi:D30,mode:0}...
                    ^
>fc
={  }

```

gfwilliams, 2020-08-31:
Oops - thanks! Should have tested that :) Just fixed properly :)

fanoush, 2020-08-31:
It works, bug is fixed.

And now even `s=SPI()` and `s=new SPI()` seems to do same right thing. I hope I understand `new` keyword better now, however the confusion what is better to call `x=new Date()` or `x=Date()` is still there.

from https://stackoverflow.com/a/383503 it looks like without `new` the constructor is called as normal function with 'this' pointing elsewhere. And there  is proposed way to 'fix' it in constructor method so it doesn't matter whether `new` is used or not
```
function foo()
{
   // if user accidentally omits the new keyword, this will 
   // silently correct the problem...
   if ( !(this instanceof foo) )
      return new foo();
```
And with the fix the `s=SPI()` works and I can't see any difference between resulting created objects with or without `new`.
So i guess `new` is better/safer/proper way but it can work without it too. Confusing :-)

commit 3f5c48c1 (2020-08-31): Actual fix for Ensure software SPI/I2C/Serial don't leak a variable when initialised (fix #1923)