espruino/Espruino #1926: Support ESP32-S2

sfranzyshen, 2020-09-05:
Are there plans to support the ESP32-S2 platforms ...

sfranzyshen, 2020-09-07:
It looks like the biggest challenge in supporting the ESP32-S2 is that it requires a minimum of ESP-IDF v4.2. This project is currently on v3.1.3 ... perhaps it could be handled as a separate build target  ... have both the esp32 and esp32s2 targets ...

jumjum123, 2020-09-07:
Some months ago, I checked version V4.1 for Espruino.
After a lot of changes, it was possible to compile.
There was no remarkable benefit of the new version.
On the other hand, V4.1 again eats more RAM and Size of binary was growing
a lot.
Compilation for ESP32-S2 would need additional work. Lot of libs are
different between ESP32 and ESP32S2.
Adding the fact, that speed of ESP32 is not the big winner, and Espressif
does not support Espruino in any way, ESP32 became low priority for me.

closed 2020-09-24 by MaBecker