espruino/Espruino #1927: setInterval/clearInterval - global['\xff'].timers array still growing

fanoush, 2020-09-07:
I have following code that switches between different screens each updating itself via scheduled setInterval. When button is pressed previous interval is cleared and new screen is called to create another interval
``` javascript
function first(){
print("First init");
  return setInterval(function(){
    print("First update");
  },2000);
}
function second(){
print("Second init");
  return setInterval(function(){
    print("Second update");
  },2000);
}
var screens=[first,second];
var currscr=0;
var currint=0;
setWatch(function(s){
  currscr++;if (currscr>=screens.length) currscr=0;
  if (currint>0) {print(`clearInterval(${currint})`);clearInterval(currint);}//
//  setTimeout(function(){
    currint=screens[currscr]();
    print(`currint=${currint}`);
//  },3000);
},BTN1,{ repeat:true, edge:'rising' }
);

```
You can upload to bangle emulator and press BTN few times.

For some reason the interval id is still growing and the   `global['\xff'].timers` is growing with all items in array except the last being undefined.

When trying to schedule the screen init code later via setTimeout (hoping that clearInterval(currint); needs some delay) it does not solve the issue (because timeouts and intervals are kept in same table, and it needs last item in array to be cleared to not to grow?).

I did not find any code that is trying to shrink the array or use free/undefined spots inside the array. Yet there perhaps is some because when I try from console line by line to run setInterval and clearInterval it _does_ reuse last id. Also with the code above running it is enough to run this from console
``` javascript
clearInterval(currint);currint=0;
```
and the last interval is cleared and array is shrinked.
Maybe it is not shrinking  because of clearInterval/setInterval being run inside setWatch handler?

Growing `global['\xff'].timers` array slowly eats memory variables on each button press/screen change, not good when running on the watch for long time.

fanoush, 2020-09-07:
What I found so far:
setTimeout/setInterval goes to [jsiTimerAdd](https://github.com/espruino/Espruino/blob/0991c8ed676ff6dfe9c807b46109a7cf5876b0b3/src/jsinteractive.c#L2421) which can only add to the end of the array

clearTimeout/Interval goes to [this part](https://github.com/espruino/Espruino/blob/fbaa3bbcf54a742caa63ba815bc13f74ecdf1030/src/jswrap_interactive.c#L599) which only removes item from arra/set to undefined(?)

Anf then there is magic starting [here](https://github.com/espruino/Espruino/blob/0991c8ed676ff6dfe9c807b46109a7cf5876b0b3/src/jsinteractive.c#L2052) but it looks mostly unrelated, don't see any array shrinking there.

Anyway, double checked now in bangle emulator and I see the issue of id increasing and array growing but I don't see free variables going down. This is really strange. I see the array `global['\xff'].timers.length` being =109 but `process.memory().free` is steady around 2368! On my watch it is decreasing steadily each screen switch. Will check more.

fanoush, 2020-09-07:
Oh, one more thing, this is related to debounce setWatch setting, watch on digital pin without debounce stays at same interval id, with e.g. debounce:50 it skips by two and always  increases. Also tried on nrf52 board only with above code and it *does* leak variables (version 2v06.120). Also after stopping it via explicit `clearInterval(currint)` the memory goes back up, see below. So either bangle emulator is older version or it is becasue of emscripten build.
```
>process.memory().free
=2473
Second init
currint=3
Second update
>process.memory().free
=2452
Second update
Second update
clearInterval(3)
First init
currint=5
First update
>process.memory().free
=2452
First update
First update
>process.memory().free
=2452
First update
clearInterval(5)
Second init
currint=7
Second update
>process.memory().free
=2450
Second update
clearInterval(7)
First init
currint=9
First update
clearInterval(9)
Second init
currint=11
Second update
>process.memory().free
=2448
Second update
Second update
clearInterval(11)
First init
currint=13
First update
clearInterval(13)
Second init
currint=15
Second update
clearInterval(15)
First init
currint=17
First update
clearInterval(17)
Second init
currint=19
Second update
clearInterval(19)
First init
currint=21
First update
>process.memory().free
=2444
First update
First update
>clearInterval(currint);currint=0
=0
>process.memory().free
=2463

```

gfwilliams, 2020-09-14:
Thanks - I think the interval number raising is expected, but it shouldn't be leaking memory. I'll see if I can figure out what the issue is.

closed 2020-09-14 by gfwilliams

fanoush, 2020-09-14:
OK, with new version process.memory().free is steady again, thanks.

Also while `global['\xff'].timers` evaluates to long list of 'undefined' which looks a bit scary
``` javascript
=[
  undefined,
  undefined,
  undefined,
  undefined,
  undefined,  ...
  undefined,
  undefined,
  undefined,
  undefined,
  { time: 2050272, interval: 2097152,
    callback: function () { ... }
   }
 ]
```
it looks like the array is in fact sparse and iterating over it should hopefully not be increasingly slower (?) when handling timers in interpreter loop as only one item is printed when iterating like this
``` javascript
>global['\xff'].timers.forEach((i)=>{print(i);})
{ "time": 1027808, "interval": 2097152,
  "callback": function () {print("First update");}
 }
=undefined
>for (i in global['\xff'].timers)print(i);
452
=undefined
```
and also the index is not limited to e.g. 16 bit value so it can possibly go almost forever - well, maybe until double precision make n++ same as n(?), which when tested is around 1e15(?)
```
>y=1e15
=1000000000000000
>++y
=1000000000000001
>++y
=1000000000000002
>y=1e16
=10000000000000000
>++y
=10000000000000000
>++y
=10000000000000000
```

gfwilliams, 2020-09-15:
Yes, the array growing is fine, and expected. Although it should probably wrap at 2^16 just for sanity's sake.

I seem to recall I actually had to add the behaviour because someone complained (before it did the first available free number) - re-using numbers meant that broken code that called clearInterval with an old handle ended up removing new intervals.

commit c143e49c (2020-09-14): Fix memory leak during setWatch with debounce after regression from https://github.com/espruino/Espruino/commit/853cd64c1a05ddc8a6a5384e7f8f806813ded639 (fix #1927)