espruino/Espruino #1928: .bootrst clarification - code vs https://www.espruino.com/Saving

fanoush, 2020-09-11:
the https://www.espruino.com/Saving#boot-process says

> If hasBeenReset wasn't set, it looks in Storage for a file named .bootcde and executes it (see Save on Send below)
If hasBeenReset was set, it looks in Storage for a file named .bootrst and executes it (see Save on Send below)

This says that it runs one or another depending on hasBeenReset flag (related to holding BTN1 at startup or reset() call).

In reality it
- always runs .bootrst no matter whether button is pressed or not at boot time
- never runs .bootcde if .bootrst is present in storage

It this bug or feature?

curent code is here https://github.com/espruino/Espruino/blob/ac33ff7ab453484c06a169bf10f636f8df975dbd/src/jsflash.c#L834
``` c
  JsVar *resetCode = jsfReadFile(jsfNameFromString(SAVED_CODE_BOOTCODE_RESET),0,0);
  if (isReset || resetCode) return resetCode;
  return jsfReadFile(jsfNameFromString(SAVED_CODE_BOOTCODE),0,0);
```
so if resetCode is not null (= .bootrst exists) it always boots it no matter what and skips .bootcde

gfwilliams, 2020-09-14:
Ahh, thanks - I should update the docs.

This seems intentional to me, especially with the IDE's wording - `Executed even after 'reset()' is called`

fanoush, 2020-09-14:
OK, thanks. 

Actually I am not sure why there is the choice in IDE like it is now. if you think about your code you write as a whole and have those two choices to upload for all your code, which one you would chose and why? To me it is like 'brick my device yes or no' type of question.

It would make more sense to me if each was for different type of code e.g. .bootrst would always run at reset (and boot) time so it is suitable for e.g. starting watchdog and .bootcde would be for the normal stuff which can then be prevented from running by holding BTN1 at boot time if there is a bug. But maybe that is what .boot0 and friends are, except that those do not run at reset() time so setting up watchdog in .boot0 while having 'reset on upload' will trigger reboot with long uploads.

Anyway, doesn't matter, I can manage somehow even with current options. Maybe what is more worrying to me with current IDE save to flash buttons is that .bootrst run always and cannot be prevented from running by holding BTN1 at startup. That is what  the `Executed even after 'reset()' is called` IDE label does not tell you. That is also intentional that it works like that?

The only part that mentions this is on the bottom of https://www.espruino.com/Saving 
> On most boards, holding down a button while applying power can be used to force the device to boot without loading any of the saved code.

so it is not clear whether .bootrst should or should not run when holding BTN1 at boot time.

gfwilliams, 2020-09-14:
> Maybe what is more worrying to me with current IDE save to flash buttons is that .bootrst run always and cannot be prevented from running by holding BTN1 at startup

Yes, that is very intentional. The idea is there are cases when you want code to run always - for example if you're making your own bootloader, trying to lock the device down, or making your own home PC sort of thing: http://www.espruino.com/Pixl.js+Home+Computer

However a lot of this is a bit 'legacy' - from before there was even a filesystem. I think it might be sensible to actually remove the 'execute always' option from the IDE as it's trivial to click 'Storage', 'New File', and enter '.bootrst' - which ensures you know what you're doing :)

fanoush, 2020-09-14:
just few days ago there was a thread http://forum.espruino.com/conversations/353479/#15507367 but I see that elsewhere too, when guys get e.g. P8 or DS-D6 or another watch and upload espruino and start writing their first lines -  for some reason they often pick this `Executed even after 'reset()' is called` option as a first one. Maybe they think the code would otherwise be lost after each reboot ? One guy even put 'save()' as a last line of such .bootrst just to be sure it is really saved properly :-)

With your official devices you can at least always enter bootloader by holding button for < 3 seconds and maybe erase storage from there but with stock nordic bootloader there is no easy recovery except SWD or trying to interrupt startup somehow or connect quickly before their code e.g. removes uart service.

gfwilliams, 2020-09-14:
Ok, sorted - hopefully. The option is still there in the Settings page, but it should sort out 90% of the accidental usage issues.

We could always add a `NO_BOOTRST_CODE` #define and then define that for P8/etc ports, since there is probably no good reason to have it on those, and as you say without a proper bootloader it can brick them.

fanoush, 2020-09-14:
Looks great, thank you :-)

As for removing  .bootrst via NO_BOOTRST_CODE, then there would be no way to run code after reset() at all.  Maybe more useful in such case would be .bootrst to run only after reset but not on boot or to be also skipped when holding BTN1 - both are introducing complexity to existing code and changing current way it works (see below). So hiding it from IDE is probably good enough.

 @enaon  is using .bootrst now as a watchdog starter and then possibly eval other storage files from it based on some flags and/on BTN1 state basically emulating like if .bootcde would start after .bootrst.  But it can also be switched to some persistent 'devmode' state so that after each boot/reset() it stays in mostly clean state  waiting for file uploads. Such devmode state solves issue with uploading/overwriting multiple pieces of code that would be otherwise already running. Also if IDE has 'reset before upload' set it would still keep watchdog happy after every reset() but not start partial code until all pieces are uploaded. Then the mode is exited by some further button presses when everything is ready.  Any change to how/when/if .bootrst runs would break this early bootloader/devmode setup or make it more complicated to catch all reboot/reset() cases.

gfwilliams, 2020-09-15:
Ok, closing this now then :)

closed 2020-09-15 by gfwilliams

trin4ik, 2022-02-20:
ok, never mind, ty for help. 
try other devices and other os`s later.