espruino/Espruino #1930: jsiDumpHardwareInitialisation for ESPXX

MaBecker, 2020-09-23:
conversation http://forum.espruino.com/comments/15534057/

It is amazing, there are still so many things I have not realized how the work.

I would suggest to keep those pin settings for ESPXX boards as they, because changing this could break a lot of user code and pepper jsinteractive.c with  ifdef like this 

```
#if defined(ESP32) || defined(ESP8266)
      // don't bother with normal inputs, as they come up in this state (ish) anyway
      if (statem != JSHPINSTATE_GPIO_IN_PULLUP && statem != JSHPINSTATE_ADC_IN) {
        // use getPinMode to get the correct string (remove some duplication)
        JsVar *s = jswrap_io_getPinMode(pin);
        if (s) cbprintf(user_callback, user_data, "pinMode(%p, %q%s);\n",pin,s,jshGetPinStateIsManual(pin)?"":", true");
        jsvUnLock(s);
      }
#else
      // don't bother with normal inputs, as they come up in this state (ish) anyway
      if (statem != JSHPINSTATE_GPIO_IN && statem != JSHPINSTATE_ADC_IN) {
        // use getPinMode to get the correct string (remove some duplication)
        JsVar *s = jswrap_io_getPinMode(pin);
        if (s) cbprintf(user_callback, user_data, "pinMode(%p, %q%s);\n",pin,s,jshGetPinStateIsManual(pin)?"":", true");
        jsvUnLock(s);
      }
#endif
``` 

Please let me know if this is ok.

gfwilliams, 2020-09-24:
I'm not sure I understand what your suggestion is?

I guess one option might be to add a `jshIsPinStateDefault(pin, state)` which we could define as __WEAK in `jshardware_common.c`:

```
__attribute__((weak)) bool jshIsPinStateDefault(Pin pin, JshPinState state) {
  return state == JSHPINSTATE_GPIO_IN || state == JSHPINSTATE_ADC_IN;
}
```

Then ESPxx could override this by adding its own function in `jshardware.c` :

```
bool jshIsPinStateDefault(Pin pin, JshPinState state) {
  return state == JSHPINSTATE_GPIO_IN_PULLUP || state == JSHPINSTATE_ADC_IN;
}
```

MaBecker, 2020-09-24:
Thanks, much better than just removing it from being printed out, will do some tests and send a pr for this.

commit 89426ec7 (2020-09-24): fix #1930
commit a1024cd1 (2020-09-24): Merge pull request #1934 from espruino/fix_#1930

fix #1930
closed 2020-09-24 by gfwilliams