espruino/Espruino #1933: [bug] Changing the BLE uart state after ~5mins leads to a reboot.

enaon, 2020-09-24:
Hello again, I  am not sure again if this is a bug really, but it has been troubling me for some time and I finally pinpointed how to reproduce it. Tested in sdk11 .129 firmware and sdk12  .129 firmware, but it was there before. Those are builds for the dsd6 smartband, but @fanoush said that if the bug is there on the sdk12 build, then most propably it affects all espruinos, so I thought I report it. 

How to reproduce:

1.Reboot the espruino and connect through serial, change the console to serial allways (that part is not needed if you have another way to disable/enable BLE uart), and disable/enable the BLE uart like so. 
```
NRF.setServices(undefined,{uart:false}); NRF.setServices(undefined,{uart:true}); (or with {} instead of undefined)
```
2.Wait at least 4 minutes, 6 for 100% success :) (strange I know, but it needs ~5 minutes from last change)
3.Disable/enable the uart again, and in a few secs(not instantly) the board will reboot, giving no errors in console.

gfwilliams, 2020-09-24:
Thanks - so to reproduce this you call both `setServices` in one line? Or is just one of them enough to do it?

enaon, 2020-09-24:
no one line is not needed, I do it line by line. 
disable it/enable it/wait 5-6 minutes/disable again, that should be enouph.

gfwilliams, 2020-09-24:
Ok, and when does it crash? 

```
NRF.setServices(undefined,{uart:false}); 
NRF.setServices(undefined,{uart:true});
// wait 5-6 mins
NRF.setServices(undefined,{uart:false}); // here?
NRF.setServices(undefined,{uart:true}); // here?
```

enaon, 2020-09-24:
disable/enable/wait/disable=reboot after a few secs.

enaon, 2020-09-24:
No connection to BT durring that time, on the p8 I do it using the gui, on the dsd6 using serial1.

fanoush, 2020-09-24:
I tried it today morning with DS-D6, SDK12 build 2.06.129 but cannot replicate it. tried like 3 times. Over serial with no bluetooth connection run that line with both uart false/true, had longer intervals like 10,15 minutes. Also I had other watch code there. Also had no watchdog enabled, do you have watchdog in manual mode, with your own setInterval kicking it? What watchdog interval and setInterval values you used?

If console still works all the time until reboot and you use manual watchdog, maybe the setInterval watchdog kicking stops firing or its timing gets wrong so it fires too late.

enaon, 2020-09-24:
NIce, I will try without a watchdog. I checked the timer, I have some time bebore it reboots, the watchdog timers was running, but did not check the time. 

thos are the values for the watchdog:
```
function KickWd(){
  if(!BTN1.read())E.kickWatchdog();
}
var wdint=setInterval(KickWd,1000);
E.enableWatchdog(6,false)
```

fanoush, 2020-09-24:
will try with that
btw, adding some print to the kicking interval with current time could help to see when/if it stops firing before it hangs

enaon, 2020-09-24:
on. you were correct. the timer stops, and the watchdog forces a reboot when I change the uart after a few minutes. 
```
Date: Thu Jan 1 1970 00:08:06 GMT+0000
Date: Thu Jan 1 1970 00:08:07 GMT+0000
Date: Thu Jan 1 1970 00:08:08 GMT+0000
Date: Thu Jan 1 1970 00:08:09 GMT+0000
Date: Thu Jan 1 1970 00:08:10 GMT+0000
>NRF.setServices(undefined,{uart:false});
=undefined
>
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v06.129 (c) 2019 G.Williams
```

gfwilliams, 2020-09-24:
Ok, that's interesting. 

So with no extra code, and no watchdog, if you just do a `setInterval(...print...)` then the interval stops being called after the second setServices - but *only* if around 5 mins have passed first?

enaon, 2020-09-24:
Yes, empty code,
With no watchdog, I see no problems, maybe I should look closer. 

With the watchdog, a reboot after a change, only after around 5 mins have passed yes. 
Indeed the time dodnot show, so the timer stoped it seems, but I am allmost sure I checked and the id was changing, I will recheck. 

Can you please try to reproduse it, with the watchdog on?

enaon, 2020-09-24:
yes, the time goes back as you said, that is why the timer is still alive but is not firing. 

Date: Thu Jan 1 1970 00:05:20 GMT+0000
Date: Thu Jan 1 1970 00:05:21 GMT+0000
>NRF.setServices(undefined,{uart:false});print(Date());
Date: Wed Dec 31 1969 23:56:49 GMT+0000
=undefined
>NRF.setServices(undefined,{uart:false});print(Date());

enaon, 2020-09-24:
ok so the title needs changing, is this indeed happens to all espruinos. 

the uart change causes time drift, which causes timers to not work. 

This is on a clean system after E.reboot(), no watchdog, 

```
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>
>NRF.setServices(undefined,{uart:false});print(Date());
Date: Thu Jan 1 1970 00:00:11 GMT+0000
=undefined
>NRF.setServices(undefined,{uart:true});print(Date());
Date: Thu Jan 1 1970 00:00:20 GMT+0000
=undefined
>print(Date());
Date: Thu Jan 1 1970 00:03:43 GMT+0000
=undefined
>print(Date());
Date: Thu Jan 1 1970 00:06:40 GMT+0000
=undefined
>NRF.setServices(undefined,{uart:false});print(Date());
Date: Wed Dec 31 1969 23:58:14 GMT+0000
=undefined
```

enaon, 2020-09-24:
ok, I found a way to walk around it, but it is strange. 

If during the first 5 minutes I call NRF.setServices(undefined,{uart:true}); every minute or so, then the problem dissapears, meaning that start waiting at the 5th minute, when I call setServices again at the 15th minute, the problem is not there anymore.

gfwilliams, 2020-09-24:
Ok, thanks. I think I've got a good idea what the issue is - it depends on whether the top bit of the 24 bit RTC is set when the softdevice restarts :)

enaon, 2020-09-24:
nice, thanks :)

closed 2020-09-24 by gfwilliams

gfwilliams, 2020-09-24:
Thanks - thinking about it I reckon this could fix some instability issues on Bangle.js as well since we have the watchdog on that.

enaon, 2020-09-24:
it was driving me nuts, It was messing up the gui, now I know it is because ot timers gone wild, and was rebooting the p8, but it looked so random, and also it could be avoided if one played with the on/off for some time to test, that I am really happy you found it. :)

enaon, 2020-09-24:
I  am sure you know, but as a confirmation, I just tested and the bug is gone, thanks again :)

commit 810d36c1 (2020-09-24): nRF52: Restarting softdevice no longer changes the system time by 5 mins sometimes (fix #1933)