espruino/Espruino #1935: nRF52 writable NFC

gfwilliams, 2020-09-25:
Right now you can make your own NFC tag with Puck/Pixl.js with the NFCtag module or going bare metal like  http://forum.espruino.com/conversations/354213/

Thing is we already support reading NFC tags for `nfcURL`/etc - there's no reason that code (in the handler for `BLEP_NFC_RX`) couldn't optionally handle writing too, since the data is in RAM. It'd make doing this sort of thing much easier.

DanTheMan827, 2021-08-06:
It depends on the polling device, but I found is that depending on the device it may or may not work within the timing of the event loop.

I made a writable tag mostly conforming to the NTAG215 protocol by putting the hal_nfc function pointers inside of an array and exposing the pointer to that array as a javascript variable.

That method works and the tag is readable and writable by devices expecting an NTAG215 (although the protocol isn't 100% implemented yet)

Perhaps it would be a good idea to add a signature like `NRF.nfcStart([native function])` so people could implement their own tag protocols? It would however require exposing at minimum `hal_nfc_send` and `hal_nfc_send_rsp`

The idea is that you'd have the user give a function conforming to `hal_nfc_callback_t` which would be used as the NFC callback bypassing the event loop completely

Just a thought.

Here's the code I wrote making use of the hal functions directly, and aside from signal issues due to the antenna size it works fine.
```c
// void init(int)
// void setTagPointer(int)
// void setBufferPointer(int)
// bool getTagWritten()
// int startNfc()
// int stopNfc()
// int restartNfc()
typedef unsigned int size_t;
typedef unsigned int ret_code_t;
typedef unsigned char uint8_t;
typedef enum {
    HAL_NFC_EVENT_FIELD_ON,
    HAL_NFC_EVENT_FIELD_OFF,
    HAL_NFC_EVENT_DATA_RECEIVED,
    HAL_NFC_EVENT_DATA_TRANSMITTED
} hal_nfc_event_t;
typedef enum {
    HAL_NFC_PARAM_ID_TESTING,
    HAL_NFC_PARAM_ID_NFCID1,
    HAL_NFC_PARAM_ID_INTERNAL,
    HAL_NFC_PARAM_ID_UNKNOWN
} hal_nfc_param_id_t;
typedef void (* hal_nfc_callback_t)(void          * p_context,
                                    hal_nfc_event_t event,
                                    const uint8_t * p_data,
                                    size_t          data_length);
unsigned int *native;
ret_code_t hal_nfc_setup(hal_nfc_callback_t callback, void * p_context){
  return ( (ret_code_t(*)(hal_nfc_callback_t, void *))native[0])(callback, p_context);
}
ret_code_t hal_nfc_parameter_set(hal_nfc_param_id_t id, const void * p_data, size_t data_length){
  return ( (ret_code_t(*)(hal_nfc_param_id_t, const void *, size_t))native[1])(id, p_data, data_length);
}
ret_code_t hal_nfc_parameter_get(hal_nfc_param_id_t id, void * p_data, size_t * p_max_data_length){
  return ( (ret_code_t(*)(hal_nfc_param_id_t, void *, size_t *))native[2])(id, p_data, p_max_data_length);
}
ret_code_t hal_nfc_start(void){
  return ( (ret_code_t(*)())native[3])();
}
ret_code_t hal_nfc_send(const uint8_t * p_data, size_t data_length){
  return ( (ret_code_t(*)(const uint8_t *, size_t))native[4])(p_data, data_length);
}
ret_code_t hal_nfc_send_rsp(const uint8_t p_data, size_t data_length){
  return ( (ret_code_t(*)(const uint8_t, size_t))native[5])(p_data, data_length);
}
ret_code_t hal_nfc_stop(void){
  return ( (ret_code_t(*)())native[6])();
}
ret_code_t hal_nfc_done(void){
  return ( (ret_code_t(*)())native[7])();
}
ret_code_t startNfc(void);
ret_code_t stopNfc(void);
ret_code_t restartNfc(void);
void print(const char *text){
  JsVar *p = jspGetNamedVariable("print");
[... 279 lines cut ...]
```
Then in the javascript side of things...
```javascript
var buffer = new Uint8Array(24);
var tag = new Uint8Array(572);
var native = E.compiledC(...);
function onInit(){
  tag[0] = 0x04;
  tag[1] = Math.round(Math.random() * 255);
  tag[2] = Math.round(Math.random() * 255);
  tag[3] = tag[0] ^ tag[1] ^ tag[2] ^ 0x88;
  tag[4] = Math.round(Math.random() * 255);
  tag[5] = Math.round(Math.random() * 255);
  tag[6] = Math.round(Math.random() * 255);
  tag[7] = Math.round(Math.random() * 255);
  tag[8] = tag[4] ^ tag[5] ^ tag[6] ^ tag[7];
  tag.set([0x48, 0x00, 0x00, 0xE1, 0x10, 0x3E, 0x00, 0x03, 0x00, 0xFE], 0x09);
  tag.set([0xBD, 0x04, 0x00, 0x00, 0xFF, 0x00, 0x05], 0x20B);

  native.init(process.env.NFCPTR);
  native.setBufferPointer(E.getAddressOf(buffer, true));
  native.setTagPointer(E.getAddressOf(tag, true));
  native.startNfc();
}
```

DanTheMan827, 2021-08-13:
@gfwilliams Would this be a change that would maybe be worthwhile to make? or perhaps something similar to it...

Could expose these methods to the online compiler service and in the case of a device without them, just set the pointer to an empty function.
```diff
diff --git a/src/jswrap_process.c b/src/jswrap_process.c
index f6c738c3..b8bd9cbe 100644
--- a/src/jswrap_process.c
+++ b/src/jswrap_process.c
@@ -23,6 +23,9 @@
 #ifdef PUCKJS
 #include "jswrap_puck.h" // process.env
 #endif
+#ifdef USE_NFC
+#include "hal_t2t/hal_nfc_t2t.h"
+#endif

 /*JSON{
   "type" : "class",
@@ -93,6 +96,18 @@ const void *exportPtrs[] = {
 };
 #endif

+#ifdef USE_NFC
+const void *nfcPtrs[] = {
+    hal_nfc_setup,
+    hal_nfc_parameter_set,
+    hal_nfc_parameter_get,
+    hal_nfc_start,
+    hal_nfc_send,
+    hal_nfc_send_rsp,
+    hal_nfc_stop,
+    hal_nfc_done,
+    0
+};
+#endif
+
 /*JSON{
   "type" : "staticproperty",
   "class" : "process",
@@ -127,6 +142,7 @@ JsVar *jswrap_process_env() {
   // Pointer to a list of predefined exports - eventually we'll get rid of the array above
   jsvObjectSetChildAndUnLock(obj, "EXPTR", jsvNewFromInteger((JsVarInt)(size_t)exportPtrs));
 #endif
+#ifdef USE_NFC
+  jsvObjectSetChildAndUnLock(obj, "NFCPTR", jsvNewFromInteger((JsVarInt)(size_t)nfcPtrs));
+#endif
   return obj;
 }
```

gfwilliams, 2021-08-16:
Honestly, I don't see this as something that's going to benefit many users. I could be wrong (please post up if you read this and you want it too!), but it feels like it's literally just you. And even then it's not useful by itself.

As discussed on the forum threads we'd really have to have some kind of shortcut built into the NFC library that called compiled code without hitting the event loop as well - and then if that happens, and someone had previously used JsVars in their code (or the `compiled` keyword in JS) it'd likely make their code completely unstable.

I spent pretty much a whole day last week trying to get the code size of the Puck.js build down enough that it'll still fit in available flash, so I'm afraid right now I'm not really feeling like pulling in things that negate all that work.

However I'm happy to pull this code and other changes for NFC into a branch so it's easy for others to compile, and we can see how it works?

DanTheMan827, 2021-08-16:
I won't deny that being able to access NFC at this low of a level is a niche, but it does seem that calling it in this fashion from compiled C allows the NFC callback to completely bypass the JS event loop since it's called from an interrupt (correct me if I'm wrong)

The only change I've made to the firmware in my build is exposing the pointers to the nfc hal functions, from there I call them from native C code compiled through the web ide.

When testing, I did tweak the NFC code to remove the increased max delay and set it to the NFC spec of 302us and my native code was able to run within those constraints, even the simplest javascript making use of the rx callback failed.

mmarquez76, 2021-11-08:
Chiming in to say this would be pretty useful for me as well, and would make the Puck a much more powerful NFC emulator.