espruino/Espruino #1941: save() cause ERROR: >> jshKickWatchDog Not implemented,using taskwatchdog from RTOS

MaBecker, 2020-10-08:
save () works as designed, but cause many ERROR for ESP32

```js
>save()
=undefined
WARNING: gap set scan error code = 103
WARNING: set rssi scan not implemeted yet
Compacting Flash...
ERROR: >> jshKickWatchDog Not implemented,using taskwatchdog from RTOS
......
ERROR: >> jshKickWatchDog Not implemented,using taskwatchdog from RTOS
Calculating Size...
Writing..
Compressed 65600 bytes to 2597
```

For ESP32 jsError is used and this is causing the output.

```c
 // Kick the watchdog
void jshKickWatchDog() {
    jsError(">> jshKickWatchDog Not implemented,using taskwatchdog from RTOS");
}
```

**What about this, as long as there is not implementation for  taskwatchdog?**
```c
void jshEnableWatchDog(JsVarFloat timeout) {
  UNUSED(timeout);
#ifdef DEBUG
  jsError(">> jshEnableWatchDog Not implemented,using taskwatchdog from RTOS");
#endif
}

// Kick the watchdog
void jshKickWatchDog() {
#ifdef DEBUG
  jsError(">> jshKickWatchDog Not implemented,using taskwatchdog from RTOS");
#endif
}
```

MaBecker, 2020-10-08:
Just tested and as result process is missing env, memoy and version - any hints?

```js
>process.
apply               bind
call                replaceWith
clone               emit
hasOwnProperty      length
on                  removeAllListeners
removeListener      toString
valueOf
```

gfwilliams, 2020-10-09:
Yes, seems fine to remove the error if it's not causing problems.

Your missing stuff might be because you `save()`d your code (including `process`) then uploaded new firmware.

Normally that wouldn't be an issue because the git commit hash is checked, but since you made changes and I guess didn't commit them yet, the git commit hash is the same :)

commit 17ff204c (2020-10-09): fix #1941
commit fd26bfb9 (2020-10-09): Merge pull request #1942 from espruino/esp32_improvement

fix #1941
closed 2020-10-09 by gfwilliams