espruino/Espruino #1953: E.sum() hangs if the result is 0 on latest builds

AkosLukacs, 2020-10-30:
So, looks like printing the result of a sum, if that sum is zero hangs Espruino on the latest builds.
Tested on Pixl.JS and Pico, the 2.08 release works fine, `E.sum([])` or `E.sum(new Uint32Array(60))` returns `0` just fine.

But on 2v08.73 and 2v08.75 hangs if the sum of all elements is zero (found out this with an empty array). But for example `E.sum([42])`, or if there is one  does work on the latest.

Haven't tried earlier builds, so can't pinpoint the break, but since looks like it hangs on printing the result.
Here are some examples:

```JS
>E.sum([12, -12.0001])  // this works
=-0.0001

>E.sum([]) // hangs, time to reset it...

>E.sum(new Uint32Array(60))  // hangs, time to reset it...

>E.sum([12, -12]) // hangs, time to reset it...

>var foo = E.sum([12, -12]); var bar = 'x' // the second statement is here to prevent immediate printing, or it would hang right here...
="x"
>bar
="x"
>foo == bar
=false
>foo == 0
=true  // it is zero...
>foo 
=  // and hangs here

gfwilliams, 2020-10-30:
Thanks - I think I know what the issue is: https://github.com/espruino/Espruino/commit/1e6586a80cd61063a11f63bb026ac5db35735d53

I'll get on this

gfwilliams, 2020-10-30:
Ok, just fixed - thanks for the clear bug report.

It turns out even just typing `0.0` would kill it :)

commit 9b223914 (2020-10-30): Fix regression printing 0.0, fix #1953
closed 2020-10-30 by gfwilliams