espruino/Espruino #1954: E.reboot() on ESP8266 causes  ASSERT(jsvIsArray(arr)) FAILED AT src/jsvar.h:721

MaBecker, 2020-11-01:
Just working on some code and replaced `require("ESP8266").reboot` with `E.reboot()` and ran into this.

snippet

```js
onInit = () => {
  console.log("up and running");

  setTimeout(E.reboot,5E3);
  //setTimeout(require("ESP8266").reboot,5E3);

};

setTimeout(save,1E3);

``` 

output

```js
>ASSERT(jsvIsArray(arr)) FAILED AT src/jsvar.h:721
  #1[r0,l1] Object {
    #2[r1,l2] Name String [1 blocks] "\xFF"      #3[r1,l0] Object {
        #6[r1,l2] Name String [2 blocks] "timers"          #8[r1,l0] Array(0) [ ]
        #9[r1,l2] Name String [2 blocks] "watches"          #11[r1,l0] Array(0) [ ]
        #14[r1,l2] Name String [1 blocks] "net"          #5[r1,l1] String [2 blocks] "\4\0\0\0\0\xFF\xFF\xFF\0\0\0\0"
        #29[r1,l2] Name String [2 blocks] "modules"          #31[r1,l0] Object {
            #33[r1,l2] Name String [1 blocks] "Wifi"              #32[r1,l0] NativeFunction 0x4022d5c4 (0) { }
          }
        #28[r1,l2] Name String [1 blocks] "tz"= int 60
        #17[r1,l2] Name String [2 blocks] "history"          #18[r1,l0] Array(3) [
            #20[r1,l2] Name Integer 0              #24[r1,l0] String [4 blocks] "setTime(1604230927.18);E.setTimeZone(1)"
            #53[r1,l2] Name Integer 1              #39[r1,l0] String [7 blocks] "onInit = () => {\n  console.log(\"up and running\");\n\n  setTimeout(E.reboot,5E3);\n\n};"
            #54[r1,l2] Name Integer 2              #55[r1,l0] String [2 blocks] "setTimeout(save,1E3);"
          ]
        #58[r1,l2] Name String [1 blocks] "init"          #57[r1,l0] String [1 blocks] ""
      }
    #26[r1,l2] Name String [1 blocks] "E"      #27[r1,l0] NativeFunction 0x4022d560 (1) { }
    #38[r1,l2] Name String [2 blocks] "onInit"      #36[r1,l0] Function {
        #35[r1,l2] Name Param "\xFFundefined"           undefined
        #19[r1,l2] Name String [1 blocks] "\xFFcod"          #46[r1,l0] FlatString [5 blocks] "console.log(\"up and running\");\n\n  setTimeout(E.reboot,5E3);"
        #52[r1,l2] Name String [1 blocks] "\xFFlin"= int 2
        #51[r1,l2] Name String [1 blocks] "\xFFths"          undefined
      }
    #67[r1,l2] Name String [2 blocks] "console"      #66[r1,l0] NativeFunction 0x4022d568 (1) { }
  }
---console end---
```

runtime infos
```js
>process.env
={
  VERSION: "2v08.76",
  GIT_COMMIT: "9b223914b",
  BOARD: "ESP8266_4MB",
  FLASH: 0, STORAGE: 196608, RAM: 81920,
  SERIAL: "84f3eb82-0ef6",
  CONSOLE: "Serial1",
  MODULES: "Flash,Storage,hea" ... "r,crypto,neopixel",
  EXPTR: 1073643664 }
> 
```

MaBecker, 2020-11-03:
Any hints, how to fix this?

gfwilliams, 2020-11-03:
Not really... You'd need to know where the error was. However the assertion seems to be in `jsvArrayIsEmpty` which really isn't used much - just in jsinteractive.c - so you could add some print statements to `jswrap_espruino_reboot` and jsinteractive to try and track down when/why it's being called.

Honestly about the only the thing I can think is that `jshReboot` isn't actually rebooting immediately, and then it exits, some more stuff gets run, and it fails because everything has been torn down at that point.

MaBecker, 2022-06-08:
look like it's been fixed  http://forum.espruino.com/comments/16429229/

so let's close this

closed 2022-06-08 by MaBecker