espruino/Espruino #1956: Difficulty rendering complex graphics in a standard way (SVG support?)

rapport, 2020-11-09:
I am still learning Espruino but finding it unusable on a watch and other devices with display because of lack of support for complex graphics. Are there any plans or a current way to add support for rending of SVG...  I think this would solve all the problems I have. If we could for instance render SVG you could create presentations for time, apps and data on a watch just like you see on all the popular smart watches out there! Is there for example a way to include the Cairo rendering engine in the build that does a great job rendering SVG to pixels on the screen?

gfwilliams, 2020-11-09:
Honestly, I think the RAM/processing requirements of the Cairo engine are far too high for Bangle.js. It'd be nice to be proven wrong though.

I think by 'popular smart watches' you're probably referring to Apple Watch and various Android watches? These have maybe 512MB or more of RAM and multi-core GHz processors - so roughly 10,000 times more RAM and 100x more power. That's a huge difference. 

There is however a project to offline-convert an SVG into something that'll render on Bangle.js: http://forum.espruino.com/conversations/345963/

That might help you out, but it sounds like you want to have a DOM of the SVG in RAM and just tweak it there, and while there may be a way to make something a bit like that after much modification of the above code, I think ultimately you will be disappointed.

All I can suggest is to design the UI inside an SVG, and then to export the relevant bits you need to bitmaps.

closed 2020-11-09 by gfwilliams

rapport, 2020-11-09:
Thanks for taking the time to consider this. I think you are right that there is no chance of SVG running on the Bangle.js but if reconsider the implementation for the SMA Q3 I can see it is feasible for that! I think you are also right that the Cairo engine again may be too much for the SMA as well. However, what if we said instead look at something with a much smaller footprint like the resvg https://github.com/RazrFalcon/resvg C library for SMA Q3 could that be feasible to use in Esprunio with this implementation? If so I think that is worthwhile because it would then make it feasible to hack the watch and make it better than the original in addition to the hack to build various novelty application with it!

gfwilliams, 2020-11-10:
With the Q3 there's definitely more headroom, but even so the resvg page says:

> Right now, a resvg CLI application is less than 4MiB and doesn't require any external dependencies.

> resvg is aiming to support only the static SVG subset; e.g. no ... animations.

So it's about 8x larger than available flash memory even so, and I guess one of the things you wanted was probably the ability to modify it on the fly, which it looks like it doesn't handle?

Perhaps you could give me an example of exactly what you'd like to do with it? Maybe an SVG as an example?

It may be possible to offline-convert the SVG to draw calls, so you could get it looking ok and then with minimal work you could animate it.

rapport, 2020-11-10:
Thanks again for your interest and considering this! I can see SVG running on devices like the SMA Q3 so know it is possible although I am not sure what libraries they are running on the devices doing that.

It usually doesn't matter if the SVG animation features are not supported either, animations can be also be created with JavaScript as a separate process such as where you want to flash a symbol on and off as a warning or move a hand. 

Regards what is possible with SVG rendering... So much more that I don't know where to start with describing that! Easier is to compare and you can replicate almost all the visual displays like users now expect to see from experience with using something like a Apple Watch or another high end smart watch with this kind of setup! 

Regards what I do like to do... I like to adapt technology for people with disability, particularly those with dementia, learning difficulties and cognitive impairment. I have worked on many assistive wearable devices for this user group and we generally need a lot of graphics capability even though the screens should be as simple as possible because you often need to combine symbols, pointers, words all together to make it clearer and easier to understand! It is common that people with Dementia can not read a standard clock (as usually provided by vendor) and I often have to change how the clock works to make it easier to understand for users such as removing some of the hands, or creating rising and falling symbols for time or even horizontal timelines for events. Most of my designs will run on higher end smart watches but they are really expensive for the users and they also often don't require most of the APPS and features only in the higher end devices either!  You can see one the prototype designs I made for a Galaxy watch running here in the browser  https://dementia.watch/ but they are not all as complicated as that one.

FYI Lup Yuen Lee has made a load of interesting videos working with the PineTime in Rust and I am also looking at that as a low cost hackable option if Espruino isn't the right tool for the job...

https://www.youtube.com/channel/UCUd12xvwfFCHj5WM_Co1zYg