espruino/Espruino #1957: STM32F4 pull up not set properly at boot - gpio clock not enabled yet when calling jshInitDevices()

fanoush, 2020-11-13:
I have [STM32F411CEU6 board](https://stm32-base.org/boards/STM32F411CEU6-WeAct-Black-Pill-V2.0.html) that needs "input_pullup" on button. Even with correct pull setting in board file it is not set at boot time.

`BTN1.getMode()` shows `"input"` instead of `"input_pullup"`. However after calling `reset()` it is correct. When debugging it looks like `jshPinSetState` correctly triggers write to  `GPIOx->PUPDR` but it has no effect. Looks like this line 
https://github.com/espruino/Espruino/blob/157c4c77ab11a4271a2207095e00084ff8a89f2e/targets/stm32/jshardware.c#L1105
is called too early as GPIO clocks are not yet enabled here. It helped me to move this line below, after clocks are enabled, over here https://github.com/espruino/Espruino/blob/157c4c77ab11a4271a2207095e00084ff8a89f2e/targets/stm32/jshardware.c#L1152

Seems to work for me but not sure if calling it later may not break something else. However I hope it is OK because `jshInitDevices()` is not platform specific so perhaps the STM32 platform should be initialized first before calling common code(?).

Found the solution here https://os.mbed.com/questions/79074/Nucleo-144-boards-and-GPIO-HAL-Not-all-p/ someone had similar issue and solution was to enable GPIO Clock first before setting up GPIO registers.

Not sending PR as maybe you  want to reshuffle it in different way, like moving only clock enabling code to the beginning of the method instead?

gfwilliams, 2020-11-16:
Thanks! Your solution is great, although maybe we should move it later, for instance to: https://github.com/espruino/Espruino/blob/157c4c77ab11a4271a2207095e00084ff8a89f2e/targets/stm32/jshardware.c#L1180

Since it looks like its effect would be overwritten on the original Espruino board anyway. Please could you do a PR? Or I can change it.

That's pretty interesting though as the Espruino WiFi uses an STM32F411 and has a button without a pullup, so I'd have expected there would have been issues (but maybe not because the bootloader already set the clock up?).

closed 2020-11-16 by gfwilliams

fanoush, 2020-11-16:
OK, here it is.
As for bootloader - it may be the reason, I don't use bootloader on this board.

PR #1958 (2020-11-16): move jshInitDevices() after GPIO clocks are enabled
before GPIO clocks are enabled setting up GPIO registers inside jshInitDevices() has no effect (e.g. setting pull up on pin does nothing)
resolves #1957