espruino/Espruino #1959: ERROR: Socket bind failed on Espruino under linux.

Louro, 2020-11-16:
Hey guys,

Trying to make module in espruino that creates a socket at port 1234 and waits for a connection.

On the module  file myModule.js I have.

var net = require('net');

var _q;
var myModule= function(optns){
	var opts = optns || {};
	this.prt = opts.port || 502;
	
	var server = net.createServer(function(c) {
		console.log("New client connected.");
		console.log(c);

		c.on('data', function(data) {onDat(data);});
			
	});
	server.listen(this.prt);
	
	this.server = server;
	
	_q = this;
};

then on the main project in the IDE i have

mymodule = require("myModule").create('1234');

If i call the code from the module it fails.

ERROR: Socket bind failed
Uncaught InternalError: Unable to create socket

running directly from the IDE main project works correctly.

Any ideas?

gfwilliams, 2020-11-16:
You're calling create with a string for `optns`, so `prt` ends up being the default of 502.

On Linux ports below 1024 are only available if you run as `sudo` so chances are that's your problem

Louro, 2020-11-16:
Bah,

What a basic mistake.

Thanks alot,

I can confirm that it works.

closed 2020-11-16 by Louro