espruino/Espruino #1960: Espruino WiFi neopixel disables IRQ

gfwilliams, 2020-11-26:
From: http://forum.espruino.com/conversations/356951

We have two options:

* Properly implement SPI DMA via jshSPISendMany, and then use that to push the NeoPixel data
* Ensure that neopixel send flips the CTS pins on any serial devices that have CTS - this would fix the WiFi case nicely but may still cause issues with other things

gfwilliams, 2020-11-26:
Workaround for now is to manually flip CTS:

```
A15.set();
neopixel.write...
A15.reset();
```

GermanWarez, 2022-06-10:
GPIO15 is free to use, I don't think the workaround will work, unless the serial console is set to "AT+UART 3" during initialization. But this could break comms for some boards and users.

gfwilliams, 2022-06-13:
>  unless the serial console is set to "AT+UART 3" 

It's set to 2 already here: https://github.com/espruino/EspruinoDocs/blob/master/devices/EspruinoWiFi.js#L282

Will that not do it?