espruino/Espruino #1966: NRF.setServices 'advertise' puts advertisements in Scan Response

gfwilliams, 2020-12-16:
As found at:

* http://forum.espruino.com/conversations/357737/ 
* https://github.com/espruino/Espruino/issues/1961

As well as some other posts - it's really catching a lot of people out. Unless they won't fit, services should go in the standard Advertising packet.

The API is also a bit rubbish since the service should really be specified in 'setAdvertising' - we should maybe deprecate the old method.

Also related to https://github.com/espruino/Espruino/issues/1371 Espruino uses the Nordic API to work out the advertising packet, but there isn't really a nice way of detecting when there's too much stuff to advertise. It's so simple we should probably make Espruino create the advertising packet, then we might have some options about how we notify the user and/or start moving advertised services into the scan response.

fanoush, 2020-12-16:
just comment about idea 'put there everything what fits', there are low power guides that mention making advertising packet smallest and return rest only on request (=in scan response). So should overridable in some way for those wanting to maximize battery life, see https://www.novelbits.io/ble-power-consumption-optimization/  part Advertising state - Advertising data length

gfwilliams, 2020-12-16:
Thanks - good idea. Maybe it's best not to even bother doing it automatically and to just error. One issue is the Nordic UART and HID service IDs get automatically added to the scan response when enabled, but maybe we actually just ignore that and when you specify a scan response the UART/HID stuff is overwritten.

Right now `NRF.setAdvertising` basically uses `NRF.getAdvertisingData` to get the data internally.

What I'm hoping is where you want to stick stuff in a scan response, you do `NRF.setScanResponse(NRF.getAdvertisingData(...))`.

**Note to me:** Scan Response is currently broken for SDK15 builds because of the way it needs setting alongside advertising data. This would be a good opportunity to fix that.

gfwilliams, 2023-05-10:
Just a note that some modules/code has to hack around it - like ble_midi which specifies the raw advertising data because it wants to try and get the MIDI UUID in the main advertising packet: https://www.espruino.com/modules/ble_midi.js