espruino/Espruino #1969: Puck.js v2 IR stays on after transmitting

plisiecki, 2021-01-01:
Puck.IR() on Puck.js v2 leaves the IR LED on.  For example, I measured current into the board at ~.4mA.  I then issued Puck.IR([2000]) and saw current increase to 22mA for 2 seconds, then it increased to 84.6mA as stayed there forever, unless I ran another Puck.IR([2000]), in which case, the power dropped to 22mA for 2 seconds and then returned to 84.6mA.  Puck.IR(..., FET) also leaves the FET on.  I hooked a resistor from the FET pin to 3V and saw the same behavior.

The issue appears to be that the "cathode-only" mode assumes that the LED's cathode is connected to the pin and the anode to 3V (presumably with a series resistor in there somewhere).  In that case, "off" is a logical 1.  But, the IR output and FET output each have N-channel MOSFETs, so 1 is actually "on."

```
void _jswrap_puck_IR_off() {
  jshPinOutput(_jswrap_puck_IR_pin, 1);
}
```

I worked around the issue for my project simply by changing "1" to "0", and then the power returns to normal after sending is complete.  This fixed the power issue for both on-board IR and via the FET.

It appears that "0" is the correct value here for IR and FET operation, based on the schematic, my observations, and the initialization code:
```
  // If Puck.js v2 ensure FETs are forced off
  if (isPuckV2) {
    jshPinSetValue(IR_FET_PIN, 0);
    jshPinSetState(IR_FET_PIN, JSHPINSTATE_GPIO_OUT);
    jshPinSetValue(FET_PIN, 0);
    jshPinSetState(FET_PIN, JSHPINSTATE_GPIO_OUT);
  }
```

My simple fix, though, would break any application that may be using a direct-cathode connection, or otherwise assume that "1" is "off."  I am guessing that maintaining that existing behavior for other cathode-only uses would be a requirement here.  (For IR and FET pins, though, it's hard to imagine that this is anything but a bug.)

Perhaps the output off polarity should be a fourth option that defaults to "0" for IR or FET pins and "1" for anything else?  That way, IR and FET output issue is fixed without immediately changing the behavior of other pins? And applications can choose which polarity they want if the default isn't right.

Or Puck.IR could check whether the cathode (only) pin is already an output with value 0, and if so, use 0 for off, else use 1 for off?  (The end result is similar to the previous approach, without actually needing to add an extra argument and treat particular pins specially: Just set the pin to the chosen off state before calling Puck.IR().)

If it would help, I would be happy to work up a patch with some guidance as to the preferred solution (one of the above or something else).

gfwilliams, 2021-01-04:
Wow, thanks for this - I'll get a fix in today.

It's a shame as some fixes got put in a few months ago, but somehow it seems I never tested this!

gfwilliams, 2021-01-04:
Thanks - just fixed. I've actually just added a check to see if the cathode pin is either FET pin on Puck.js and to invert if so.

Annoyingly it looks like this worked in the original 2v05 release and I broke it in an update after that :(

commit d2ed4518 (2021-01-04): Puck.js: Fix regression where IR LED was left on after transmission (fix #1969)
closed 2021-01-04 by gfwilliams