espruino/Espruino #1976: serviceData for 128bit uuids

bettse, 2021-01-30:
With the ultimate goal of displaying it on my pixljs, I was trying to find serviceData I was advertising from another device using a 128 bit service uuid.  I tried setScan and then findDevices and finally realized the issue was that service data for 128 bit uuids are not parsed out (https://github.com/espruino/Espruino/blob/2bb197159b602a462de3de0afe88dadbf15b6056/libs/bluetooth/jswrap_bluetooth.c#L1641).  

I imagine it is probably not worth the time to actually do the code for it, but given that I ran into this, I'd like to suggest adding it as a note in the docs.

bettse, 2021-01-30:
Anyone else running into this, can do as I did and just parse the data themselves.

```javascript

const HEX = 0x10;
const SERVICE_DATA_128BIT_UUID = 0x21;

function parse128BitServiceData(device) {
  const { data, serviceData = {} } = device;
  if (data) {
    dataArray = new Uint8Array(data);
    var index = 0;
    do {
      const length = data[index++];
      const field_type = data[index];
      if (field_type === SERVICE_DATA_128BIT_UUID) {
        const field = dataArray.slice(index + 1, index + 1 + length);
        const uuid = field.slice(0, 16);
        const uuidArray = [].map.call(uuid, x=>x.toString(HEX));
        uuidArray.reverse();
        const uuidHex = uuidArray.join('');
        console.log(uuidArray, uuidHex);
        const value = field.slice(uuid.length);
        serviceData[uuidHex] = value;
        device.serviceData = serviceData;
      }
      index += length;
    } while (index < data.length);
  }
  return device;
}
```

gfwilliams, 2021-02-01:
Thanks! I'll try and get this into the 2v09 release. It should only be a pretty minor tweak.

Not sure how I missed that!

commit b794ce9c (2021-02-02): BLE: Allow 128-bit service data to be decoded (fix #1976)
closed 2021-02-02 by gfwilliams