espruino/Espruino #1986: Espruino does not sleep when low on memory

fanoush, 2021-02-21:
When memory is low the interpreter gets 'randomly' to state when it is busy almost all the time and battery drains over night. In DK08 watch with complex code uploaded to RAM (multiple setWatch and  setIntervals) it can happen when there are still e.g. 340 free variables reported by process.memory.

I can reproduce it with https://www.espruino.com/binaries/travis/master/espruino_2v08.187_mdbt42q.zip flashed to bare nrf52 board with LED on pin 30 and following code pasted line by line into bluetooth console
```
>setBusyIndicator(D30);
=undefined
>var a = new Uint8Array(34000);
=new Uint8Array(34000)
>var b = new Uint8Array(1000);
=new Uint8Array(1000)
>var c = new Uint8Array(1800);
=new Uint8Array(1800)
>process.memory()
={ free: 151, usage: 2349, total: 2500, history: 22,
  gc: 0, gctime: 0.42724609375, blocksize: 16, stackEndAddress: 536928864, flash_start: 0,
  flash_binary_end: 432620, flash_code_start: 442368, flash_length: 524288 }
>c=null
=null
>process.memory()
={ free: 266, usage: 2234, total: 2500, history: 27,
  gc: 0, gctime: 0.6103515625, blocksize: 16, stackEndAddress: 536928864, flash_start: 0,
  flash_binary_end: 432620, flash_code_start: 442368, flash_length: 524288 }
>var c = new Uint8Array(1800); // <-- here it happens !!!
=new Uint8Array(1800)
>process.memory()
={ free: 114, usage: 2386, total: 2500, history: 27,
  gc: 0, gctime: 0.64086914062, blocksize: 16, stackEndAddress: 536928864, flash_start: 0,
  flash_binary_end: 432620, flash_code_start: 442368, flash_length: 524288 }
> 
```
All is good until the second `var c = new Uint8Array(1800);` then LED changes and dimly glows as the busy indicator blinks it. When running `c=null` it goes away and is no longer busy. There are no memory warnings during those tests.

Also notice that first time there are 151 variables free and next time there are only 114. Looks like memory leak but maybe it is just overhead of calling process.memory() for the first time - loading 'process' variable etc?

With clean memory and no code (and no history?) it is only reproducible when  free variables  go really low like this - 150 is quite low indeed. However when code is uploaded to RAM it goes through history line by line so in such case number of free variables reported is much higher when this happens and system appears to otherwise work with no visible issues.
With complex code it is pretty random, it can happen that it goes away after cycling through some watch activities via button and next time it starts again at 'random' time.

BTW I also got twice to the point that just arrow up in console history stopped the effect and arrow down (=go to new empty line) started it. That is also interesting because it looks like when not being on last line of history there is one more free variable reported and when you go down to new empty line then one more variable is allocated (just try `setInterval(()=>{console.log(process.memory().free);},2000)`). And this difference of one variable triggered the effect.

I guess it is related to garbage collect code here https://github.com/espruino/Espruino/blob/master/src/jsinteractive.c#L2278 as the JS_VARS_BEFORE_IDLE_GC is maybe triggered by this issue and I don't see any flag there to stop it, so maybe when 'free' space goes  below this value it calls garbage collection in each idle cycle doing this indefinitely and drain battery (becasue each gc takes relatively long in this state). So if the history is quite full (and is indeed counted into used variables here) it can be triggered even if memory is still in relatively good shape, only filled by long history of uploaded code.

Not sure what is the solution here if it is indeed this idle GC.  Counting history as free variables could at least push this to really low memory situation but still not sure if it makes sense to drain battery by doing repeatedly GC like this if the memory cannot be freed and system still somehow works.

fanoush, 2021-02-21:
I have commented out this 'idle GC if' part there https://github.com/espruino/Espruino/blob/master/src/jsinteractive.c#L2276 and the issue is gone so that was indeed it. Will use it like that and see.  Not sure in what cases it can improve things, without it it seems to work so far and no memory is leaking. It probably never gets executed when memory is not low and then if it gets low it causes this battery drain.

Checked history and before this commit https://github.com/espruino/Espruino/commit/74a6ffaaa07cb4d1c395cae4a50b3739ac7e9e08 it did GC only once before going to sleep(?), that looks saner to me. When resetting `loopsIdling`  to 0 when being 1 it is basically scheduled again over and over? Will try without that line.

fanoush, 2021-02-21:
OK< with line [loopsIdling = 0;](https://github.com/espruino/Espruino/blob/master/src/jsinteractive.c#L2281) removed the issue is not reproducible, I can go below 100 free variables and it is OK. So that might be good compromise - it gets called before going to sleep but only once before each sleep.

gfwilliams, 2021-02-22:
Thanks for this! I think the initial idea was that when it had time it would try and GC before sleep, so the next time around the loop it didn't end up stalling having to GC. `loopsIdling = 0;` definitely feels like a bug though.

I *think* the reasoning might have been that GC takes a while and if an event happened during that time we want to go around the loop again and handle it before going to sleep - but there are better ways of handling that.

fanoush, 2021-02-22:
As for

> Also notice that first time there are 151 variables free and next time there are only 114. Looks like memory leak but maybe it is just overhead of calling process.memory() for the first time - loading 'process' variable etc?

this part is unrelated to this issue but also strange, later I noticed I called process.memory() already above it so not sure what is taking extra variables second time the same array is allocated to same variable. Tried it again on different board and indeed it happens again with slightly different numbers.

gfwilliams, 2021-02-22:
> there are 151 variables free and next time there are only 114

That is very strange. Maybe you could post a new issue with some test code? I just tried allocating stuff here and calling `process.memory()` twice and I don't see any big increases (maybe 2 vars from allocating and storing `process`)

fanoush, 2021-02-22:
OK, can you easily try with that mdbt42q firmware linked and paste same steps if you see it too? But will check more and create another issue or post here explanation if it is something harmless, I guess there is this dumpVariables method I can use to check what changes.

gfwilliams, 2021-02-22:
Ok, thanks - just tried it and yes, I see what you mean now. Will see if I can debug this somehow ;)

fanoush, 2021-02-22:
If you saw the variables go down lower second time I am happy, because now I can't replicate it when trying. But have only DS-D6 here with different slightly older 2.08 build. Have set busy indicator to motor so I can replicate the issue with no sleep as it is vibrating steadily when variables go down however I don't see that difference in variables when allocating same variable second time. Need to try at home with that specific build or I am imagining it.

gfwilliams, 2021-02-22:
I think the issue is not the second call, but the arraybuffer free and reallocation that may cause a leak.

gfwilliams, 2021-02-22:
No, you're right - I can't reproduce it now either!

gfwilliams, 2021-02-22:
Just to add - defining a Uint8Array/etc can use a few vars for sure since it has to declare `Uint8Array/ArrayBuffer` and `process`:

```
>trace()
#1[r2,l1] Object {
  #2[r1,l2] Name String [1 blocks] "\xFF"    #3[r1,l1] Object {
      #6[r1,l2] Name String [2 blocks] "timers"        #8[r2,l0] Array(0) [ ]
      #9[r1,l2] Name String [2 blocks] "watches"        #11[r2,l0] Array(0) [ ]
    }
}
=undefined
>var b = new Uint8Array(1000);
=new Uint8Array(1000)
>process.memory()
={ free: 2149, usage: 101, total: 2250, history: 7,
  gc: 0, gctime: 3.173828125, blocksize: 16, stackEndAddress: 536924248, flash_start: 0,
  flash_binary_end: 411380, flash_code_start: 442368, flash_length: 524288 }
>delete b
=true
>trace()
#1[r2,l1] Object {
  #2[r1,l2] Name String [1 blocks] "\xFF"    #3[r1,l1] Object {
      #6[r1,l2] Name String [2 blocks] "timers"        #8[r2,l0] Array(0) [ ]
      #9[r1,l2] Name String [2 blocks] "watches"        #11[r2,l0] Array(0) [ ]
      #19[r1,l2] Name String [2 blocks] "history"        #17[r1,l0] Array(4) [
          #20[r1,l2] Name Integer 0            #16[r1,l0] String [1 blocks] "trace()"
          #28[r1,l2] Name Integer 1            #26[r1,l0] String [3 blocks] "var b = new Uint8Array(1000);"
          #105[r1,l2] Name Integer 2            #103[r1,l0] String [2 blocks] "process.memory()"
          #24[r1,l2] Name Integer 3            #108[r1,l0] String [1 blocks] "delete b"
        ]
    }
  #25[r1,l2] Name String [2 blocks] "Uint8Array"    #21[r2,l0] NativeFunction 0x410a5 (2313) {
      #29[r1,l2] Name String [2 blocks] "prototype"        #31[r1,l0] Object {
          #32[r1,l2] Name String [2 blocks] "constructor"            #21[r2,l0] ...
 
        }
    }
  #35[r1,l2] Name String [2 blocks] "process"    #104[r1,l0] NativeFunction 0x3f791 (1) { }
}
```

closed 2021-02-22 by gfwilliams

fanoush, 2021-02-23:
Flashed that very same firmware again, copy pasted the same thing (even twice) and see it again - 156 and then only 116, it is pretty repeatable in that setup, here it is with trace() added
```
>reset()
=undefined
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v08.187 (c) 2019 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>setBusyIndicator(D30);
=undefined
>var a = new Uint8Array(34000);
=new Uint8Array(34000)
>var b = new Uint8Array(1000);
=new Uint8Array(1000)
>var c = new Uint8Array(1800);
=new Uint8Array(1800)
>process.memory()
={ free: 153, usage: 2347, total: 2500, history: 16,
  gc: 0, gctime: 0.42724609375, blocksize: 16, stackEndAddress: 536928864, flash_start: 0,
  flash_binary_end: 432620, flash_code_start: 442368, flash_length: 524288 }
>trace()
#1[r2,l1] Object {
  #2[r1,l2] Name String [1 blocks] "\xFF"    #3[r1,l1] Object {
      #6[r1,l2] Name String [2 blocks] "timers"        #8[r2,l0] Array(0) [ ]
      #9[r1,l2] Name String [2 blocks] "watches"        #11[r2,l0] Array(0) [ ]
      #16[r1,l2] Name String [2 blocks] "history"        #21[r1,l0] Array(5) [
          #19[r1,l2] Name Integer 0            #17[r1,l0] String [2 blocks] "setBusyIndicator(D30);"
          #29[r1,l2] Name Integer 1            #27[r1,l0] String [3 blocks] "var a = new Uint8Array(34000);"
          #34[r1,l2] Name Integer 2            #2168[r1,l0] String [3 blocks] "var b = new Uint8Array(1000);"
          #2167[r1,l2] Name Integer 3            #2239[r1,l0] String [3 blocks] "var c = new Uint8Array(1800);"
          #2240[r1,l2] Name Integer 4            #2358[r1,l0] String [2 blocks] "process.memory()"
        ]
    }
  #25[r1,l2] Name String [1 blocks] "a"    #2165[r1,l0] Uint8Array (offs 0, len 34000)      #2164[r1,l0] ArrayBuffer (offs 0, len 34000)        #38[r1,l0] FlatString [2126 blocks] 
  #2237[r1,l2] Name String [1 blocks] "c"    #2357[r1,l0] Uint8Array (offs 0, len 1800)      #2356[r1,l0] ArrayBuffer (offs 0, len 1800)        #2242[r1,l0] FlatString [114 blocks] 
  #2359[r1,l2] Name String [2 blocks] "process"    #2238[r1,l0] NativeFunction 0x48495 (1) { }
}
=undefined
>c=null
=null
>process.memory()
={ free: 268, usage: 2232, total: 2500, history: 23,
  gc: 0, gctime: 0.6103515625, blocksize: 16, stackEndAddress: 536928864, flash_start: 0,
  flash_binary_end: 432620, flash_code_start: 442368, flash_length: 524288 }
>var c = new Uint8Array(1800);
=new Uint8Array(1800)
>process.memory()
={ free: 116, usage: 2384, total: 2500, history: 23,
  gc: 0, gctime: 0.64086914062, blocksize: 16, stackEndAddress: 536928864, flash_start: 0,
  flash_binary_end: 432620, flash_code_start: 442368, flash_length: 524288 }
>trace()
#1[r2,l1] Object {
  #2[r1,l2] Name String [1 blocks] "\xFF"    #3[r1,l1] Object {
      #6[r1,l2] Name String [2 blocks] "timers"        #8[r2,l0] Array(0) [ ]
      #9[r1,l2] Name String [2 blocks] "watches"        #11[r2,l0] Array(0) [ ]
      #16[r1,l2] Name String [2 blocks] "history"        #21[r1,l0] Array(10) [
          #19[r1,l2] Name Integer 0            #17[r1,l0] String [2 blocks] "setBusyIndicator(D30);"
          #29[r1,l2] Name Integer 1            #27[r1,l0] String [3 blocks] "var a = new Uint8Array(34000);"
          #34[r1,l2] Name Integer 2            #2168[r1,l0] String [3 blocks] "var b = new Uint8Array(1000);"
          #2389[r1,l2] Name Integer 5            #2363[r1,l0] String [1 blocks] "trace()"
          #2357[r1,l2] Name Integer 6            #5[r1,l0] String [1 blocks] "c=null"
          #2167[r1,l2] Name Integer 8            #2244[r1,l0] String [3 blocks] "var c = new Uint8Array(1800);"
          #2240[r1,l2] Name Integer 9            #2169[r1,l0] String [2 blocks] "process.memory()"
        ]
    }
  #25[r1,l2] Name String [1 blocks] "a"    #2165[r1,l0] Uint8Array (offs 0, len 34000)      #2164[r1,l0] ArrayBuffer (offs 0, len 34000)        #38[r1,l0] FlatString [2126 blocks] 
  #26[r1,l2] Name String [2 blocks] "Uint8Array"    #22[r2,l0] NativeFunction 0x48e65 (2313) {
      #30[r1,l2] Name String [2 blocks] "prototype"        #32[r1,l0] Object {
          #33[r1,l2] Name String [2 blocks] "constructor"            #22[r2,l0] ...
 
        }
    }
  #2166[r1,l2] Name String [1 blocks] "b"    #2236[r1,l0] Uint8Array (offs 0, len 1000)      #2235[r1,l0] ArrayBuffer (offs 0, len 1000)        #2171[r1,l0] FlatString [64 blocks] 
  #2237[r1,l2] Name String [1 blocks] "c"    #2409[r1,l0] Uint8Array (offs 0, len 1800)      #2408[r1,l0] ArrayBuffer (offs 0, len 1800)        #2242[r1,l0] String [151 blocks] 
  #2359[r1,l2] Name String [2 blocks] "process"    #2238[r1,l0] NativeFunction 0x48495 (1) { }
}
=undefined
> 
```
Do you see anything there what makes the difference? Can I run something else to see more?
BTW I copy pasted first unique lines but after c=null, I used history, cursor up to run same lines again (process.memory and var c= new ...) if it makes any difference. Also the busy indicator blinks fast as it still has that bug, can that make the variable leak somehow?

fanoush, 2021-02-23:
here is also screenshot just in case I copy pasted wrong, had to delete zeroes from the trace output 
[image]

fanoush, 2021-02-23:
Oh see it now! the c variable takes 114 blocks in first output and 151 blocks in second! having same (offs 0 , len 1800)

gfwilliams, 2021-02-24:
Wow, good spot! But actually that is expected and is a 'feature':

```
#2356[r1,l0] ArrayBuffer (offs 0, len 1800)        #2242[r1,l0] FlatString [114 blocks] 
#2408[r1,l0] ArrayBuffer (offs 0, len 1800)        #2242[r1,l0] String [151 blocks] 
```

The other difference is it's a String, not a FlatString.

So what happens is you're so low on memory/it is so fragmented that a Flat String cannot be allocated because there's not enough contiguous space. Rather than breaking, Espruino allocates the array as a String. It's slower, but it at least can be allocated even when the data is fragmented.

fanoush, 2021-02-24:
Oh, I see. So normal string  takes so much more space than flat string? OK then.

fanoush, 2021-02-24:
Anyway it is great that there is no nasty bug :-)

commit 955e6994 (2021-02-22): Fix Espruino not sleeping when very low on free memory (fix #1986)