espruino/Espruino #1987: Uncaught Error: Unableto find or create file

vidalouiswang, 2021-02-22:
Hi, here is a problem on espruino, when I write file it shows:

`
>fs.write(`entry0`, `);net.on("res",e=>{if(e.err)print("err:"+e.err);else{log("data:");print(e.data)}})}})}else log("no available wifi info");fs.write("*entry","1");val("avaEntry",val("entry"));log("entry initialized")})();`, 2048);
Uncaught Error: Unableto find or create file
at line 1 col 230
...y initialized")})();`, 2048);
^
`

This information is from serial after write some files.

vidalouiswang, 2021-02-22:
[image]

vidalouiswang, 2021-02-22:
After that, I try to reboot, then:

`
>fs.write(`entry0`, `);net.on("res",e=>{if(e.err)print("err:"+e.err);else{log("data:");print(e.data)}})}})}else log("no available wifi info");fs.write("*entry","1");val("avaEntry",val("entry"));log("entry initialized")})();`, 2048);
Uncaught Error: Unableto find or create file
at line 1 col 230
...y initialized")})();`, 2048);
^
>E.reboot();
=undefined

ets Jan 8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x40100000, len 2408, room16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

2nd bootversion : 1.6
SPI Speed: 80MHz
SPI Mode : QIO

SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000

����o�;��g|�lld$`#��|{�l�n��o�l ��{�l�l�Storage is corrupt.
Erasing Storage Area...
Erase complete.
[?7l
____ _
| __|___ ___ ___ _ _|_|___ ___
| __|_ -| . | _| | | | | . |
|____|___| _|_| |___|_|_|_|___|
|_| espruino.com
2v06.37 (c) 2019 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:1024/1024, manuf 0x5e chip 0x4016

>
`

Espruino erased all files.

vidalouiswang, 2021-02-22:
Also,
`
>fs.write(`entry0`, `);net.on("res",e=>{if(e.err)print("err:"+e.err);else{log("data:");print(e.data)}})}})}else log("no available wifi info");fs.write("*entry","1");val("avaEntry",val("entry"));log("entry initialized")})();`, 2048);
=true
>E.reboot();
=undefined

ets Jan 8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x40100000, len 2408, room16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

2nd bootversion : 1.6
SPI Speed: 80MHz
SPI Mode : QIO

SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000

����o�;��g|�lld$`#��|{�$�n��o�l`��s�l�l�Storage is corrupt.
Erasing Storage Area...
Erase complete.
[?7l
____ _
| __|___ ___ ___ _ _|_|___ ___
| __|_ -| . | _| | | | | . |
|____|___| _|_| |___|_|_|_|___|
|_| espruino.com
2v08 (c) 2019 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:1024/1024, manuf 0x5e chip 0x4016

>
`

No errors but espruino shows Storage is corrupt.

gfwilliams, 2021-02-23:
You were using an old firmware. Quite a few flash storage write issues were fixed in 2v08.

In 2v08 it'd detect invalid flash storage and erase it to stop having problems, but it should then work

closed 2021-02-23 by gfwilliams

vidalouiswang, 2021-02-23:
> You were using an old firmware. Quite a few flash storage write issues were fixed in 2v08.
> 
> In 2v08 it'd detect invalid flash storage and erase it to stop having problems, but it should then work

Actually I tested two versions of Espruino, please notice the last reply, it's the latest version 2.08. And the problem still exist.

gfwilliams, 2021-02-23:
Can you paste up your full code, on the ESP8266 forum - if that is the device you're using? http://forum.espruino.com/microcosms/925/

We can't tell anything useful from one line of a stack trace I'm afraid - it's not even obvious what device you're using.

vidalouiswang, 2021-02-23:
> Can you paste up your full code, on the ESP8266 forum - if that is the device you're using? http://forum.espruino.com/microcosms/925/
> 
> We can't tell anything useful from one line of a stack trace I'm afraid - it's not even obvious what device you're using.

When this happened Espruino running nothing, no any code in it.
And the actions is just write some large file into the storage. 
First I write those files directly, but the interpreter will stuck in there like : >require("Storage").write("aFile",`some data...`): and it will shows =true when the process done, but it won't. So I write large file slices with 512 bytes.

The whole procdure is I will write some code into the .boot0, code:
`
g = global;
fs = require("Storage");
readFile = fs.read;
write = fs.write;
{
        function DB() {
            this.name = "db";
            if (!readJSON(this.name)) { writeJSON(this.name, {}); }
            this.queue = [];
            this.cache = [];
            this.running = false;
        };
        DB.prototype.findInArray = function(key, cache) {
            if (cache) {
                return this.cache.find(obj => { return obj.key == key });
            } else {
                return this.queue.find(obj => { return obj.key == key });
            }
        };
        DB.prototype.innerUpdate = function(key, value, flush, cache) {
            var target = this.findInArray(key, cache);
            if (target) {
                target.value = value;
                if (!cache) {
                    target.t = time;
                }
            } else {
                target = { key: key, value: value, t: flush ? 0 : time };
                this.queue.push(target);
            }
            this.start();
        };
        DB.prototype.update = function(key, value, flush) {
            this.innerUpdate(key, value, flush); //更新queue中的存在项
            this.innerUpdate(key, value, flush, true); //更新cache中的存在项
        };
        DB.prototype.start = function() {
            if (this.running) { return; }
            this.timer && clearInterval(this.timer);
            this.running = true;
            this.timer = setInterval(() => {
                var t = time;
                var targets = this.queue.filter(obj => { return t - obj.t >= 6000 });
                if (targets.length) {
                    //print("pending " + targets.length);
                    let json = readJSON("db") || {};
                    for (let i of targets) {
                        json[i.key] = i.value;
                    }
                    writeJSON("db", json);
                    this.queue = this.queue.filter(obj => { return t - obj.t < 6000 });
                    if (!this.queue.length) {
                        clearInterval(this.timer);
                        this.running = false;
                    }
                }
            }, 1000);
        };
        DB.prototype.innerBuildCache = function(key, value) {
            var target = this.findInArray(key, true);
            if (target) {
                target.times++;
            } else {
                if (value || typeof value == "string") {
                    this.cache.push({
                        key: key,
                        value: value,
                        times: 1
                    });
                    //print("cleared:" + (this.clearCache() || "0"));
                }
            }
            return value;
        };

        DB.prototype.clearCache = function(length) {
            length = length || 10;
            if (this.cache.length > length) {
                var min = { times: 8e8 };
                for (let i of this.cache) {
                    if (i.times < min.times) {
                        min = i;
                    }
                }
                if (min.hasOwnProperty(key)) {
                    this.cache = this.cache.filter(obj => { return obj.key != min.key });
                    return 1;
                }
            }
        };
        DB.prototype.read = function(key) {
            if (!key) { return; }
            var target = this.findInArray(key, true) || this.findInArray(key);
            if (target) {
                //print("read from cache or queue");
                this.innerBuildCache(key);
                return target.value;
            } else {
                let json = readJSON("db");
                if (json.hasOwnProperty(key)) {
                    return this.innerBuildCache(key, json[key]);
                }
            }
        };

        g.db = new DB();
    }

    g.valuer = loadModule("valuer0");

    val = (key, value, flush) => {
        if (value || typeof value == "string") db.update(key, value, flush);
        else return db.read(key);
    };
    eval(readFile("entry0");
`
And write the entry0, then reboot.
As you can see the entry0 is just wrote, after the reboot, the problem show up.

MaBecker, 2021-02-23:
@vidalouiswang 
connect your ESP826 to an access point, then switch to 
`SETTINGS - COMMUNICATION - Connect via TCP` in the WebIDE and add  your ip-address.
Now you can use a much quicker connection type compared to serial connection.

vidalouiswang, 2021-02-23:
> @vidalouiswang
> connect your ESP826 to an access point, then switch to
> `SETTINGS - COMMUNICATION - Connect via TCP` in the WebIDE and add your ip-address.
> Now you can use a much quicker connection type compared to serial connection.

I think the problem is not about the speed, I write code in vscode and I built myseld a serial tool to upload code into the board.
I can use other methods to upload the code as fast as your says.